MiczFlor/RPi-Jukebox-RFID #1135: Please Help

OlafStyle, 2020-11-11:
Es wird keine Musik abgespielt weder in der Webapp noch per Karte und RFID-Reader

Ich habe hier eine Karte vor mir liegen auf der eine Playlist Names playlist gespeichert ist. In dieser Playlist befindet sich eine Audiodatei die auf diese Karte gelinkt worden ist, aber obwohl dies der Fall ist wird diese Audio-Datei nicht abgespielt. Weder per Karte noch in der Webapp. In der Kommandozeilesteht dann nur bash: (dann die Nummer der Karte) und Kommando nicht gefunden. Außerdem, in der Webapp, wird mir kein "Musikbibliotehk scannen?" vorgeschlagen.
Und in der Webapp kann ich die Lautstärke nicht aufdrehen. Es wird zwar angezeigt, aber dennoch wird es nicht gespeichert.

Was ich benutze:
RasberryPie 3 Modell B (Ich weiß, dass es Pi heißt)
Keine Sound card
Audio output Headset
Stretch

Vielen Dank
MfG Olaf Style

s-martin, 2020-11-17:
Was ist das Ergebnis von `tail -n 500 /var/log/syslog`?

s-martin, 2020-11-21:
Was ist das Ergebnis von `cd /home/pi/RPi-Jukebox-RFID/ && git status | head -2`?

Ist die Karte mit einer Playlist oder einem einzigen Song verknüpft? Hast du versucht die Karte neu zu verknüpfen?

OlafStyle, 2020-12-05:
> Was ist das Ergebnis von `tail -n 500 /var/log/syslog`?

tail -n 500 /var/log/syslog
Dec  5 09:42:46 raspberrypi rsyslogd:  [origin software="rsyslogd" swVersion="8.1901.0" x-pid="384" x-info="https://www.rsyslog.com"] rsyslogd was HUPed
Dec  5 09:42:47 raspberrypi rsyslogd:  [origin software="rsyslogd" swVersion="8.1901.0" x-pid="384" x-info="https://www.rsyslog.com"] rsyslogd was HUPed
Dec  5 09:42:48 raspberrypi systemd[1]: logrotate.service: Succeeded.
Dec  5 09:42:48 raspberrypi systemd[1]: Started Rotate log files.
Dec  5 09:42:49 raspberrypi systemd[1]: man-db.service: Succeeded.
Dec  5 09:42:49 raspberrypi systemd[1]: Started Daily man-db regeneration.
Dec  5 09:42:50 raspberrypi systemd[1]: apt-daily.service: Succeeded.
Dec  5 09:42:50 raspberrypi systemd[1]: Started Daily apt download activities.
Dec  5 09:42:50 raspberrypi systemd[1]: Starting Daily apt upgrade and clean activities...
Dec  5 09:42:51 raspberrypi systemd[1]: systemd-hostnamed.service: Succeeded.
Dec  5 09:42:53 raspberrypi systemd[1]: apt-daily-upgrade.service: Succeeded.
Dec  5 09:42:53 raspberrypi systemd[1]: Started Daily apt upgrade and clean activities.
Dec  5 09:43:17 raspberrypi systemd[1]: phoniebox-idle-watchdog.service: Succeeded.
Dec  5 09:44:19 raspberrypi kernel: [  132.834490] usb 1-1.4: new full-speed USB device number 6 using dwc_otg
Dec  5 09:44:19 raspberrypi kernel: [  132.967808] usb 1-1.4: New USB device found, idVendor=413d, idProduct=2107, bcdDevice= 0.00
Dec  5 09:44:19 raspberrypi kernel: [  132.967827] usb 1-1.4: New USB device strings: Mfr=0, Product=0, SerialNumber=0
Dec  5 09:44:19 raspberrypi kernel: [  132.974979] input: HID 413d:2107 as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4/1-1.4:1.0/0003:413D:2107.0004/input/input3
Dec  5 09:44:19 raspberrypi kernel: [  133.047813] hid-generic 0003:413D:2107.0004: input,hidraw3: USB HID v1.11 Keyboard [HID 413d:2107] on usb-3f980000.usb-1.4/input0
Dec  5 09:44:19 raspberrypi mtp-probe: checking bus 1, device 6: "/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4"
Dec  5 09:44:19 raspberrypi mtp-probe: bus: 1, device: 6 was not an MTP device
Dec  5 09:44:19 raspberrypi systemd-udevd[2340]: Process '/usr/sbin/th-cmd --socket /var/run/thd.socket --passfd --udev' failed with exit code 1.
Dec  5 09:44:19 raspberrypi mtp-probe: checking bus 1, device 6: "/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4"
Dec  5 09:44:19 raspberrypi mtp-probe: bus: 1, device: 6 was not an MTP device
Dec  5 09:47:17 raspberrypi mpd: player: played "Test/Bleibe hier bei uns.mp3"
Dec  5 09:47:48 raspberrypi dbus-daemon[1054]: [session uid=1000 pid=1054] Activating via systemd: service name='org.gtk.vfs.Metadata' unit='gvfs-metadata.service' requested by ':1.7' (uid=1000 pid=1163 comm="pcmanfm --desktop --profile LXDE-pi ")
Dec  5 09:47:48 raspberrypi systemd[1034]: Starting Virtual filesystem metadata service...
Dec  5 09:47:48 raspberrypi dbus-daemon[1054]: [session uid=1000 pid=1054] Successfully activated service 'org.gtk.vfs.Metadata'
Dec  5 09:47:48 raspberrypi systemd[1034]: Started Virtual filesystem metadata service.
Dec  5 09:51:36 raspberrypi dbus-daemon[389]: [system] Activating via systemd: service name='org.freedesktop.hostname1' unit='dbus-org.freedesktop.hostname1.service' requested by ':1.28' (uid=1000 pid=1367 comm="/usr/lib/chromium-browser/chromium-browser-v7 --di")
Dec  5 09:51:37 raspberrypi systemd[1]: Starting Hostname Service...
Dec  5 09:51:37 raspberrypi dbus-daemon[389]: [system] Successfully activated service 'org.freedesktop.hostname1'
Dec  5 09:51:37 raspberrypi systemd[1]: Started Hostname Service.
Dec  5 09:51:37 raspberrypi dbus-daemon[1054]: [session uid=1000 pid=1054] Activating service name='ca.desrt.dconf' requested by ':1.19' (uid=1000 pid=1367 comm="/usr/lib/chromium-browser/chromium-browser-v7 --di")
Dec  5 09:51:37 raspberrypi dbus-daemon[1054]: [session uid=1000 pid=1054] Successfully activated service 'ca.desrt.dconf'
Dec  5 09:52:12 raspberrypi lighttpd[782]: chown: changing ownership of '/home/pi/RPi-Jukebox-RFID/shared/audiofolders/Test/Bleibe hier bei uns.mp3': Operation not permitted
Dec  5 09:52:17 raspberrypi mpd: update: updating Test/Bleibe hier bei uns.mp3
Dec  5 09:52:30 raspberrypi systemd[1]: systemd-hostnamed.service: Succeeded.

> Was ist das Ergebnis von `cd /home/pi/RPi-Jukebox-RFID/ && git status | head -2`?
> 
> Ist die Karte mit einer Playlist oder einem einzigen Song verknüpft? Hast du versucht die Karte neu zu verknüpfen?

Ja habe ich

cd /home/pi/RPi-Jukebox-RFID/ && git status | head -2
Aktualisiere Index: 100% (390/390), Fertig.
Auf Branch master
Ihr Branch ist auf demselben Stand wie 'origin/master'.

s-martin, 2020-12-05:
Hast du, bevor du das log abgerufen hast, versucht den Song abzuspielen? Wenn nein, bitte nochmal versuchen den Song abzuspielen und dann log ausgeben.

OlafStyle, 2020-12-05:
Ich habe zu danken, es klappt jetzt ich weiß zwar grad nicht woran es gelegen hat dass es geklappt hat aber es wird jetzt abgespielt

closed 2020-12-15 by s-martin