MiczFlor/RPi-Jukebox-RFID #1136: 🐛 | Shutdown via gpio_control executes immediately

RiekeIcke, 2020-11-11:
## Bug

When pressing the shutdown button, the script for shutdown executes immediately.

### What I did

I installed the raspberry pi with above mentioned buster image and ran the installer script. I configures the gpio_settings.ini and changed the buttons to the classic layout. Every button works as expected but not the shutdown-button.

### What happened

Phoniebox shuts down immediately.

### I expected this to happen

I wanted to press the button at least 4 seconds before the shutdown begins. The LED should blink a few times (as in previous versions of Phoniebox).

### Further information that might help

This is, what I configured in the gpio_settings.ini. I am using open push buttons.

```
[Shutdown]
enabled: True
Type:  Button
Pin: 3
pull_up: True
hold_time: 4
functionCall: functionCallShutdown
```

## Software

### Base image and version

Raspbian GNU/Linux 10 (buster)

### Branch / Release

develop
up to date with 'origin/develop'.

### Installscript

scripts/installscripts/buster-install-default.sh

## Hardware

### RaspberryPi version

Raspberry Pi 3 Model B Rev 1.2

### RFID Reader

16c0:27db HXGCoLtd Keyboard

### Soundcard

1130:1620 Tenx Technology, Inc.

### Other notable hardware

GPIO Control Settings:
```[Shutdown]
enabled: True
Type:  Button
Pin: 3
pull_up: True
hold_time: 4
functionCall: functionCallShutdown
```

s-martin, 2020-11-21:
Thanks for pointing that out.

I just saw we have a dedicated class defined for shutdown: https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/components/gpio_control/GPIODevices/shutdown_button.py

But this class is never used in https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/components/gpio_control/gpio_control.py and the example configs also don’t reflect this.

 I think we could implement this. @veloxidSchweiz,was there any reason not to use the shutdown class or was it just missed?

veloxidSchweiz, 2020-11-21:
Hi @s-martin ,
The shutdownButton is special that it also has a led option.
I think i did not test it properly and i do have quite some issues with my test hardware at the moment. 
I did the required changes in the gpio_control class
maybe it works:
https://github.com/veloxidSchweiz/RPi-Jukebox-RFID/tree/feature/ShutdownButton

PR #1151 (2021-01-05): activation of shutdown button class in gpio_control
Fixes #1136
closed 2021-01-05 by MiczFlor