MiczFlor/RPi-Jukebox-RFID #1139: 🐛 | GPIO PrevNextControl Buttons sometimes do not work

deMerl, 2020-11-14:
Hi, i have the problem that sometimes the "Previous" and "Next" GPIO Buttons do not work.

From the Log I can see the following:
-If it works
#BUTTON PRESSED
Nov 12 21:25:00 raspberrypi PhonieboxGPIOControl[503]: INFO:GPIODevices.simple_button:PrevNextControlBtn2: executre callback
Nov 12 21:25:00 raspberrypi PhonieboxGPIOControl[503]: INFO:GPIODevices.simple_button:PrevNextControlBtn2: action
Nov 12 21:25:00 raspberrypi PhonieboxGPIOControl[503]: INFO:GPIODevices.two_button_control:functionCall2
Nov 12 21:25:00 raspberrypi PhonieboxGPIOControl[503]: Broilers - Ihr da oben
Nov 12 21:25:00 raspberrypi PhonieboxGPIOControl[503]: [playing] #7/13   0:00/3:55 (0%)
Nov 12 21:25:00 raspberrypi PhonieboxGPIOControl[503]: volume: 53%   repeat: off   random: off   single: off   consume: off

-If it does not work

#BUTTON PRESSED
Nov 12 21:25:13 raspberrypi PhonieboxGPIOControl[503]: INFO:GPIODevices.simple_button:PrevNextControlBtn2: executre callback
Nov 12 21:25:13 raspberrypi PhonieboxGPIOControl[503]: INFO:GPIODevices.simple_button:PrevNextControlBtn2: action
#BUTTON PRESSED
Nov 12 21:25:16 raspberrypi PhonieboxGPIOControl[503]: INFO:GPIODevices.simple_button:PrevNextControlBtn2: executre callback
Nov 12 21:25:16 raspberrypi PhonieboxGPIOControl[503]: INFO:GPIODevices.simple_button:PrevNextControlBtn2: action
#BUTTON PRESSED
Nov 12 21:25:17 raspberrypi PhonieboxGPIOControl[503]: INFO:GPIODevices.simple_button:PrevNextControlBtn2: executre callback
Nov 12 21:25:17 raspberrypi PhonieboxGPIOControl[503]: INFO:GPIODevices.simple_button:PrevNextControlBtn2: action
`
Looks like it does not trigger the "GPIODevices.two_button_control"

This happens randomly. Does work for some minutes than does not for some minutes and so on. Makes no difference if "Forward" or "Rewind".
The rotary for volume and the GPIO for Play/Pause do work all the time.

I hope someone can help push me into the right direction or tell me how to troubleshoot further.

Many thanks and keep up the absolutely fantastic work on this project.

s-martin, 2020-11-17:
@veloxidSchweiz ?

s-martin, 2020-11-21:
Please try to adjust the log level, so we get more info in the log file.

In file `components/gpio_control/GPIODevices/two_button_control.py`

Replace in line 94 
`logging.basicConfig(level='INFO')`
with
`logging.basicConfig(level='DEBUG')`

and then post the log file again.

veloxidSchweiz, 2020-11-21:
@deMerl 
Sorry for the later answer,
This is a strange behaviour....
 which RPi are you using?
Here a debugging branch which might give some additional information:
https://github.com/veloxidSchweiz/RPi-Jukebox-RFID/tree/bufgix/1139_GPIO_PrevNextControl_Buttons_sometimes_do_not_work

deMerl, 2020-11-21:
No Problem, I can bypass this issue by using "Simple Button" Config for Next and Prev Song.
Many thanks for the debug files. Still not sure why it sometimes passes the correct Value and sometimes not. I´m using an RPi 4.
Issue is with: 
> Btn1 False, Btn2 False

> GPIO26 not working

```
Nov 21 20:49:15 raspberrypi PhonieboxGPIOControl[486]: DEBUG:GPIODevices.simple_button:args before: (26,)
Nov 21 20:49:15 raspberrypi PhonieboxGPIOControl[486]: DEBUG:GPIODevices.simple_button:args after: ()
Nov 21 20:49:15 raspberrypi PhonieboxGPIOControl[486]: INFO:GPIODevices.simple_button:PrevNextControlBtn2: execute callback
Nov 21 20:49:15 raspberrypi PhonieboxGPIOControl[486]: INFO:GPIODevices.simple_button:PrevNextControlBtn2: get action - <function functionCallTwoButtons.<locals>.functionCallTwoButtons at 0xb6520a98>
Nov 21 20:49:15 raspberrypi PhonieboxGPIOControl[486]: INFO:GPIODevices.two_button_control:FunctionCallTwoButtons: Btn1 False, Btn2 False
Nov 21 20:49:15 raspberrypi PhonieboxGPIOControl[486]: DEBUG:GPIODevices.two_button_control:No Button Pressed: no action
Nov 21 20:49:15 raspberrypi PhonieboxGPIOControl[486]: INFO:GPIODevices.simple_button:PrevNextControlBtn2: get action - <function functionCallTwoButtons.<locals>.functionCallTwoButtons at 0xb6520a98>
Nov 21 20:49:15 raspberrypi PhonieboxGPIOControl[486]: DEBUG:GPIODevices.simple_button:PrevNextControlBtn2: executed action: <function functionCallTwoButtons.<locals>.functionCallTwoButtons at 0xb6520a98> with args=()
```

> GPIO26 Working

```
Nov 21 20:49:38 raspberrypi PhonieboxGPIOControl[486]: DEBUG:GPIODevices.simple_button:args before: (26,)
Nov 21 20:49:38 raspberrypi PhonieboxGPIOControl[486]: DEBUG:GPIODevices.simple_button:args after: ()
Nov 21 20:49:38 raspberrypi PhonieboxGPIOControl[486]: INFO:GPIODevices.simple_button:PrevNextControlBtn2: execute callback
Nov 21 20:49:38 raspberrypi PhonieboxGPIOControl[486]: INFO:GPIODevices.simple_button:PrevNextControlBtn2: get action - <function functionCallTwoButtons.<locals>.functionCallTwoButtons at 0xb6520a98>
Nov 21 20:49:38 raspberrypi PhonieboxGPIOControl[486]: INFO:GPIODevices.two_button_control:FunctionCallTwoButtons: Btn1 True, Btn2 False
Nov 21 20:49:38 raspberrypi PhonieboxGPIOControl[486]: DEBUG:GPIODevices.two_button_control:Btn1 is pressed, secondary Btn not pressed, action: functionCall1
Nov 21 20:49:38 raspberrypi PhonieboxGPIOControl[486]: INFO:GPIODevices.two_button_control:functionCall1
Nov 21 20:49:38 raspberrypi PhonieboxGPIOControl[486]: INFO:GPIODevices.simple_button:PrevNextControlBtn2: get action - <function functionCallTwoButtons.<locals>.functionCallTwoButtons at 0xb6520a98>
Nov 21 20:49:38 raspberrypi PhonieboxGPIOControl[486]: DEBUG:GPIODevices.simple_button:PrevNextControlBtn2: executed action: <function functionCallTwoButtons.<locals>.functionCallTwoButtons at 0xb6520a98> with args=()
Nov 21 20:49:39 raspberrypi PhonieboxGPIOControl[486]: Broilers - Irgendwas in mir
Nov 21 20:49:39 raspberrypi PhonieboxGPIOControl[486]: [playing] #5/13   0:00/2:50 (0%)
Nov 21 20:49:39 raspberrypi PhonieboxGPIOControl[486]: volume: 46%   repeat: off   random: off   single: off   consume: off
```

> GPIO1 not working

```
Nov 21 20:52:16 raspberrypi PhonieboxGPIOControl[486]: DEBUG:GPIODevices.simple_button:args before: (1,)
Nov 21 20:52:16 raspberrypi PhonieboxGPIOControl[486]: DEBUG:GPIODevices.simple_button:args after: ()
Nov 21 20:52:16 raspberrypi PhonieboxGPIOControl[486]: INFO:GPIODevices.simple_button:PrevNextControlBtn2: execute callback
Nov 21 20:52:16 raspberrypi PhonieboxGPIOControl[486]: INFO:GPIODevices.simple_button:PrevNextControlBtn2: get action - <function functionCallTwoButtons.<locals>.functionCallTwoButtons at 0xb6520a98>
Nov 21 20:52:16 raspberrypi PhonieboxGPIOControl[486]: INFO:GPIODevices.two_button_control:FunctionCallTwoButtons: Btn1 False, Btn2 False
Nov 21 20:52:16 raspberrypi PhonieboxGPIOControl[486]: DEBUG:GPIODevices.two_button_control:No Button Pressed: no action
Nov 21 20:52:16 raspberrypi PhonieboxGPIOControl[486]: INFO:GPIODevices.simple_button:PrevNextControlBtn2: get action - <function functionCallTwoButtons.<locals>.functionCallTwoButtons at 0xb6520a98>
Nov 21 20:52:16 raspberrypi PhonieboxGPIOControl[486]: DEBUG:GPIODevices.simple_button:PrevNextControlBtn2: executed action: <function functionCallTwoButtons.<locals>.functionCallTwoButtons at 0xb6520a98> with args=()
```

> GPIO1 working
```
Nov 21 20:52:54 raspberrypi PhonieboxGPIOControl[486]: DEBUG:GPIODevices.simple_button:args before: (1,)
Nov 21 20:52:54 raspberrypi PhonieboxGPIOControl[486]: DEBUG:GPIODevices.simple_button:args after: ()
Nov 21 20:52:54 raspberrypi PhonieboxGPIOControl[486]: INFO:GPIODevices.simple_button:PrevNextControlBtn2: execute callback
Nov 21 20:52:54 raspberrypi PhonieboxGPIOControl[486]: INFO:GPIODevices.simple_button:PrevNextControlBtn2: get action - <function functionCallTwoButtons.<locals>.functionCallTwoButtons at 0xb6520a98>
Nov 21 20:52:54 raspberrypi PhonieboxGPIOControl[486]: INFO:GPIODevices.two_button_control:FunctionCallTwoButtons: Btn1 False, Btn2 True
Nov 21 20:52:54 raspberrypi PhonieboxGPIOControl[486]: DEBUG:GPIODevices.two_button_control:Btn2 is pressed, action: functionCall2
Nov 21 20:52:54 raspberrypi PhonieboxGPIOControl[486]: INFO:GPIODevices.two_button_control:functionCall2
[... cut ...]
Nov 21 20:52:54 raspberrypi PhonieboxGPIOControl[486]: volume: 46%   repeat: off   random: off   single: off   consume: off
```

> gpio_settings.ini
```
[DEFAULT]
enabled: True
used_pinds = [13]

[... 73 lines cut ...]

[NextSong]
enabled: False
Type:  Button
Pin: 26
pull_up: True
hold_time: 0.3
functionCall: functionCallPlayerNext

[PrevSong]
enabled: False
Type:  Button
Pin: 20
pull_up: True
hold_time: 0.3
functionCall: functionCallPlayerPrev

[Halt]
enabled: False
Type:  Button
Pin: 21
pull_up: True
hold_time: 0.3
functionCall: functionCallPlayerPauseForce
```

s-martin, 2020-11-22:
Could be a electrical issue, too?

veloxidSchweiz, 2020-11-27:
It is interesting it might be related with the bouncetime, i.e. more or less a hardware thing (pull-up/down resistor, capacities on the lines, etc)
At the moment the callback is called and then it check the status of the GPIO pins to decide if it should be a one or two button action. One can see that the readback is False/False in non working cases which is quite strange. 
I could think about a solution that there is a fallback behaviour in case non of the pins is set to true.
Might get that working on the weeekend.

veloxidSchweiz, 2020-11-27:
maybe this helps?
https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/1165

s-martin, 2020-11-30:
PR #1165 provided by @veloxidSchweiz (Thanks!) was just merged to develop branch and could help to debug this issue.

@deMerl, please test this and let us know, if this helps to fix the issue.

s-martin, 2021-02-28:
@deMerl do you have some info, if #1165 helps to fix your issue?

that-coding-human, 2021-03-15:
I am struggling with the same behaviour. At first I was suspecting an electrical issue, and double checked all cabling. I even soldered the cables directly to the arcade button contacts, to eliminate the cable shoes as a possible source for the error.
But it looks like the electrical part is absolutely ok.

When having a look at the journal it looks like every button pressed is correctly recognized, but the corresponding function is not always called. The behaviour seemed really strange. Finally I found this issue, which describes the absolutely exact same behaviour I am experiencing.

Following the example journal output that shows what happens:

`Mar 14 20:18:07 phoniebox PhonieboxGPIOControl[829]: INFO:GPIODevices.simple_button:PrevNextControlBtn2: executre callback
Mar 14 20:18:07 phoniebox PhonieboxGPIOControl[829]: INFO:GPIODevices.simple_button:PrevNextControlBtn2: action
Mar 14 20:18:31 phoniebox PhonieboxGPIOControl[829]: INFO:GPIODevices.simple_button:VolumeControlBtn2: holdAndRepeatHandler
Mar 14 20:18:31 phoniebox PhonieboxGPIOControl[829]: INFO:GPIODevices.simple_button:VolumeControlBtn2: action
Mar 14 20:18:40 phoniebox PhonieboxGPIOControl[829]: INFO:GPIODevices.simple_button:VolumeControlBtn2: holdAndRepeatHandler
Mar 14 20:18:40 phoniebox PhonieboxGPIOControl[829]: INFO:GPIODevices.simple_button:VolumeControlBtn2: action
Mar 14 20:18:41 phoniebox PhonieboxGPIOControl[829]: INFO:GPIODevices.simple_button:VolumeControlBtn2: holdAndRepeatHandler
Mar 14 20:18:41 phoniebox PhonieboxGPIOControl[829]: INFO:GPIODevices.simple_button:VolumeControlBtn2: action
Mar 14 20:18:41 phoniebox PhonieboxGPIOControl[829]: INFO:GPIODevices.two_button_control:functionCall1
Mar 14 20:18:41 phoniebox PhonieboxGPIOControl[829]: OK MPD 0.21.4
Mar 14 20:18:41 phoniebox PhonieboxGPIOControl[829]: OK
Mar 14 20:18:47 phoniebox PhonieboxGPIOControl[829]: INFO:GPIODevices.simple_button:VolumeControlBtn2: holdAndRepeatHandler
Mar 14 20:18:47 phoniebox PhonieboxGPIOControl[829]: INFO:GPIODevices.simple_button:VolumeControlBtn2: action
Mar 14 20:18:48 phoniebox PhonieboxGPIOControl[829]: INFO:GPIODevices.simple_button:VolumeControlBtn2: holdAndRepeatHandler
Mar 14 20:18:48 phoniebox PhonieboxGPIOControl[829]: INFO:GPIODevices.simple_button:VolumeControlBtn2: action
Mar 14 20:18:49 phoniebox PhonieboxGPIOControl[829]: INFO:GPIODevices.simple_button:VolumeControlBtn2: holdAndRepeatHandler
Mar 14 20:18:49 phoniebox PhonieboxGPIOControl[829]: INFO:GPIODevices.simple_button:VolumeControlBtn2: action
Mar 14 20:18:50 phoniebox PhonieboxGPIOControl[829]: INFO:GPIODevices.simple_button:VolumeControlBtn2: holdAndRepeatHandler
Mar 14 20:18:50 phoniebox PhonieboxGPIOControl[829]: INFO:GPIODevices.simple_button:VolumeControlBtn2: action
Mar 14 20:18:50 phoniebox PhonieboxGPIOControl[829]: INFO:GPIODevices.simple_button:VolumeControlBtn2: holdAndRepeatHandler
Mar 14 20:18:50 phoniebox PhonieboxGPIOControl[829]: INFO:GPIODevices.simple_button:VolumeControlBtn2: action
Mar 14 20:18:51 phoniebox PhonieboxGPIOControl[829]: INFO:GPIODevices.simple_button:VolumeControlBtn2: holdAndRepeatHandler
Mar 14 20:18:51 phoniebox PhonieboxGPIOControl[829]: INFO:GPIODevices.simple_button:VolumeControlBtn2: action
Mar 14 20:18:51 phoniebox PhonieboxGPIOControl[829]: INFO:GPIODevices.two_button_control:functionCall1
Mar 14 20:18:51 phoniebox PhonieboxGPIOControl[829]: OK MPD 0.21.4
Mar 14 20:18:51 phoniebox PhonieboxGPIOControl[829]: OK
Mar 14 20:18:52 phoniebox PhonieboxGPIOControl[829]: INFO:GPIODevices.simple_button:VolumeControlBtn2: holdAndRepeatHandler
Mar 14 20:18:52 phoniebox PhonieboxGPIOControl[829]: INFO:GPIODevices.simple_button:VolumeControlBtn2: action
Mar 14 20:19:15 phoniebox PhonieboxGPIOControl[829]: INFO:GPIODevices.simple_button:VolumeControlBtn2: holdAndRepeatHandler
Mar 14 20:19:15 phoniebox PhonieboxGPIOControl[829]: INFO:GPIODevices.simple_button:VolumeControlBtn2: action
Mar 14 20:19:39 phoniebox PhonieboxGPIOControl[829]: INFO:GPIODevices.simple_button:VolumeControlBtn2: holdAndRepeatHandler
Mar 14 20:19:39 phoniebox PhonieboxGPIOControl[829]: INFO:GPIODevices.simple_button:VolumeControlBtn2: action
Mar 14 20:19:39 phoniebox PhonieboxGPIOControl[829]: INFO:GPIODevices.two_button_control:functionCall2
Mar 14 20:19:39 phoniebox PhonieboxGPIOControl[829]: OK MPD 0.21.4
Mar 14 20:19:39 phoniebox PhonieboxGPIOControl[829]: OK
Mar 14 20:19:49 phoniebox PhonieboxGPIOControl[829]: INFO:GPIODevices.simple_button:VolumeControlBtn2: holdAndRepeatHandler
Mar 14 20:19:49 phoniebox PhonieboxGPIOControl[829]: INFO:GPIODevices.simple_button:VolumeControlBtn2: action
Mar 14 20:19:49 phoniebox PhonieboxGPIOControl[829]: INFO:GPIODevices.two_button_control:functionCall2
Mar 14 20:19:49 phoniebox PhonieboxGPIOControl[829]: OK MPD 0.21.4
Mar 14 20:19:49 phoniebox PhonieboxGPIOControl[829]: OK
Mar 14 20:19:54 phoniebox PhonieboxGPIOControl[829]: INFO:GPIODevices.simple_button:VolumeControlBtn2: holdAndRepeatHandler
Mar 14 20:19:54 phoniebox PhonieboxGPIOControl[829]: INFO:GPIODevices.simple_button:VolumeControlBtn2: action
Mar 14 20:20:01 phoniebox PhonieboxGPIOControl[829]: INFO:GPIODevices.simple_button:VolumeControlBtn2: holdAndRepeatHandler
Mar 14 20:20:01 phoniebox PhonieboxGPIOControl[829]: INFO:GPIODevices.simple_button:VolumeControlBtn2: action
Mar 14 20:20:01 phoniebox PhonieboxGPIOControl[829]: INFO:GPIODevices.two_button_control:functionCall2
Mar 14 20:20:01 phoniebox PhonieboxGPIOControl[829]: OK MPD 0.21.4
Mar 14 20:20:01 phoniebox PhonieboxGPIOControl[829]: OK
`
You clearly see that the putton presses are recognized, but the functionCall of the two_button_control is only executed sometines.

As the originator of this issue seems to be absent, I would like to pick this up and help you in resolving it.
I will try the mentioned commit and let you know.

that-coding-human, 2021-03-15:
My setup is:
- Raspberry Pi3b+
- Hifiberry miniamp
- Neuftech RFID Scanner
- Pimoroni OnOff-Shim 
- Arcade Buttons for Vol Up, Vol Down, Prev, Next, Play/Pause

I just did some testing with the following steps:
- I switched to #1165 and had a look at the changes performed.
- I took the simple_button.py and the two_button_control.py from the actual develop branch and replaced the files on my pi with them.
- Performed a reboot
- Started playing a track and then tested all buttons by pressing them in fast order and randomly with longer presses and longer pauses between the presses.

I can confirm that the behaviour of missed function calls did not appear anymore. I was experiencing it regularly and a lot before.
The mentioned fix #1165 seems to work for this.

The only thing I observed, that might be worth a second look, is the following:
- When I pressed the vol up or vol down button multiple times, it looked like exactly one call gets lost. 
e.g. when I started with 1% volume, and have configured 5% vol change steps, I pressed the button 5 times. By incresing 5x5% I should have ended up with 26%, right? But it was 21%. The same when lowering the volume again.
Repeated presses of these buttons seem to result in exactly 1 function call less than button presses.

@s-martin I think the fix for the original behaviour works. Is there any reason to not include it in the 2.3 milestone again? 
Can you look into the behaviour with the exactly 1 lost call when pressing the vol function repeatedly?

s-martin, 2021-03-17:
Fixed with #1165

closed 2021-03-17 by s-martin