MiczFlor/RPi-Jukebox-RFID #1140: 🚀 | Switch from youtube-dl to youtube-dlc

joeyvanderbie, 2020-11-14:
## Feature Description
Better support for youtube links and playlists from web interface.
Youtube-dl is not always working as expected, and does not fully support playlists.

### What functionality would you like to see in your phoniebox?
It would be great if we can not only support individual youtube links, but also the playlists from the web interface.
Limitations currently are connected to youtube-dl.
Youtube-dlc appears to be faster with updates and supports playlists better (in my experience)
The repository I use can be found at https://github.com/blackjack4494/yt-dlc
I suggest we switch to youtube-dlc and implement support for youtube playlists in the interface.

### How do you envision the feature to work from a users perspective?

Implementation in the web interface could for now be just an extra form field for playlists next to the current youtube form field

### Further information that might help

impact on files
at least: nc.formCardEdit.php needs to have a form field for the playlist
and inc.processCheckCardEditRegister.php, change line 265 
from             $exec = "cd '".$foldername."'; youtube-dl -f bestaudio --extract-audio --audio-format mp3 ".$post['YTstreamURL']." > ".$conf['shared_abs']."/youtube-dl.log; chmod 777 ".$foldername."/* 2>&1 &";
to             $exec = "cd '".$foldername."'; youtube-dlc -f bestaudio --extract-audio --audio-format mp3 ".$post['YTstreamURL']." > ".$conf['shared_abs']."/youtube-dl.log; chmod 777 ".$foldername."/* 2>&1 &";

and add around there somewhere the playlist check with a line like this: 
            $exec = "cd '".$foldername."'; youtube-dlc -f bestaudio --extract-audio --audio-format mp3 --playlist-random --yes-playlist ".$post['YTstreamURL']." > ".$conf['shared_abs']."/youtube-dl.log; chmod 777 ".$foldername."/* 2>&1 &";

s-martin, 2020-11-21:
I think, if yt-dlc is actively maintained, we could switch to have the playlists feature.

A PR is always welcome :)

s-martin, 2022-05-29:
Closing. Please feel free to reopen this issue.

closed 2022-05-29 by s-martin