MiczFlor/RPi-Jukebox-RFID #1142: Phoniebox with EQUALIZER not working

ralle211, 2020-11-16:
My Phoniebox got just completed but the sound is not as clear.
I am using a Hifiberry miniamp an i startet searching and found a equalizer for Raspberry with sounds very easy to use.

https://www.hifiberry.com/docs/software/guide-adding-equalization-using-alsaeq/

I tried now a lot of configs for the asounf.conf but nothing worked. I have no sound or 100% loudness without the equalizer.
I think that i just have to set the "Master" name at the right position or change "Master" to "equal" (when using the example).

Can somebody help me out here?

Many THANKS!!!!!!!

s-martin, 2020-11-21:
What’s the output of 

`sudo cat /etc/asound.conf`

`cat /proc/asound/modules` 

`sudo cat /usr/share/alsa/alsa.conf`

ralle211, 2020-11-27:
**sudo cat /etc/asound.conf**

`pcm.!default {
type plug
slave.pcm "sftvol"
}
pcm.sftvol {
type softvol
slave.pcm "plughw:0"
control {
name "Master"
card 0
}
}
ctl.equal {
  type equal;
  controls "/home/pi/.alsaequal.bin"
}
pcm.plugequal {
  type equal;
  slave.pcm "plughw:0";
  controls "/home/pi/.alsaequal.bin"
}
  pcm.equal {
  type plug;
  slave.pcm plugequal;
}`

**cat /proc/asound/modules**
`0 snd_soc_rpi_simple_soundcard`

**sudo cat /usr/share/alsa/alsa.conf**
`@hooks [
	{
		func load
		files [
			"/etc/alsa/conf.d"
			"/etc/asound.conf"
			"~/.asoundrc"
		]
		errors false
	}
]

cards.@hooks [
	{
		func load
		files [
			{
				@func concat
				strings [
					{ @func datadir }
					"/cards/aliases.conf"
				]
			}
		]
	}
	{
		func load_for_all_cards
		files [
			{
				@func concat
				strings [
					{ @func datadir }
					"/cards/"
					{ @func private_string }
					".conf"
				]
			}
		]
[... 530 lines cut ...]
		default {
			@func refer
			name defaults.timer.device
		}
	}
	@args.SUBDEV {
		type integer
		default {
			@func refer
			name defaults.timer.subdevice
		}
	}
	type hw
	class $CLASS
	sclass $SCLASS
	card $CARD
	device $DEV
	subdevice $SUBDEV
	hint {
		description "Direct timer device"
		device $DEV
	}
}`

Many THANKS!!!

ralle211, 2020-11-27:
I also tried:
sudo cat /etc/asound.conf
pcm.!default {
type plug
slave.pcm "**equal**"
}

AND

sudo cat /etc/asound.conf
pcm.!default {
type plug
slave.pcm "**plugeqal**"
}

but this didnt work

s-martin, 2023-12-05:
Closing as no interaction for a couple of months.
Feel free to reopen the issue, if the problem still occurs.

closed 2023-12-05 by s-martin