MiczFlor/RPi-Jukebox-RFID #1145: 🐛 | BUG SUMMARY

ElGauchooooo, 2020-11-19:
## Bug

### What I did

Installed phoniebox using the provided installer script on a raspberry pi, added one folder including several files and assigned an RFID card to it, set the folder to _"RESUME: ON", "SHUFFLE: ON", "SINGLE: OFF"_

### What happened

Approximately 90% of the time, phoniebox plays instantly without any issues. Yet in the other 10% of the cases, phoniebox does not start playing when a card is being supplied. Tried starting it via the webapp using the play button on the playlist:
[image]
The button then _sometimes_ turns red, sometimes it already works on this button click. Last option to start the playlist is the big playbutton at the top - using this one, the phoniebox starts playing again: 
[image]

### I expected this to happen
Phoniebox should start playing instantly using the RFID card.

### Further information that might help
find logfiles at https://paste.ubuntu.com/p/W9cftsQFXQ/

Not quite sure if the logs covered one of the incidents, but should. I've enabled all logs on phoniebox webinterface, if you only need specific ones, please let me know, so that i can set them up prior to the next bug appearance. Bug appears approximately every 2 days.

## Software

### Base image and version
`pi@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
`

### Branch / Release
`pi@raspberrypi:~ $ cd /home/pi/RPi-Jukebox-RFID/ && git status | head -2
On branch master
Your branch is up to date with 'origin/master'.`

### Installscript

`scripts/installscripts/buster-install-default.sh`
## Hardware

### RaspberryPi version

pi@raspberrypi:~/RPi-Jukebox-RFID $ sudo cat /sys/firmware/devicetree/base/model
Raspberry Pi 3 Model B Rev 1.2

### RFID Reader

RC522

### Soundcard

RPI OnBoard.

s-martin, 2020-11-21:
Do you use classic or Spotify edition?

Nothing special in the log. Unfortunately this looks like just sometimes the swipe isn’t recognized. I suppose you haven’t noticed anything before the swipe doesn’t work?

s-martin, 2020-11-21:
Could you check, if #1083 is related?

ElGauchooooo, 2020-11-21:
@s-martin how to check if the bug is related? As far as i can see, there's no solution to that issue mentioned, nor a way to reproduce it. It might be, but cannot tell for sure (and i'm not using a USB RFID reader - mine is connected via GPIO).

I'm using classic edition - but had the same issue with spotify edition (which i had installed previously, then removed as i thought somehow spotify might interfere when playing). 

The only thing i've noticed is, that sometimes the phoniebox isn't reachable with wifi (connection timed out on webinterface). Might it be related to this one?

s-martin, 2020-11-22:
Is the behavior described in #1083 similar to the behavior you are experiencing?

ElGauchooooo, 2020-11-22:
Yes, looks completely identical, might be the same. My reader however has no beeper nor does it blink never (working or not), yet (if its the same issue) it appears not to be related to the reader.

s-martin, 2020-11-25:
We released 2.2 a couple of days ago with fixes for both resume and shuffle, so you might want to check, if this fixes also your issue.

ElGauchooooo, 2020-12-01:
Updated to 2.2, havn't used it for 1-2 days. Today, phoniebox didn't work on the first try. Starting it via webinterface works like a charm. debug.log can be found here, have removed the first log entries right before 12:00 when it switches to 15:xx o'clock (where the issue happened) https://paste.ubuntu.com/p/3y2R6pJNvQ/

ElGauchooooo, 2020-12-01:
Next error occurred - after one song was finished, the phoniebox did not continue playing with the next song, although repeat is on and shuffle aswell (see first image in this issue). Any clues whats going on here?

s-martin, 2022-12-27:
Can you check, if the issue still persists with current 2.4 version?

s-martin, 2024-05-17:
No response, so I’m closing the issue.
Feel free to reopen the issue, if you are still experiencing issues with a recent V2.

closed 2024-05-17 by s-martin