MiczFlor/RPi-Jukebox-RFID #1146: Record when button is push and stop when released

wackoracoon, 2020-11-19:
Hi,
I want to set a button to record voice only when it's pressed. I expect a call to functionCallRecordStop when released.
A saw here in issue topics about that here : #437 I think this belong to the old (?) file RPi-Jukebox-RFID/scripts/gpio-buttons.py and changes for GPIO was made for release 2.1.

Old file contains :
```
reco.when_pressed = recordstart_action
reco.when_released = recordstop_action
```

Is there is a way to do that with latest release ?
Thanks !

Bonus behavior : when push button is released, it stops the record AND play the latest record

s-martin, 2020-11-21:
This should work, if you use one of the function calls for recording defined in https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/components/gpio_control/function_calls.py in your `gpio_settings.ini`.

E.g.

```
[Record]
enable: True
Type: Button
Pin: 27
pull_up: True
functionCall: functionCallRecordStart
```

This is untested and I’m not sure how to set it up to stop recording when releasing the button.

wackoracoon, 2020-11-21:
Yes, this setting is what I understood to do for a record button if there is an other button to stop the record.
I made some tests and I my idea was to create a button defined in the .ini with 
`edge=both`
and create a new functionCall : functionCallRecordStartAndStop that can get args like the state of (GPIO.input(button)).
0 is pushed so start record and when it's 1 (released) stop record.
Easy to say but not easy to do (for me...)

madmooose, 2020-12-05:
I would really appreciate something like that. I stumpled upon the same problem when trying to add a switch for turning playback on and off.

jorgemia, 2021-02-27:
@madmooose @wackoracoon were you able to get something working? I want to set up a button that will play the music whilst the button is pressed and stops the music when the button is released

jorgemia, 2021-02-27:
It seems like maybe we can use the functions playerpauseforce and playerplay instead of playerpause (which should be called playertoggle) from here:

https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/dfeed95eac288489409267ee7ea4cdf374ab0936/scripts/playout_controls.sh#L683-L739

I will try and see if I can get it to work