MiczFlor/RPi-Jukebox-RFID #1150: Audio works on the Pi , but not via Phoniebox 🐛

mrc-sudo, 2020-11-21:
Hi there,

I really like the idea of this project and I'm very keen on realizing it. But unfortunately I'm experiencing an audio issue.

I'm running Phoniebox on a Pi4 with the newest version of Buster. So far I'm using the internal sound with the AV Jack. If I'm playing the sounds via the Buster Desktop there is no problem (eg. .mp3 / youtube). But if I'm accessing the phoniebox (via a different device) the Audio doesn't work. I get a pop up saying xyz is playing but there is no sound.

I guess important to mention-I get the start up sound but not the shut down sound.

Thanks for any advice or help!

s-martin, 2020-11-22:
Could you please post the output of `tail -n 500 /var/log/syslog`?

Did you check the steps in https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#audio-is-not-working ?

mrc-sudo, 2020-11-23:
Hi, the output of `tail -n 500 /var/log/syslog` is the following (I tried one local file and one song via Spotify):

`:~ $ tail -n 500 /var/log/syslog
Nov 23 10:24:03 raspberrypi bash[426]: AUDIOVOLMINLIMIT="1"
Nov 23 10:24:03 raspberrypi bash[426]: AUDIOVOLSTARTUP="50"
Nov 23 10:24:03 raspberrypi bash[426]: VOLCHANGEIDLE="TRUE"
Nov 23 10:24:03 raspberrypi bash[426]: IDLETIMESHUTDOWN="0"
Nov 23 10:24:03 raspberrypi bash[426]: SHOWCOVER="ON"
Nov 23 10:24:03 raspberrypi bash[426]: READWLANIPYN="OFF"
Nov 23 10:24:03 raspberrypi bash[426]: EDITION="plusSpotify"
Nov 23 10:24:03 raspberrypi bash[426]: LANG="en-UK"
Nov 23 10:24:03 raspberrypi bash[426]: VERSION="2.1.1 - 70f989b - master"
Nov 23 10:24:03 raspberrypi bash[426]: CMDVOLUP=""
Nov 23 10:24:03 raspberrypi bash[426]: CMDVOLDOWN=""
Nov 23 10:24:03 raspberrypi bash[426]: CMDNEXT=""
Nov 23 10:24:03 raspberrypi bash[426]: CMDPREV=""
Nov 23 10:24:03 raspberrypi bash[426]: CMDREWIND=""
Nov 23 10:24:03 raspberrypi bash[426]: CMDSEEKFORW=""
Nov 23 10:24:03 raspberrypi bash[426]: CMDSEEKBACK=""
Nov 23 10:24:03 raspberrypi bash[426]: 50
Nov 23 10:24:03 raspberrypi dhcpcd[423]: dev: loaded udev
Nov 23 10:24:03 raspberrypi rsyslogd: imuxsock: Acquired UNIX socket '/run/systemd/journal/syslog' (fd 3) from systemd.  [v8.1901.0]
Nov 23 10:24:03 raspberrypi rsyslogd:  [origin software="rsyslogd" swVersion="8.1901.0" x-pid="422" x-info="https://www.rsyslog.com"] start
Nov 23 10:24:03 raspberrypi dphys-swapfile[409]: want /var/swap=100MByte, checking existing: keeping it
Nov 23 10:24:03 raspberrypi udisksd[421]: udisks daemon version 2.8.1 starting
Nov 23 10:24:03 raspberrypi kernel: [    8.690347] 8021q: 802.1Q VLAN Support v1.8
Nov 23 10:24:03 raspberrypi dhcpcd[423]: forked to background, child pid 490
Nov 23 10:24:03 raspberrypi systemd[1]: Started System Logging Service.
Nov 23 10:24:03 raspberrypi systemd[1]: Started dhcpcd on all interfaces.
Nov 23 10:24:03 raspberrypi systemd[1]: Started Login Service.
Nov 23 10:24:03 raspberrypi systemd[1]: Started WPA supplicant.
Nov 23 10:24:03 raspberrypi wpa_supplicant[418]: Successfully initialized wpa_supplicant
Nov 23 10:24:03 raspberrypi avahi-daemon[408]: Successfully called chroot().
Nov 23 10:24:03 raspberrypi systemd[1]: Started Avahi mDNS/DNS-SD Stack.
Nov 23 10:24:03 raspberrypi avahi-daemon[408]: Successfully dropped remaining capabilities.
Nov 23 10:24:03 raspberrypi avahi-daemon[408]: No service file found in /etc/avahi/services.
Nov 23 10:24:03 raspberrypi systemd[1]: Reached target Network.
Nov 23 10:24:03 raspberrypi avahi-daemon[408]: Network interface enumeration completed.
Nov 23 10:24:03 raspberrypi avahi-daemon[408]: Server startup complete. Host name is raspberrypi.local. Local service cookie is 3954776602.
Nov 23 10:24:03 raspberrypi systemd[1]: Started Phoniebox Idle Watchdog Service.
Nov 23 10:24:03 raspberrypi systemd[1]: Started Phoniebox RFID-Reader Service.
Nov 23 10:24:03 raspberrypi systemd[1]: Starting Permit User Sessions...
Nov 23 10:24:03 raspberrypi systemd[1]: Condition check resulted in fast remote file copy program daemon being skipped.
Nov 23 10:24:03 raspberrypi systemd[1]: Reached target Network is Online.
Nov 23 10:24:03 raspberrypi systemd[1]: Starting Samba NMB Daemon...
Nov 23 10:24:03 raspberrypi udisksd[421]: failed to load module crypto: libbd_crypto.so.2: cannot open shared object file: No such file or directory
Nov 23 10:24:03 raspberrypi udisksd[421]: failed to load module mdraid: libbd_mdraid.so.2: cannot open shared object file: No such file or directory
Nov 23 10:24:03 raspberrypi systemd[1]: Starting LSB: exim Mail Transport Agent...
Nov 23 10:24:03 raspberrypi kernel: [    9.014517] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
Nov 23 10:24:03 raspberrypi systemd[1]: Starting Lighttpd Daemon...
Nov 23 10:24:03 raspberrypi systemd[1]: Starting Mopidy music server...
Nov 23 10:24:03 raspberrypi systemd[1]: Starting /etc/rc.local Compatibility...
Nov 23 10:24:03 raspberrypi systemd[1]: Started dphys-swapfile - set up, mount/unmount, and delete a swap file.
Nov 23 10:24:03 raspberrypi systemd[1]: Started Permit User Sessions.
Nov 23 10:24:03 raspberrypi systemd[1]: Started Mopidy music server.
Nov 23 10:24:03 raspberrypi systemd[1]: Started /etc/rc.local Compatibility.
Nov 23 10:24:03 raspberrypi systemd[1]: Starting Hold until boot process finishes up...
Nov 23 10:24:03 raspberrypi systemd[1]: Starting Light Display Manager...
Nov 23 10:24:03 raspberrypi udisksd[421]: Failed to load the 'mdraid' libblockdev plugin
Nov 23 10:24:03 raspberrypi udisksd[421]: Failed to load the 'crypto' libblockdev plugin
Nov 23 10:24:03 raspberrypi dbus-daemon[399]: [system] Activating via systemd: service name='org.freedesktop.PolicyKit1' unit='polkit.service' requested by ':1.5' (uid=0 pid=421 comm="/usr/lib/udisks2/udisksd ")
Nov 23 10:24:03 raspberrypi systemd[1]: Starting Authorization Manager...
Nov 23 10:24:04 raspberrypi dhcpcd-run-hooks[547]: wlan0: starting wpa_supplicant
Nov 23 10:24:04 raspberrypi raspi-config[392]: Checking if shift key is held down: No. Switching to ondemand scaling governor.
Nov 23 10:24:04 raspberrypi systemd[1]: Started LSB: Switch to ondemand cpu governor (unless shift key is pressed).
Nov 23 10:24:04 raspberrypi kernel: [    9.769931] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
Nov 23 10:24:04 raspberrypi dhcpcd[490]: wlan0: connected to Access Point `'
[... cut ...]
Nov 23 10:27:44 raspberrypi mopidy[528]: INFO     [MpdSession-85] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:59494
Nov 23 10:27:44 raspberrypi mopidy[528]: INFO     [MpdSession-86] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:59498
`

Regarding the troubleshooting. I've thought this is for external sound cards only. But I checked the audio iFace File and there is 'Headphone" saved.

The next step is external card only, right? 

`cat /proc/asound/modules
 0 snd_bcm2835
 1 snd_bcm2835
`

0x0lli, 2020-11-23:
I am wondering: did you have a mp3 on the  card and set the path within the installer/configuration right ? 
The appearance of your dashboard looks pretty naked to me. If there is no link to a file, the box doesnt have anything to playback. Sorry if that is off

mrc-sudo, 2020-11-23:
> I am wondering: did you have a mp3 on the card and set the path within the installer/configuration right ?
> The appearance of your dashboard looks pretty naked to me. If there is no link to a file, the box doesnt have anything to playback. Sorry if that is off

Hi, thanks for your input. I have mp3s in the standard folder and there are also shwon in my dashbord. I just wanted to make another screenshot for you, so I ramped up the Pi but didn't connect my screen. And guess what, it works :D I double checked it, connected my screen and rebooted the Pi-didn't work.

So the problem in this case was the connected monitor. @s-martin Is this normal behaviour and a attached screen is not foreseen?

Anyway, thanks a lot guys!

s-martin, 2020-11-23:
Attached screen was probably never tested, because not the main use case.

mrc-sudo, 2020-11-23:
> Attached screen was probably never tested, because not the main use case.

True. In my case it was just still attached from installation. Will use it without screen as well ;)

s-martin, 2020-11-23:
So if this occurs only with an attached screen you could close the issue.

mrc-sudo, 2020-11-24:
> So if this occurs only with an attached screen you could close the issue.

You're right. Since it works smoothly I close this one.

closed 2020-11-24 by mrc-sudo