MiczFlor/RPi-Jukebox-RFID #1153: 🐛 | Mopidy dies after receiving wrong Spotify links

bubble1999, 2020-11-21:
## Bug

### What I did

I installed the raspberry pi with above mentioned buster image and ran the installer script including spotify

### What happened

Startup sound is coming, but no sounds from spotify streams (RFID swipe and web interface)

### I expected this to happen

Play the music I selected

### Further information that might help

    I had a lot of trouble getting the spotify streams running. Same issue as decribed here [https://github.com/natumbri/mopidy-youtube/issues/58](link) or here [https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/1067](link)
    I got it running by going through this procedure: [https://forum-raspberrypi.de/forum/thread/48179-phoniebox-spotify-ich-bekomme-es-nicht-zum-laufen/](link)
    I think the main difference from what I did is that I installed the raspian packages for the desktop environment (because I thought that I will not need them on the headles raspberry) However, it seems that there are some drivers hidden which are needed to get Spotify streams running.

EDIT: perhaps it turns out that I somehow got mopidy in troubles.
After going through the mentioned procedure I put in a spotify link (with add new card):
https://open.spotify.com/album/0Lya0qWV4qfHVArhaXYSju
Actually, I still do not know which format a Spotify link must have; it would be most convenient to just copy from spotify web page. However, using this link leads to such error message:

`Nov 21 20:18:43 Phoniebox mopidy[947]: WARNING  [MainThread] mopidy.audio.actor Could not find a text/html decoder to handle media.
Nov 21 20:18:43 Phoniebox mopidy[947]: WARNING  [MainThread] mopidy.audio.gst GStreamer warning: No decoder available for type 'text/html'.
Nov 21 20:18:43 Phoniebox mopidy[947]: ERROR    [MainThread] mopidy.audio.gst GStreamer error: Your GStreamer installation is missing a plug-in.
`

After this happened, all other links to spotify just did not play and produced this error message. Only after reboot, a spotify link such as
spotify:album:0Lya0qWV4qfHVArhaXYSju
did play the musik.

Perhaps it would be possible to convert http links or fetch them from executing.

This is my first bug report so please tell me if I can provide more information.

## Software

### Base image and version

### Branch / Release

### Installscript

## Hardware

### RaspberryPi version

### RFID Reader

### Soundcard

### Other notable hardware

s-martin, 2020-11-22:
So to clarify:
If you use a wrong link for Spotify, mopidy dies?
In the meantime you know which formats the links have to be and Phoniebox/mopidy works?

Did I understand that correctly ?

bubble1999, 2020-11-23:
I noticed that the Spotify link format should be
spotify:album:XXXX

This works until I use a http-link.
Without reboot, also a spotify:album:XXX link then does not work.

After reboot, spotify:album:XXX will work again.

s-martin, 2020-11-23:
I don't know, if there was an intention to support the http links or if this is possible (I don't use Spotify myself)

s-martin, 2020-11-29:
From my point of view someone who is using Spotify edition or has in depth experience could investigate, if using the URLs directly could be implemented.

Until know this use case wasn't considered.

keesanalytics, 2020-12-01:
I've experienced the same issue. I think what happens is:
You set up an rfid card with a wrong spotify link such as https://open.spotify/playlist/<playlistcode>. This links to the webpage for the playlist, not the actual playlist itself. So mopidy gets an html file, which it doesn't know how to handle and it crashes/disconnects. You can see that it is disconnected in the info screen. 
To play music again you have to restart the mopidy service. For example, by rescanning the library (this stops and restarts the mopidy service). 

However, the bad link is still there. So whenever you swipe that same card, it will crash mopidy again. The solution I found is to delete the card, delete the playlist from the Phoniebox and delete the folder for the playlist from the Phoniebox. Then register the card again with the correct spotify link in the form of spotify:playlist:XXXX. This got the playlist working.

Hope it helps!

bubble1999, 2020-12-15:
This is great, thank you very much!

closed 2021-01-07 by s-martin