MiczFlor/RPi-Jukebox-RFID #1156: 🚀 | GPIO Buttons over USB Encoder

Dieter4712, 2020-11-23:
## Feature Description
Support the GPIO-button functionality that are connected via USB Encoder ( https://www.amazon.de/gp/product/B01N0GZQZI/ )

### What functionality would you like to see in your phoniebox?

I would like to map the USB Encoder Buttons in the install script or in the folder gpio_control\example_configs.

### How do you envision the feature to work from a users perspective?

I am afraid of soldering. Two years ago I build my Phoniebox and I bought this USB Encoder. Everytime I upgrade the Phoniebox I override the GPIO functions (I wrote my own py-Function to control the buttons, the USB Encoder reacts like an new input device). On every update I am struggeling with the GPIO-Service and the new functions that are available with the GPIO Controls are awesome.

### Further information that might help

I can send my gpio_control.py script (its not that good but it works), that I overwrite in /etc/systemd/system/phoniebox-gpio-control.service.

Thank you for your awesome work!

s-martin, 2020-11-23:
Please post your GPIO file here as starting point

MiczFlor, 2020-11-23:
Hi @Dieter4712 
that's awesome :) I also hate soldering :) I would love to try this and am looking forward to the GPIO file. Attach it here or open a pull request.
All the best, micz

Dieter4712, 2020-11-23:
Thank you for your fast feedback.

I attached the file (i was not allowed to add a *.py file, so i renamed it to *.txt. Unfortunately it is not generic code. The variables are not stored in constants and so on, sorry for that bad coding :-/

I can support you if there are any questions.

[gpio-buttons.txt](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/5585486/gpio-buttons.txt)

s-martin, 2020-11-24:
I think the implementation is easy enough, so I suggest we add this as an additional component. I would prepare the infrastructure and then you could help implementing or testing.

Is there a description somewhere how this usb encoder needs to be installed or used?

Dieter4712, 2020-11-24:
The USB encoder only needs to be pluged in. You dont need to install any drivers. 
After pluging in, it acts like an input device. 
The script has to be variable because the input device number depends on the devices that are attached. 
The other variable stuff is how you map the GPIO on the keyboard. There are many possibilities to map the buttons.

I attached the documentation. I can help you testing or answer questions! 
[image]

MiczFlor, 2020-11-27:
Hi @Dieter4712 
I ordered the hardware, now I only need the time to follow your lead.
[image]

s-martin, 2020-11-30:
I started a branch for a possible addition of USB Encoder to Phoniebox: https://github.com/s-martin/RPi-Jukebox-RFID/tree/add-usb-encoder

I don't have the hardware, so please test it and adapt it as necessary.

Some hints:
* the input device has to be configured in the py file
* I don't know, if all the relative paths work, there could be some issues...

Dieter4712, 2020-12-02:
Hi @s-martin ,

very nice to see, that the changes are allready implemented.

I want to test your changes and I dont know what i made wrong. I want to make a clean install, so i get the https://raw.githubusercontent.com/s-martin/RPi-Jukebox-RFID/add-usb-encoder/scripts/installscripts/buster-install-default.sh file from your branch. I ran the one-line-installation script at your branch, but after that I cant find a folder RPi-Jukebox-RFID/components/control/buttons-USB-Encoder

Its late, but I dont know what i´ve been doing wrong.

s-martin, 2020-12-02:
The install script uses the branch name and the Git URL in several places, so you need to use the following command, if you want to use my branch:

```cd; rm buster-install-*; wget https://raw.githubusercontent.com/s-martin/RPi-Jukebox-RFID/add-usb-encoder/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; GIT_URL=https://github.com/s-martin/RPi-Jukebox-RFID GIT_BRANCH=add-usb-encoder bash ./buster-install-default.sh```

Dieter4712, 2020-12-03:
Thank you, that worked for me.

I found some errors:
- There is a typo in line 25 (in the end, the destination file should be '/etc/systemd/system/phoniebox-usb-encoder.service' and not '/etc/systemd/system/phoniebox-usb-eoncoder.service' :-)
- the script has to be executeable, so i executed `chmod +x ./setup_USB_Encoder.sh`

After that, the service can not be startet, because there is another problem with permission.
`phoniebox-usb-encoder.service: Failed at step EXEC spawning /home/pi/RPi-Jukebox-RFID/components/control/buttons-USB-Encoder/buttons-USB-Encoder.py: Permission denied
raspberrypi systemd[1]: phoniebox-usb-encoder.service: Main process exited, code=exited, status=203/EXEC
`
I tried the folllwing two commands, but without success:
`sudo chmod +x  /home/pi/RPi-Jukebox-RFID/components/control/buttons-USB-Encoder/buttons-USB-Encoder.py

sudo chmod 644  /home/pi/RPi-Jukebox-RFID/components/control/buttons-USB-Encoder/buttons-USB-Encoder.py`

Dieter4712, 2020-12-04:
Another problem:

After I replaced the content of the file buttons-USB-Encoder.py with my old content, the buttons are working (in the old, terrible way), because I would like to test the encoder-service.
Unfortunatly, it doesnt run after startup, it has to be started manually.

Kind regards.

s-martin, 2020-12-05:
Thanks for testing, I try to look into it this weekend.

Dieter4712, 2020-12-05:
If you need further help, let me know.
I tried to look after the import python function issue, but for me it looks all good. The folder RPi-Jukebox-RFID/components/gpio_control/ contains a __init__.py file which is needed. I dont have any idea.

jeripeierSBB, 2020-12-30:
A few days ago I bought a similar product as you did: https://www.amazon.de/gp/product/B075DFNK24. 

I tested @s-martin  branch and fixed some issues which were reported by @Dieter4712. Furthermore I made the scripts more flexible. As a last step, I will introduce a script to make the button mapping in the next days. Work in progress: https://github.com/MiczFlor/RPi-Jukebox-RFID/compare/develop...jeripeierSBB:usb-encoder

jeripeierSBB, 2020-12-31:
Finally I found time to finish the work of @s-martin by creating a PR. With the device https://www.amazon.de/gp/product/B075DFNK24 it works very well. Maybe @Dieter4712 and @MiczFlor can test their devices too?

Additionally I ask you if it would be a good idea to move the existing `gpio_control` folder also inside the new `controls` folder?

s-martin, 2021-01-01:
Implemented by #1249

closed 2021-01-01 by s-martin

Dieter4712, 2021-01-11:
@jeripeierSBB  Great work! 
I will test it in a few days. Bescause of corona the box is mich more in use :-)