MiczFlor/RPi-Jukebox-RFID #1157: GPIO Buttons aren't working anymore

lerniberni, 2020-11-23:
## Bug
Vol Up / Down & Next / Prev Track Buttons do not work anymore

### What I did
Since a few weeks our phoniebox took a long time to boot so I installed the raspberry pi with above mentioned buster image and ran the installer script.
Still had issues with spotify so I took a step back and installed only the classic version.

### What happened

I can't get my buttons to work again. They have been working fine ever since I builded the box. I only had 4 buttons installed and tried to assign them to other functions. Shutdown worked for some reason. 
What's also very strange is, that there's no music coming from the box when hdmi is connected. Still the intro-jingle is coming. Vise versa there's no intro-jingle when no hdmi is connected but music is coming. 

As fas as I remember I wired the buttons according to the Classic Layout:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Hardware-Pinout-Overview#flexible-pinout

Any ideas?

s-martin, 2020-11-23:
Since 2.0 we have a new GPIO control component which improves GPIO usage, see https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/develop/components/gpio_control
Unfortunately there's currently a bug in the path, but 2.2 release should be out of the door soon, so you could either wait or install develop branch.

For the audio issue, I suspect this is the change in the Raspberry Pi software introduced in May 2020, please see here: https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#raspberry-pi-update-may-2020-broke-my-audio

Groovylein, 2020-11-26:
Nice, I will try the new release and will report if the GPIOs now work.

Hamsert, 2020-11-28:
I just freshly installed a new Phoniebox with version 2.2 and can confirm that unfortunately GPIO-buttons still won't work. Can I provide any logs or outputs for troubleshooting?

s-martin, 2020-11-28:
Please provide outputs of `tail -n 500 /var/log/syslog`.

Hamsert, 2020-11-28:
pi@phoniebox:~ $ tail -n 500 /var/log/syslog
Nov 28 11:30:55 phoniebox kernel: [    2.486939] [vc_sm_connected_init]: end - returning 0
Nov 28 11:30:55 phoniebox kernel: [    2.491342] Initializing XFRM netlink socket
Nov 28 11:30:55 phoniebox kernel: [    2.493730] NET: Registered protocol family 17
Nov 28 11:30:55 phoniebox kernel: [    2.496204] Key type dns_resolver registered
Nov 28 11:30:55 phoniebox kernel: [    2.499251] Registering SWP/SWPB emulation handler
Nov 28 11:30:55 phoniebox kernel: [    2.501899] registered taskstats version 1
Nov 28 11:30:55 phoniebox kernel: [    2.504276] Loading compiled-in X.509 certificates
Nov 28 11:30:55 phoniebox kernel: [    2.507157] Key type ._fscrypt registered
Nov 28 11:30:55 phoniebox kernel: [    2.509471] Key type .fscrypt registered
Nov 28 11:30:55 phoniebox kernel: [    2.522544] uart-pl011 3f201000.serial: cts_event_workaround enabled
Nov 28 11:30:55 phoniebox kernel: [    2.524958] 3f201000.serial: ttyAMA0 at MMIO 0x3f201000 (irq = 81, base_baud = 0) is a PL011 rev2
Nov 28 11:30:55 phoniebox kernel: [    2.531623] bcm2835-power bcm2835-power: Broadcom BCM2835 power domains driver
Nov 28 11:30:55 phoniebox kernel: [    2.535578] mmc-bcm2835 3f300000.mmcnr: mmc_debug:0 mmc_debug2:0
Nov 28 11:30:55 phoniebox kernel: [    2.538016] mmc-bcm2835 3f300000.mmcnr: DMA channel allocated
Nov 28 11:30:55 phoniebox kernel: [    2.566922] sdhost: log_buf @ (ptrval) (f7507000)
Nov 28 11:30:55 phoniebox kernel: [    2.605303] mmc1: queuing unknown CIS tuple 0x80 (2 bytes)
Nov 28 11:30:55 phoniebox kernel: [    2.609320] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
Nov 28 11:30:55 phoniebox kernel: [    2.613206] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
Nov 28 11:30:55 phoniebox kernel: [    2.618306] mmc1: queuing unknown CIS tuple 0x80 (7 bytes)
Nov 28 11:30:55 phoniebox kernel: [    2.620415] mmc0: sdhost-bcm2835 loaded - DMA enabled (>1)
Nov 28 11:30:55 phoniebox kernel: [    2.625120] of_cfs_init
Nov 28 11:30:55 phoniebox kernel: [    2.627458] of_cfs_init: OK
Nov 28 11:30:55 phoniebox kernel: [    2.630464] Waiting for root device PARTUUID=45a0f9f7-02...
Nov 28 11:30:55 phoniebox kernel: [    2.646965] Indeed it is in host mode hprt0 = 00021501
Nov 28 11:30:55 phoniebox kernel: [    2.688473] random: fast init done
Nov 28 11:30:55 phoniebox kernel: [    2.711781] mmc0: host does not support reading read-only switch, assuming write-enable
Nov 28 11:30:55 phoniebox kernel: [    2.720418] mmc0: new high speed SDHC card at address aaaa
Nov 28 11:30:55 phoniebox kernel: [    2.723900] mmcblk0: mmc0:aaaa SC32G 29.7 GiB
Nov 28 11:30:55 phoniebox kernel: [    2.732273]  mmcblk0: p1 p2
Nov 28 11:30:55 phoniebox kernel: [    2.755136] mmc1: new high speed SDIO card at address 0001
Nov 28 11:30:55 phoniebox kernel: [    2.758872] EXT4-fs (mmcblk0p2): INFO: recovery required on readonly filesystem
Nov 28 11:30:55 phoniebox kernel: [    2.761094] EXT4-fs (mmcblk0p2): write access will be enabled during recovery
Nov 28 11:30:55 phoniebox kernel: [    2.856620] EXT4-fs (mmcblk0p2): orphan cleanup on readonly fs
Nov 28 11:30:55 phoniebox kernel: [    2.856861] usb 1-1: new high-speed USB device number 2 using dwc_otg
Nov 28 11:30:55 phoniebox kernel: [    2.861164] Indeed it is in host mode hprt0 = 00001101
Nov 28 11:30:55 phoniebox kernel: [    2.872649] EXT4-fs (mmcblk0p2): 2 orphan inodes deleted
Nov 28 11:30:55 phoniebox kernel: [    2.874943] EXT4-fs (mmcblk0p2): recovery complete
Nov 28 11:30:55 phoniebox kernel: [    2.892885] EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Opts: (null)
Nov 28 11:30:55 phoniebox kernel: [    2.897792] VFS: Mounted root (ext4 filesystem) readonly on device 179:2.
Nov 28 11:30:55 phoniebox kernel: [    2.909661] devtmpfs: mounted
Nov 28 11:30:55 phoniebox kernel: [    2.925455] Freeing unused kernel memory: 1024K
Nov 28 11:30:55 phoniebox kernel: [    2.947258] Run /sbin/init as init process
Nov 28 11:30:55 phoniebox kernel: [    3.097213] usb 1-1: New USB device found, idVendor=0424, idProduct=9514, bcdDevice= 2.00
Nov 28 11:30:55 phoniebox kernel: [    3.102492] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
Nov 28 11:30:55 phoniebox kernel: [    3.106008] hub 1-1:1.0: USB hub found
Nov 28 11:30:55 phoniebox kernel: [    3.108815] hub 1-1:1.0: 5 ports detected
Nov 28 11:30:55 phoniebox kernel: [    3.426888] usb 1-1.1: new high-speed USB device number 3 using dwc_otg
Nov 28 11:30:55 phoniebox kernel: [    3.557200] usb 1-1.1: New USB device found, idVendor=0424, idProduct=ec00, bcdDevice= 2.00
Nov 28 11:30:55 phoniebox kernel: [    3.562548] usb 1-1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
Nov 28 11:30:55 phoniebox kernel: [    3.568263] smsc95xx v1.0.6
Nov 28 11:30:55 phoniebox kernel: [    3.661417] smsc95xx 1-1.1:1.0 eth0: register 'smsc95xx' at usb-3f980000.usb-1.1, smsc95xx USB 2.0 Ethernet, b8:27:eb:xx
Nov 28 11:30:55 phoniebox kernel: [    3.667641] usb 1-1-port2: over-current change #1
Nov 28 11:30:55 phoniebox kernel: [    3.704841] NET: Registered protocol family 10
Nov 28 11:30:55 phoniebox kernel: [    3.709427] Segment Routing with IPv6
Nov 28 11:30:55 phoniebox kernel: [    4.266932] usb 1-1.3: new low-speed USB device number 4 using dwc_otg
Nov 28 11:30:55 phoniebox kernel: [    4.410070] usb 1-1.3: New USB device found, idVendor=16c0, idProduct=27db, bcdDevice= 0.01
[... cut ...]
Nov 28 11:58:36 phoniebox mopidy[4120]: INFO     [MpdSession-31] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39390
Nov 28 11:58:36 phoniebox daemon_rfid_reader.py[4065]: volume: 30%   repeat: off   random: off   single: off   consume: off
Nov 28 11:58:36 phoniebox mopidy[4120]: INFO     [MpdSession-32] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:39394
Nov 28 11:58:36 phoniebox daemon_rfid_reader.py[4065]: OK MPD 0.19.0
Nov 28 11:58:36 phoniebox daemon_rfid_reader.py[4065]: OK

Here you go. Thanks.

s-martin, 2020-11-29:
I see two potential issues, first your configured buttons in GPIO control state "not enabled", @veloxidSchweiz is this expected?

@Hamsert please post the output of `cat ~/RPi-Jukebox-RFID/settings/gpio_settings.ini`

Second there is an (maybe unrelated) error in Mopidy Iris. 
You want to use Spotify edition, right?

Groovylein, 2020-11-29:
I've done an installation and modified the gpio-settings `~/RPi-Jukebox-RFID/settings/gpio_settings.ini`.
Everything works as expected!! :partying_face:

Hamsert, 2020-11-29:
> 
> 
> I see a potential issues, your configured buttons in GPIO control state "not enabled“.
> 
> Do you have `enabled: True` for every button in your settings file?
> 
> @Hamsert please post the output of `cat ~/RPi-Jukebox-RFID/settings/gpio_settings.ini`

pi@phoniebox:~ $ cat ~/RPi-Jukebox-RFID/settings/gpio_settings.ini
[DEFAULT]
enabled: True
[VolumeControl]
enabled: True
Type: TwoButtonControl ;or RotaryEncoder
PinUp: 5
PinDown: 6
pull_up: True
hold_time: 0.3
hold_repeat: True
timeBase: 0.1 ;only for RotaryEncoder
functionCallDown: functionCallVolD
functionCallUp: functionCallVolU
functionCallTwoButtons: functionCallVol0 ;only for TwoButtonControl

[PrevNextControl]
enabled: True
Type: TwoButtonControl
Pin1: 22
Pin2: 23
functionCall1: functionCallPlayerPrev
functionCall2: functionCallPlayerNext
functionCallTwoButtons: None
pull_up: True
hold_time: 0.3
hold_repeat: False

[PlayPause]
enabled: True
Type: Button
Pin: 27
pull_up: True
hold_time: 0.3
functionCall: functionCallPlayerPause

[Shutdown]
enabled: False
Type:  Button
Pin: 3
pull_up: True
hold_time: 2
functionCall: functionCallShutdown

[Volume0]
enabled: False
Type:  Button
Pin: 17
pull_up: True
hold_time: 0.3
functionCall: functionCallVol0

[VolumeUp]
enabled: False
Type:  Button
Pin: 16
pull_up: True
hold_time: 0.3
hold_repeat: True
functionCall: functionCallVolU

[VolumeDown]
enabled: False
Type:  Button
Pin: 19
pull_up: True
hold_time: 0.3
hold_repeat: True
functionCall: functionCallVolD

[NextSong]
enabled: False
Type:  Button
Pin: 26
pull_up: True
hold_time: 0.3
functionCall: functionCallPlayerNext

[PrevSong]
enabled: False
Type:  Button
Pin: 20
pull_up: True
hold_time: 0.3
functionCall: functionCallPlayerPrev

[Halt]
enabled: False
Type:  Button
Pin: 21
pull_up: True
hold_time: 0.3
functionCall: functionCallPlayerPauseForce

Do I have to connect the wires to the same pins as in version 2.0 and before or do I have to connect something differently? In the config everything seems not enabled but PlayPause is enabled but it also doesn't work (with the wiring from version 2.0).

s-martin, 2020-11-29:
The gpio_settings.ini looks fine, but you have to change the values for `pin` to the actual **GPIO number** where you have connected your button.

See here for numbering:
[image]

Second cable for each button should connect to a **GND** pin, but you don't have to enter the second pin in the file.

Hamsert, 2020-11-29:
> @Hamsert
> The gpio_settings.ini looks fine, but you have to change the values for `pin` to the actual **GPIO number** where you have connected your button.
> 
> See here for numbering:
> [image]
> 
> Second cable for each button should connect to a **Ground** pin, but you don't have to enter the second pin in the file.

Ah okay thanks.
So the standard config in the ini file does not match the standard config prior 2.1 and I basically have to change the config or reconnect the wires to match the new standard config.

s-martin, 2020-11-29:
> Ah okay thanks.
> So the standard config in the ini file does not match the standard config prior 2.1 and I basically have to change the config or reconnect the wires to match the new standard config.

Yes, that’s right. Please let us know, if you still have issues.

Hamsert, 2020-12-01:
> > Ah okay thanks.
> > So the standard config in the ini file does not match the standard config prior 2.1 and I basically have to change the config or reconnect the wires to match the new standard config.
> 
> Yes, that’s right. Please let us know, if you still have issues.

I connected the buttons today. They are all working except volume down. I already tried using another pin, another ground and I tried the button with another function to confirm it's not the button itself.

s-martin, 2020-12-05:
@Hamsert could you use different working and the not working button and afterwards post the log `tail -n 500 /var/log/syslog`?

Hamsert, 2020-12-05:
@s-martin what exactly should I do? The volume down function is not working, it doesn't matter which button I use. But I tested the button I tried for volume down for other functions and it works with other functions.

s-martin, 2020-12-05:
No, I mean just press working and not working buttons, so we can see what comes up in the log (or not).

Please post also the order you pressed the buttons, so we can check the log for which button something's missing.

Hamsert, 2020-12-06:
> No, I mean just press working and not working buttons, so we can see what comes up in the log (or not).
> 
> Please post also the order you pressed the buttons, so we can check the log for which button something's missing.

`Dec  6 13:30:29 phoniebox mopidy[4058]: INFO     [MpdSession-2235] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:51772
Dec  6 13:30:29 phoniebox mopidy[4058]: INFO     [MpdSession-2236] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:51776
Dec  6 13:30:29 phoniebox mopidy[4058]: INFO     [MpdSession-2237] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:51780
Dec  6 13:30:29 phoniebox PhonieboxGPIOControl[4018]: volume: 10%   repeat: off   random: off   single: off   consume: off
Dec  6 13:32:15 phoniebox mopidy[4058]: ERROR    [IrisFrontend-13] mopidy.listener Triggering event failed: track_playback_ended(tl_track, time_position)
Dec  6 13:32:15 phoniebox mopidy[4058]: Traceback (most recent call last):
Dec  6 13:32:15 phoniebox mopidy[4058]:   File "/usr/local/lib/python3.7/dist-packages/mopidy/listener.py", line 40, in on_event
Dec  6 13:32:15 phoniebox mopidy[4058]:     getattr(self, event)(**kwargs)
Dec  6 13:32:15 phoniebox mopidy[4058]:   File "/usr/local/lib/python3.7/dist-packages/mopidy_iris/frontend.py", line 26, in track_playback_ended
Dec  6 13:32:15 phoniebox mopidy[4058]:     iris.ioloop.add_callback(functools.partial(iris.check_for_radio_update))
Dec  6 13:32:15 phoniebox mopidy[4058]: AttributeError: 'NoneType' object has no attribute 'add_callback'
Dec  6 13:34:03 phoniebox mopidy[4058]: ERROR    [IrisFrontend-13] mopidy.listener Triggering event failed: track_playback_ended(tl_track, time_position)
Dec  6 13:34:03 phoniebox mopidy[4058]: Traceback (most recent call last):
Dec  6 13:34:03 phoniebox mopidy[4058]:   File "/usr/local/lib/python3.7/dist-packages/mopidy/listener.py", line 40, in on_event
Dec  6 13:34:03 phoniebox mopidy[4058]:     getattr(self, event)(**kwargs)
Dec  6 13:34:03 phoniebox mopidy[4058]:   File "/usr/local/lib/python3.7/dist-packages/mopidy_iris/frontend.py", line 26, in track_playback_ended
Dec  6 13:34:03 phoniebox mopidy[4058]:     iris.ioloop.add_callback(functools.partial(iris.check_for_radio_update))
Dec  6 13:34:03 phoniebox mopidy[4058]: AttributeError: 'NoneType' object has no attribute 'add_callback'
Dec  6 13:35:51 phoniebox mopidy[4058]: ERROR    [IrisFrontend-13] mopidy.listener Triggering event failed: track_playback_ended(tl_track, time_position)
Dec  6 13:35:51 phoniebox mopidy[4058]: Traceback (most recent call last):
Dec  6 13:35:51 phoniebox mopidy[4058]:   File "/usr/local/lib/python3.7/dist-packages/mopidy/listener.py", line 40, in on_event
Dec  6 13:35:51 phoniebox mopidy[4058]:     getattr(self, event)(**kwargs)
Dec  6 13:35:51 phoniebox mopidy[4058]:   File "/usr/local/lib/python3.7/dist-packages/mopidy_iris/frontend.py", line 26, in track_playback_ended
Dec  6 13:35:51 phoniebox mopidy[4058]:     iris.ioloop.add_callback(functools.partial(iris.check_for_radio_update))
Dec  6 13:35:51 phoniebox mopidy[4058]: AttributeError: 'NoneType' object has no attribute 'add_callback'
Dec  6 13:37:41 phoniebox mopidy[4058]: ERROR    [IrisFrontend-13] mopidy.listener Triggering event failed: track_playback_ended(tl_track, time_position)
Dec  6 13:37:41 phoniebox mopidy[4058]: Traceback (most recent call last):
Dec  6 13:37:41 phoniebox mopidy[4058]:   File "/usr/local/lib/python3.7/dist-packages/mopidy/listener.py", line 40, in on_event
Dec  6 13:37:41 phoniebox mopidy[4058]:     getattr(self, event)(**kwargs)
Dec  6 13:37:41 phoniebox mopidy[4058]:   File "/usr/local/lib/python3.7/dist-packages/mopidy_iris/frontend.py", line 26, in track_playback_ended
Dec  6 13:37:41 phoniebox mopidy[4058]:     iris.ioloop.add_callback(functools.partial(iris.check_for_radio_update))
Dec  6 13:37:41 phoniebox mopidy[4058]: AttributeError: 'NoneType' object has no attribute 'add_callback'
Dec  6 13:39:01 phoniebox CRON[15076]: (root) CMD (  [ -x /usr/lib/php/sessionclean ] && if [ ! -d /run/systemd/system ]; then /usr/lib/php/sessionclean; fi)
Dec  6 13:39:01 phoniebox systemd[1]: Starting Clean php session files...
Dec  6 13:39:01 phoniebox systemd[1]: phpsessionclean.service: Succeeded.
Dec  6 13:39:01 phoniebox systemd[1]: Started Clean php session files.
Dec  6 13:39:24 phoniebox mopidy[4058]: ERROR    [IrisFrontend-13] mopidy.listener Triggering event failed: track_playback_ended(tl_track, time_position)
Dec  6 13:39:24 phoniebox mopidy[4058]: Traceback (most recent call last):
Dec  6 13:39:24 phoniebox mopidy[4058]:   File "/usr/local/lib/python3.7/dist-packages/mopidy/listener.py", line 40, in on_event
Dec  6 13:39:24 phoniebox mopidy[4058]:     getattr(self, event)(**kwargs)
Dec  6 13:39:24 phoniebox mopidy[4058]:   File "/usr/local/lib/python3.7/dist-packages/mopidy_iris/frontend.py", line 26, in track_playback_ended
Dec  6 13:39:24 phoniebox mopidy[4058]:     iris.ioloop.add_callback(functools.partial(iris.check_for_radio_update))
Dec  6 13:39:24 phoniebox mopidy[4058]: AttributeError: 'NoneType' object has no attribute 'add_callback'
Dec  6 13:41:11 phoniebox mopidy[4058]: ERROR    [IrisFrontend-13] mopidy.listener Triggering event failed: track_playback_ended(tl_track, time_position)
Dec  6 13:41:11 phoniebox mopidy[4058]: Traceback (most recent call last):
Dec  6 13:41:11 phoniebox mopidy[4058]:   File "/usr/local/lib/python3.7/dist-packages/mopidy/listener.py", line 40, in on_event
[... cut ...]
Dec  6 16:04:18 phoniebox systemd[15661]: mpd.service: Main process exited, code=exited, status=1/FAILURE
Dec  6 16:04:18 phoniebox systemd[15661]: mpd.service: Failed with result 'exit-code'.
Dec  6 16:04:18 phoniebox systemd[15661]: Failed to start Music Player Daemon.
Dec  6 16:04:18 phoniebox systemd[15661]: Reached target Default.
Dec  6 16:04:18 phoniebox systemd[15661]: Startup finished in 1.371s.`

This is the log after pressing play, pause, volume down, volume down (starting at Dec  6 16:02:43). As it seems the volume down presses aren't shown in the log.

Hamsert, 2020-12-06:
This is the correct log, sorry I made mistakes in my earlier post.

Dec  6 16:57:00 phoniebox dhcpcd[390]: forked to background, child pid 3670
Dec  6 16:57:00 phoniebox systemd[1]: Started dhcpcd on all interfaces.
Dec  6 16:57:00 phoniebox systemd[1]: Reached target Network.
Dec  6 16:57:00 phoniebox systemd[1]: Reached target Network is Online.
Dec  6 16:57:00 phoniebox systemd[1]: Starting LSB: exim Mail Transport Agent...
Dec  6 16:57:00 phoniebox systemd[1]: Starting Lighttpd Daemon...
Dec  6 16:57:00 phoniebox systemd[1]: Starting Permit User Sessions...
Dec  6 16:57:00 phoniebox systemd[1]: Starting /etc/rc.local Compatibility...
Dec  6 16:57:00 phoniebox systemd[1]: Starting Samba NMB Daemon...
Dec  6 16:57:00 phoniebox systemd[1]: Condition check resulted in fast remote file copy program daemon being skipped.
Dec  6 16:57:00 phoniebox systemd[1]: Started Phoniebox GPIO Control Service.
Dec  6 16:57:00 phoniebox systemd[1]: Starting OpenBSD Secure Shell server...
Dec  6 16:57:00 phoniebox systemd[1]: Starting Mopidy music server...
Dec  6 16:57:00 phoniebox systemd[1]: Started Phoniebox RFID-Reader Service.
Dec  6 16:57:00 phoniebox systemd[1]: Started Phoniebox Idle Watchdog Service.
Dec  6 16:57:00 phoniebox systemd[1]: Started Permit User Sessions.
Dec  6 16:57:00 phoniebox systemd[1]: Started /etc/rc.local Compatibility.
Dec  6 16:57:00 phoniebox systemd[1]: Started Mopidy music server.
Dec  6 16:57:00 phoniebox systemd[1]: Started Getty on tty1.
Dec  6 16:57:00 phoniebox systemd[1]: Reached target Login Prompts.
Dec  6 16:57:00 phoniebox systemd[1]: Started OpenBSD Secure Shell server.
Dec  6 16:57:01 phoniebox systemd[1]: systemd-rfkill.service: Succeeded.
Dec  6 16:57:01 phoniebox systemd[1]: Started Lighttpd Daemon.
Dec  6 16:57:01 phoniebox PhonieboxGPIOControl[3691]: INFO:root:['VolumeControl', 'PrevNextControl', 'PlayPause', 'Shutdown', 'Volume0', 'VolumeUp', 'VolumeDown', 'NextSong', 'PrevSong', 'Halt']
Dec  6 16:57:01 phoniebox PhonieboxGPIOControl[3691]: INFO:root:adding GPIO-Device, VolumeControl
Dec  6 16:57:01 phoniebox PhonieboxGPIOControl[3691]: INFO:gpio_control:VolumeControl as TwoButtonControl
Dec  6 16:57:01 phoniebox PhonieboxGPIOControl[3691]: INFO:GPIODevices.two_button_control:adding new action
Dec  6 16:57:01 phoniebox PhonieboxGPIOControl[3691]: INFO:GPIODevices.simple_button:{}: set when_pressed
Dec  6 16:57:01 phoniebox PhonieboxGPIOControl[3691]: INFO:GPIODevices.simple_button:add new action
Dec  6 16:57:01 phoniebox PhonieboxGPIOControl[3691]: INFO:GPIODevices.simple_button:{}: set when_pressed
Dec  6 16:57:01 phoniebox PhonieboxGPIOControl[3691]: INFO:GPIODevices.simple_button:add new action
Dec  6 16:57:01 phoniebox PhonieboxGPIOControl[3691]: INFO:root:adding GPIO-Device, PrevNextControl
Dec  6 16:57:01 phoniebox PhonieboxGPIOControl[3691]: INFO:root:adding TwoButtonControl
Dec  6 16:57:01 phoniebox systemd[1]: Started Samba NMB Daemon.
Dec  6 16:57:01 phoniebox systemd[1]: Starting Samba SMB Daemon...
Dec  6 16:57:02 phoniebox PhonieboxGPIOControl[3691]: INFO:GPIODevices.two_button_control:adding new action
Dec  6 16:57:02 phoniebox PhonieboxGPIOControl[3691]: INFO:GPIODevices.simple_button:{}: set when_pressed
Dec  6 16:57:02 phoniebox PhonieboxGPIOControl[3691]: INFO:GPIODevices.simple_button:add new action
Dec  6 16:57:02 phoniebox exim4[3676]: Starting MTA: exim4.
Dec  6 16:57:02 phoniebox systemd[1]: Started LSB: exim Mail Transport Agent.
Dec  6 16:57:02 phoniebox PhonieboxGPIOControl[3691]: INFO:GPIODevices.simple_button:{}: set when_pressed
Dec  6 16:57:02 phoniebox PhonieboxGPIOControl[3691]: INFO:GPIODevices.simple_button:add new action
Dec  6 16:57:02 phoniebox daemon_rfid_reader.py[3702]: 2020-12-06 16:57:02,180 - root - INFO - Dir_PATH: /home/pi/RPi-Jukebox-RFID/scripts
Dec  6 16:57:02 phoniebox PhonieboxGPIOControl[3691]: INFO:root:adding GPIO-Device, PlayPause
Dec  6 16:57:02 phoniebox PhonieboxGPIOControl[3691]: INFO:root:Device Shutdown not enabled
Dec  6 16:57:02 phoniebox PhonieboxGPIOControl[3691]: INFO:root:Device Volume0 not enabled
Dec  6 16:57:02 phoniebox PhonieboxGPIOControl[3691]: INFO:root:Device VolumeUp not enabled
Dec  6 16:57:02 phoniebox PhonieboxGPIOControl[3691]: INFO:root:Device VolumeDown not enabled
Dec  6 16:57:02 phoniebox PhonieboxGPIOControl[3691]: INFO:root:Device NextSong not enabled
Dec  6 16:57:02 phoniebox PhonieboxGPIOControl[3691]: INFO:root:Device PrevSong not enabled
Dec  6 16:57:02 phoniebox PhonieboxGPIOControl[3691]: INFO:root:Device Halt not enabled
Dec  6 16:57:02 phoniebox PhonieboxGPIOControl[3691]: INFO:root:Ready for taking actions
Dec  6 16:57:03 phoniebox systemd[1]: Started Samba SMB Daemon.
Dec  6 16:57:05 phoniebox lighttpd[4170]: echo: write error: Broken pipe
Dec  6 16:57:06 phoniebox mopidy[3715]: INFO     [MainThread] mopidy.__main__ Starting Mopidy 3.0.2
Dec  6 16:57:06 phoniebox mopidy[3715]: INFO     [MainThread] mopidy.config Loading config from builtin defaults
Dec  6 16:57:06 phoniebox mopidy[3715]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy-local.conf
Dec  6 16:57:06 phoniebox mopidy[3715]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy.conf
Dec  6 16:57:06 phoniebox mopidy[3715]: INFO     [MainThread] mopidy.config Loading config from file:///etc/mopidy/mopidy.conf
[... cut ...]
Dec  6 16:59:09 phoniebox mpd: exception: bind to '127.0.0.1:6600' failed (continuing anyway, because binding to '[::1]:6600' succeeded): Failed to bind socket: Address already in use
Dec  6 16:59:09 phoniebox mpd: exception: Failed to set group 0: Operation not permitted
Dec  6 16:59:10 phoniebox systemd[7799]: mpd.service: Main process exited, code=exited, status=1/FAILURE
Dec  6 16:59:10 phoniebox systemd[7799]: mpd.service: Failed with result 'exit-code'.
Dec  6 16:59:10 phoniebox systemd[7799]: Failed to start Music Player Daemon.
Dec  6 16:59:10 phoniebox systemd[7799]: Reached target Default.
Dec  6 16:59:10 phoniebox systemd[7799]: Startup finished in 2.608s.

s-martin, 2020-12-15:
@Hamsert could you check, if the following fixes your issue:

1.  Replace content of requirements-spotify.txt with `Mopidy-Iris==3.54.2`
2.  `sudo pip install --upgrade --force-reinstall -r requirements-spotify.txt
3.  Reboot

Hamsert, 2020-12-15:
Where is requirements-spotify.txt located?

s-martin, 2020-12-15:
In the root directory of phoniebox

Hamsert, 2020-12-18:
> @Hamsert could you check, if the following fixes your issue:
> 
> 1. Replace content of requirements-spotify.txt with `Mopidy-Iris==3.54.2`
> 2. `sudo pip install --upgrade --force-reinstall -r requirements-spotify.txt`
> 3. Reboot

I get:
pi@phoniebox:~/RPi-Jukebox-RFID $ sudo pip install --upgrade --force-reinstall -r requirements-spotify.txt
sudo: pip: command not found

or
pi@phoniebox:~/RPi-Jukebox-RFID $ pip install --upgrade --force-reinstall -r requirements-spotify.txt
-bash: pip: command not found

s-martin, 2020-12-18:
Sorry, please try `pip3`

Hamsert, 2020-12-18:
> Sorry, please try `pip3`

The installation worked and the Phoniebox is working but VolumeDown still won't work.
Just to make sure:
This is configured:
[VolumeDown]
enabled: False
Type:  Button
Pin: 19
pull_up: True
hold_time: 0.3
hold_repeat: True
functionCall: functionCallVolD

And this is my wiring:
[image]

Hamsert, 2020-12-18:
@s-martin seems like you edited my post?

s-martin, 2020-12-18:
Sorry, wanted to answer.

Wrong click in the app ;) 
Could you try that anyway?

Hamsert, 2020-12-18:
Sure, here you go:
pi@phoniebox:~ $ journalctl -u mopidy
-- Logs begin at Fri 2020-12-18 10:29:53 GMT, end at Fri 2020-12-18 10:34:39 GMT. --
Dec 18 10:30:06 phoniebox systemd[1]: Starting Mopidy music server...
Dec 18 10:30:07 phoniebox systemd[1]: Started Mopidy music server.
Dec 18 10:30:13 phoniebox mopidy[3876]: INFO     [MainThread] mopidy.__main__ Starting Mopidy 3.0.2
Dec 18 10:30:13 phoniebox mopidy[3876]: INFO     [MainThread] mopidy.config Loading config from builtin defaults
Dec 18 10:30:13 phoniebox mopidy[3876]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy-local.conf
Dec 18 10:30:13 phoniebox mopidy[3876]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy.conf
Dec 18 10:30:13 phoniebox mopidy[3876]: INFO     [MainThread] mopidy.config Loading config from file:///etc/mopidy/mopidy.conf
Dec 18 10:30:13 phoniebox mopidy[3876]: INFO     [MainThread] mopidy.config Loading config from command line options
Dec 18 10:30:13 phoniebox mopidy[3876]: ERROR    [MainThread] mopidy.internal.log Loading logging config '/etc/mopidy/logging.conf' failed. 'formatters'
Dec 18 10:30:14 phoniebox mopidy[3876]: INFO     [MainThread] mopidy.__main__ Enabled extensions: file, spotify, softwaremixer, iris, local, m3u, mpd, stream, http
Dec 18 10:30:14 phoniebox mopidy[3876]: INFO     [MainThread] mopidy.__main__ Disabled extensions: none
Dec 18 10:30:15 phoniebox mopidy[3876]: INFO     [MainThread] mopidy.commands Starting Mopidy mixer: SoftwareMixer
Dec 18 10:30:15 phoniebox mopidy[3876]: INFO     [MainThread] mopidy.commands Mixer volume set to 30
Dec 18 10:30:15 phoniebox mopidy[3876]: INFO     [MainThread] mopidy.commands Starting Mopidy audio
Dec 18 10:30:15 phoniebox mopidy[3876]: INFO     [MainThread] mopidy.commands Starting Mopidy backends: FileBackend, M3UBackend, StreamBackend, SpotifyBackend, LocalBackend
Dec 18 10:30:15 phoniebox mopidy[3876]: INFO     [Audio-2] mopidy.audio.actor Audio output set to "alsasink"
Dec 18 10:30:16 phoniebox mopidy[3876]: INFO     [SpotifyBackend-6] mopidy_spotify.web Logged into Spotify Web API as f4iemrc0k9ohse2dzchipyy3c
Dec 18 10:30:16 phoniebox mopidy[3876]: INFO     [SpotifyEventLoop] mopidy_spotify.backend Logged in to Spotify in online mode
Dec 18 10:30:16 phoniebox mopidy[3876]: INFO     [SpotifyBackend-6] mopidy_spotify.playlists Refreshed 0 Spotify playlists
Dec 18 10:30:16 phoniebox mopidy[3876]: INFO     [MainThread] mopidy.commands Starting Mopidy core
Dec 18 10:30:16 phoniebox mopidy[3876]: INFO     [MainThread] mopidy.commands Starting Mopidy frontends: HttpFrontend, MpdFrontend, IrisFrontend
Dec 18 10:30:16 phoniebox mopidy[3876]: INFO     [HttpFrontend-12] mopidy.http.actor HTTP server running at [::ffff:0.0.0.0]:6680
Dec 18 10:30:16 phoniebox mopidy[3876]: INFO     [MainThread] mopidy_mpd.actor MPD server running at [::ffff:0.0.0.0]:6600
Dec 18 10:30:16 phoniebox mopidy[3876]: INFO     [IrisFrontend-14] mopidy_iris.core Starting Iris 3.54.2
Dec 18 10:30:16 phoniebox mopidy[3876]: INFO     [MainThread] mopidy.commands Starting GLib mainloop
Dec 18 10:30:16 phoniebox mopidy[3876]: INFO     [MpdSession-15] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:47288
Dec 18 10:30:16 phoniebox mopidy[3876]: INFO     [MpdSession-16] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:47292
Dec 18 10:30:16 phoniebox mopidy[3876]: INFO     [MpdSession-17] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:47296
Dec 18 10:30:16 phoniebox mopidy[3876]: INFO     [MpdSession-18] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:47300
Dec 18 10:30:21 phoniebox mopidy[3876]: INFO     [MpdSession-19] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:47304
Dec 18 10:31:43 phoniebox mopidy[3876]: INFO     [MpdSession-20] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:47308
Dec 18 10:31:43 phoniebox mopidy[3876]: INFO     [MpdSession-21] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:47312
Dec 18 10:31:43 phoniebox mopidy[3876]: INFO     [MpdSession-22] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:47316
Dec 18 10:31:43 phoniebox mopidy[3876]: INFO     [MpdSession-23] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:47320
Dec 18 10:31:43 phoniebox mopidy[3876]: INFO     [MpdSession-24] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:47324
Dec 18 10:31:43 phoniebox mopidy[3876]: INFO     [MpdSession-25] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:47328
Dec 18 10:31:43 phoniebox mopidy[3876]: INFO     [MpdSession-26] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:47332
Dec 18 10:31:43 phoniebox mopidy[3876]: INFO     [MpdSession-27] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:47336
Dec 18 10:31:44 phoniebox mopidy[3876]: INFO     [MpdSession-28] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:47340
Dec 18 10:31:44 phoniebox mopidy[3876]: INFO     [MpdSession-29] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:47344
Dec 18 10:31:44 phoniebox mopidy[3876]: INFO     [MpdSession-30] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:47348
Dec 18 10:31:44 phoniebox mopidy[3876]: INFO     [MpdSession-31] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:47352
[... cut ...]
Dec 18 10:34:27 phoniebox mopidy[3876]: INFO     [MpdSession-219] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:48088
Dec 18 10:34:27 phoniebox mopidy[3876]: INFO     [MpdSession-220] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:48092
Dec 18 10:34:27 phoniebox mopidy[3876]: INFO     [MpdSession-221] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:48096
Dec 18 10:34:27 phoniebox mopidy[3876]: INFO     [MpdSession-222] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:48100

Hamsert, 2020-12-18:
> > 
> 
> The installation worked and the Phoniebox is working
> 
> Ok cool, could you post the output of `journalctl -u mopidy` so I can check, if some errors are gone?
> 
> > but VolumeDown still won't work.
> > Just to make sure:
> > This is configured:
> > [VolumeDown]
> > enabled: False
> 
> This needs to be `True`

Okay, so we did some unnecessary troubleshooting since this was already false in the first post of the ini file.
Here is the output of the ini file again. What's confusing to me though is that VolumeUp is also false but it has been working from the beginning. This is why this was so strange to me...

pi@phoniebox:~ $ cat ~/RPi-Jukebox-RFID/settings/gpio_settings.ini
[DEFAULT]
enabled: True
[VolumeControl]
enabled: True
Type: TwoButtonControl ;or RotaryEncoder
PinUp: 5
PinDown: 6
pull_up: True
hold_time: 0.3
hold_repeat: True
timeBase: 0.1 ;only for RotaryEncoder
functionCallDown: functionCallVolD
functionCallUp: functionCallVolU
functionCallTwoButtons: functionCallVol0 ;only for TwoButtonControl

[PrevNextControl]
enabled: True
Type: TwoButtonControl
Pin1: 22
Pin2: 23
functionCall1: functionCallPlayerPrev
functionCall2: functionCallPlayerNext
functionCallTwoButtons: None
pull_up: True
hold_time: 0.3
hold_repeat: False

[PlayPause]
enabled: True
Type: Button
Pin: 27
pull_up: True
hold_time: 0.3
functionCall: functionCallPlayerPause

[Shutdown]
enabled: False
Type:  Button
Pin: 3
pull_up: True
hold_time: 2
functionCall: functionCallShutdown

[Volume0]
enabled: False
Type:  Button
Pin: 17
pull_up: True
hold_time: 0.3
functionCall: functionCallVol0

[VolumeUp]
enabled: False
Type:  Button
Pin: 16
pull_up: True
hold_time: 0.3
hold_repeat: True
functionCall: functionCallVolU

[VolumeDown]
enabled: True
Type:  Button
Pin: 19
pull_up: True
hold_time: 0.3
hold_repeat: True
functionCall: functionCallVolD

[NextSong]
enabled: False
Type:  Button
Pin: 26
pull_up: True
hold_time: 0.3
functionCall: functionCallPlayerNext

[PrevSong]
enabled: False
Type:  Button
Pin: 20
pull_up: True
hold_time: 0.3
functionCall: functionCallPlayerPrev

[Halt]
enabled: False
Type:  Button
Pin: 21
pull_up: True
hold_time: 0.3
functionCall: functionCallPlayerPauseForce

Thanks for your help and merry christmas!

s-martin, 2020-12-18:
But does the button work now?

Hamsert, 2020-12-18:
> But does the button work now?

Yes it works, but volume up, which is set to false also works.

s-martin, 2020-12-18:
Ok, we'll catch that in a separate issue.

closed 2020-12-18 by s-martin

Hamsert, 2020-12-18:
@s-martin should I set the VolumeUp to true as well?

s-martin, 2020-12-18:
Yes, because that's the intention.