MiczFlor/RPi-Jukebox-RFID #1163: Drop Stretch and Python 3.5 support

s-martin, 2020-11-26:
## Feature Description

### What functionality would you like to see in your phoniebox?

i.e. `I would love to cook my breakfast eggs with the help of our phoniebox`

### How do you envision the feature to work from a users perspective?

i.e. `In the mornings when I stumble to the bathroom, I want the phoniebox to react on the RFID chip injected under the skin of my right hip. I envision it to walk to the kitchen, get the eggs and a pot out, and start the cooking of the eggs on the stove.`

### Further information that might help

i.e. `the stove is operated with cooking gas, so the implemenation has to make sure the stove is handled with the greatest possible care`

s-martin, 2020-11-29:
For the moment I would just update docs explicitly stating that Stretch isn't supported anymore and remove the requirement for Python to be compatible to 3.5.

We could keep Stretch related code for the moment, but not require them to be up to date anymore.

s-martin, 2021-01-01:
New implementation for USB encoder #1249 fails on python 3.5 (see https://github.com/MiczFlor/RPi-Jukebox-RFID/runs/1633225851?check_suite_focus=true#logs).

But as I said we should drop the Stretch and Python 3.5 support

jeripeierSBB, 2021-01-01:
> New implementation for USB encoder #1249 fails on python 3.5 (see https://github.com/MiczFlor/RPi-Jukebox-RFID/runs/1633225851?check_suite_focus=true#logs).
> 
> But as I said we should drop the Stretch and Python 3.5 support

If you @s-martin like I can implement that backwards compatible?

s-martin, 2021-01-01:
If it's an easy fix you could do that for the moment.

jeripeierSBB, 2021-01-01:
@s-martin see #1253

s-martin, 2021-02-27:
I removed now Stretch from the INSTALL wiki and from CONTRIBUTING.md as well as the Python Action

closed 2021-02-27 by s-martin