MiczFlor/RPi-Jukebox-RFID #1164: 🚀 | Proposal to display WebRadio names on HD44780 LCD display

Conendrum, 2020-11-27:
Hi @davidfri, @s-martin,

I wanted to propose an improvement to the **i2c_lcd.py** script to allow displaying WebRadio names.

In my setup, i have RFID cards for:

- music playlist
- kid story
- webradio

the HD44780 LCD correctly displays the Artist and Title information when using a "music playlist" or a "kid story". However, when listening to "WebRadio", nothing is displayed under Artist (line 2 of the display) and Title (line 3 of the display).

I had a look at the code in **i2c_lcd.py** and saw that "current_song_infos['title']" and "current_song_infos['artist']" are emtpy when using a webradio (="livestream.txt" file under "shared/audiofolders/[Folder_name_of_webradio]").

When listening to webradios, I've seen that "current_song_infos['name']" is automatically populated by the MPD logic to display the name of the webradio ("Title" of the song being played in the webradio is however not populated).

My proposal would be to change the "artist" section of the i2c_lcd.py from:

```
## read in artist info                                                            #
            try:                                                                              #
                artist = current_song_infos['artist']                                     #
                artist = artist.replace("\n", "").replace("ä", "\341").replace("ö", "\357").replace("ü", "\365").replace("ß", "\342").replace("Ä", "\341").replace("Ö", "\357").replace("Ü", "\365")  # weitere codes siehe https://www.mikrocontroller.net/topic/293125                         #
            except KeyError:                                                                  #
                artist = ""                                                               #
```

to

```
## read in artist info                                                            #
            try:                                                                              #
                if playlist_length == "1"
                    artist = current_song_info['name'] # Webradio
                else:
                    artist = current_song_infos['artist']                                     #
                    artist = artist.replace("\n", "").replace("ä", "\341").replace("ö", "\357").replace("ü", "\365").replace("ß", "\342").replace("Ä", "\341").replace("Ö", "\357").replace("Ü", "\365")  # weitere codes siehe https://www.mikrocontroller.net/topic/293125                         #
            except KeyError:                                                                  #
                artist = ""                                                               #
```

For me, it works and webRadio names are now displayed. 

Regarding compatibility with music playlist and kid story:
- music playlist have always a playlist_length > 1, so no problem
- kid stories dont have an "artist" name, only a "title, so no issue here as well for me

what do you think?

cheers

s-martin, 2020-11-27:
Please check the PR #1166

davidfri, 2020-11-27:
Hi @Conendrum 
ich schreibe mal auf deutsch, da du laut Signatur aus München kommst und ich mich mit der englischen Sprache nicht so ausdrücken kann, wie ich es gerne möchte ;)
Es ist klasse, dass du die Anzeige optimieren möchtest und ich versuche so gut es geht zu unterstützen. Eine playlist_length = 1 sehe ich nicht als Indikator für einen Stream, da der ein oder andere vielleicht mal eine alte Kasette oder Platte digitalisiert hat und diese als einen Track gespeichert hat. Zudem würde ich den Standard-Fall nicht im "else" einbinden, sondern immer den Sonderfall in den "else" packen. 
Hinzu kommt, dass ""current_song_infos['title']" nicht bei jedem Stream leer ist. 
"Die Maus" hat zum Beispiel einen Titel:
```{'songid': '3', 'playlistlength': '1', 'playlist': '10', 'repeat': '1', 'consume': '0', 'mixrampdb': '0.000000', 'random': '0', 'state': 'play', 'elapsed': '621.874', 'volume': '46', 'single': '0', 'nextsong': '0', 'time': '622:0', 'song': '0', 'audio': '48000:24:2', 'bitrate': '128', 'nextsongid': '3'}
{'id': '3', 'pos': '0', 'name': 'Die Maus, Westdeutscher Rundfunk Koeln', 'file': 'https://wdr-diemaus-live.icecastssl.wdr.de/wdr/diemaus/live/mp3/128/stream.mp3', 'title': 'Die Sendung mit der Maus - Lach- und Sachgeschichten zum H?ren'}
```
Was würdest du von dem folgenden Ansatz halten (ungetestet!):
```## read in artist info                                                            #
            try:                                                                              #
                artist = current_song_infos['artist']                                     #
                artist = artist.replace("\n", "").replace("ä", "\341").replace("ö", "\357").replace("ü", "\365").replace("ß", "\342").replace("Ä", "\341").replace("Ö", "\357").replace("Ü", "\365")  # weitere codes siehe https://www.mikrocontroller.net/topic/293125                         #
            except KeyError:                                                                  #
                try:                                                                              #
                    artist = current_song_infos['name']                                     #
                    artist = artist.replace("\n", "").replace("ä", "\341").replace("ö", "\357").replace("ü", "\365").replace("ß", "\342").replace("Ä", "\341").replace("Ö", "\357").replace("Ü", "\365")  # weitere codes siehe https://www.mikrocontroller.net/topic/293125                         #

                except KeyError:                                                                  #
                    artist = ""   
```
Damit würden wir einfach bei einem nicht vorhandenem "artist" an der Stelle "name" darstellen.
Schöne Grüße
David

s-martin, 2020-11-29:
You could checkout branch https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/Improve-lcd to improve the changes.

Conendrum, 2020-11-29:
Hey @davidfri,

Danke fürs schnelle Review! Hab grad deinen Vorschlag ausprobiert und es funktioniert auch problemlos. 

Und ich gebe dir Recht: dein Vorschlag ist besser strukturiert, um die von dir vorgeschlagenen "Use-Cases" abzudecken (Digitalisierung einer Kassette/Platte usw...). Dieses Projekt ist so groß dass es irgenwie schwer an allen möglichen Use-Cases zu denken ;-)

@s-martin kannst du statt meiner Änderung David's Vorschlag im "improve-lcd" Branch mit rein nehmen?

Danke und gruß

s-martin, 2020-11-29:
Bitte PR prüfen, ob das so funktioniert

Conendrum, 2020-11-29:
Danke @s-martin

closed 2020-11-29 by s-martin

davidfri, 2020-11-29:
Top!