MiczFlor/RPi-Jukebox-RFID #1167: Phoniebox 2.2 - start sound | shutdown sound not working

fatdreads, 2020-11-28:
## Bug

### What I did

fresh install of Rasbian + Phoniebox 2.2(Spotify) -> reboot

### What happened

 start sound & shutdown sound arn´t working.

### I expected this to happen

 start sound & shutdown sound is working

### Further information that might help

sound device is ok -> hifiberry miniamp
mp3´s are working

## Software

### Base image and version

2020-08-20-raspios-buster-armhf-full.img

### Branch / Release

master - i guess

### Installscript

cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/master/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; ./buster-install-default.sh 

## Hardware

RaspberryPi 4 2GB

### Other notable hardware

PowerBlock
https://github.com/petrockblog/PowerBlock
https://www.petrockblock.com/2015/07/04/powerblock-another-power-switch-for-the-raspberry-pi/

s-martin, 2020-11-29:
Please provide outputs of `tail -n 500 /var/log/syslog`.

fatdreads, 2020-11-29:
well... 

`pi@phoniebox:~ $ tail -n 500 /var/log/syslog
Nov 29 15:24:26 phoniebox mopidy[542]: INFO     [MpdSession-90] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:40552
Nov 29 15:24:26 phoniebox mopidy[542]: INFO     [MpdSession-91] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:40556
Nov 29 15:24:26 phoniebox mopidy[542]: INFO     [MpdSession-92] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:40560
Nov 29 15:24:31 phoniebox mopidy[542]: INFO     [MpdSession-93] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:40564
Nov 29 15:24:31 phoniebox mopidy[542]: INFO     [MpdSession-94] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:40568
Nov 29 15:24:31 phoniebox mopidy[542]: INFO     [MpdSession-95] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:40572
Nov 29 15:24:31 phoniebox mopidy[542]: INFO     [MpdSession-96] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:40576
Nov 29 15:24:31 phoniebox mopidy[542]: INFO     [MpdSession-97] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:40580
Nov 29 15:24:32 phoniebox mopidy[542]: INFO     [MpdSession-98] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:40584
Nov 29 15:24:32 phoniebox mopidy[542]: INFO     [MpdSession-99] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:40588
Nov 29 15:24:32 phoniebox mopidy[542]: INFO     [MpdSession-100] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:40592
Nov 29 15:24:37 phoniebox mopidy[542]: INFO     [MpdSession-101] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:40596
Nov 29 15:24:37 phoniebox mopidy[542]: INFO     [MpdSession-102] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:40600
Nov 29 15:24:37 phoniebox mopidy[542]: INFO     [MpdSession-103] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:40604
Nov 29 15:24:37 phoniebox mopidy[542]: INFO     [MpdSession-104] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:40608
Nov 29 15:24:37 phoniebox mopidy[542]: INFO     [MpdSession-105] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:40612
Nov 29 15:24:37 phoniebox mopidy[542]: INFO     [MpdSession-106] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:40616
Nov 29 15:24:37 phoniebox mopidy[542]: INFO     [MpdSession-107] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:40620
Nov 29 15:24:37 phoniebox mopidy[542]: INFO     [MpdSession-108] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:40624
Nov 29 15:24:43 phoniebox mopidy[542]: INFO     [MpdSession-109] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:40628
Nov 29 15:24:43 phoniebox mopidy[542]: INFO     [MpdSession-110] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:40632
Nov 29 15:24:43 phoniebox mopidy[542]: INFO     [MpdSession-111] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:40636
Nov 29 15:24:43 phoniebox mopidy[542]: INFO     [MpdSession-112] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:40640
Nov 29 15:24:43 phoniebox mopidy[542]: INFO     [MpdSession-113] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:40644
Nov 29 15:24:43 phoniebox mopidy[542]: INFO     [MpdSession-114] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:40648
Nov 29 15:24:43 phoniebox mopidy[542]: INFO     [MpdSession-115] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:40652
Nov 29 15:24:43 phoniebox mopidy[542]: INFO     [MpdSession-116] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:40656
Nov 29 15:24:48 phoniebox mopidy[542]: INFO     [MpdSession-117] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:40660
Nov 29 15:24:49 phoniebox mopidy[542]: INFO     [MpdSession-118] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:40664
Nov 29 15:24:49 phoniebox mopidy[542]: INFO     [MpdSession-119] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:40668
Nov 29 15:24:49 phoniebox mopidy[542]: INFO     [MpdSession-120] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:40672
Nov 29 15:24:49 phoniebox mopidy[542]: INFO     [MpdSession-121] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:40676
Nov 29 15:24:49 phoniebox mopidy[542]: INFO     [MpdSession-122] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:40680
Nov 29 15:24:49 phoniebox mopidy[542]: INFO     [MpdSession-123] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:40684
Nov 29 15:24:49 phoniebox mopidy[542]: INFO     [MpdSession-124] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:40688
Nov 29 15:24:54 phoniebox mopidy[542]: INFO     [MpdSession-125] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:40692
Nov 29 15:24:54 phoniebox mopidy[542]: INFO     [MpdSession-126] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:40696
Nov 29 15:24:54 phoniebox mopidy[542]: INFO     [MpdSession-127] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:40700
Nov 29 15:24:55 phoniebox mopidy[542]: INFO     [MpdSession-128] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:40704
Nov 29 15:24:55 phoniebox mopidy[542]: INFO     [MpdSession-129] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:40708
Nov 29 15:24:55 phoniebox mopidy[542]: INFO     [MpdSession-130] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:40712
[... cut ...]
Nov 29 15:30:24 phoniebox mopidy[542]: INFO     [MpdSession-585] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42532
Nov 29 15:30:24 phoniebox mopidy[542]: INFO     [MpdSession-586] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42536
Nov 29 15:30:24 phoniebox mopidy[542]: INFO     [MpdSession-587] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42540
Nov 29 15:30:24 phoniebox mopidy[542]: INFO     [MpdSession-588] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42544`

s-martin, 2020-11-29:
Ok, please turn off Phoniebox and start again and afterwards call the log again.

matthias-pelger, 2020-12-01:
The problem seems to be, that mpg123
`Dec  1 12:24:23 Phoniebox PhonieboxGPIOControl[2535]: INFO:MPDStatusLED:Connection to MPD server on host localhost:6600 established
Dec  1 12:24:23 Phoniebox PhonieboxGPIOControl[2535]: INFO:root:Ready for taking actions
Dec  1 12:24:24 Phoniebox bash[398]: 0 is the mpg123 startup volume
Dec  1 12:24:24 Phoniebox bash[398]: High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
Dec  1 12:24:24 Phoniebox bash[398]: #011version 1.25.10; written and copyright by Michael Hipp and others
Dec  1 12:24:24 Phoniebox bash[398]: #011free software (LGPL) without any warranty but with best wishes
Dec  1 12:24:24 Phoniebox bash[398]: Directory: /home/pi/RPi-Jukebox-RFID/shared/
Dec  1 12:24:24 Phoniebox bash[398]: Playing MPEG stream 1 of 1: startupsound.mp3 ...
Dec  1 12:24:24 Phoniebox bash[398]: MPEG 1.0 L III cbr64 44100 j-s
Dec  1 12:24:24 Phoniebox bash[398]: Artist:  Phoniebox
`

In settings/global.conf, just edit:
from 
`AUDIOVOLSTARTUP="0"`
to
`AUDIOVOLSTARTUP="30"`

That did the trick for me.

s-martin, 2020-12-01:
Thanks for finding that bug, i opened PR #1169 to fix this.

fatdreads, 2020-12-02:
pi@phoniebox:~ $ tail -n 500 /var/log/syslog
Dec  2 21:21:30 phoniebox systemd[1]: Listening on triggerhappy.socket.
Dec  2 21:21:30 phoniebox systemd[1]: Reached target Sockets.
Dec  2 21:21:30 phoniebox systemd[1]: Reached target Basic System.
Dec  2 21:21:30 phoniebox systemd[1]: Starting Check for Raspberry Pi EEPROM updates...
Dec  2 21:21:30 phoniebox systemd[1]: Starting Deferred execution scheduler...
Dec  2 21:21:30 phoniebox systemd[1]: Starting LSB: Switch to ondemand cpu governor (unless shift key is pressed)...
Dec  2 21:21:30 phoniebox systemd[1]: Started Regular background program processing daemon.
Dec  2 21:21:30 phoniebox systemd[1]: Starting Phoniebox Startup...
Dec  2 21:21:30 phoniebox systemd[1]: Starting Avahi mDNS/DNS-SD Stack...
Dec  2 21:21:30 phoniebox cron[354]: (CRON) INFO (pidfile fd = 3)
Dec  2 21:21:30 phoniebox systemd[1]: Started D-Bus System Message Bus.
Dec  2 21:21:30 phoniebox cron[354]: (CRON) INFO (Running @reboot jobs)
Dec  2 21:21:30 phoniebox systemd[1]: Condition check resulted in Turn on SSH if /boot/ssh is present being skipped.
Dec  2 21:21:30 phoniebox systemd[1]: Starting WPA supplicant...
Dec  2 21:21:30 phoniebox rpi-eeprom-update[349]: BCM2711 detected
Dec  2 21:21:30 phoniebox rpi-eeprom-update[349]: Dedicated VL805 EEPROM detected
Dec  2 21:21:30 phoniebox systemd[1]: Starting LSB: Monitoring GPIO shutdown trigger...
Dec  2 21:21:30 phoniebox systemd[1]: Starting rng-tools.service...
Dec  2 21:21:30 phoniebox systemd[1]: Starting Disk Manager...
Dec  2 21:21:30 phoniebox systemd[1]: Starting Login Service...
Dec  2 21:21:30 phoniebox systemd[1]: Starting System Logging Service...
Dec  2 21:21:30 phoniebox systemd[1]: Starting Check for v3d driver...
Dec  2 21:21:30 phoniebox systemd[1]: Started Manage Sound Card State (restore and store).
Dec  2 21:21:30 phoniebox bash[356]: AUDIOFOLDERSPATH="/home/pi/RPi-Jukebox-RFID/shared/audiofolders"
Dec  2 21:21:30 phoniebox bash[356]: PLAYLISTSFOLDERPATH="/home/pi/RPi-Jukebox-RFID/playlists"
Dec  2 21:21:30 phoniebox bash[356]: SECONDSWIPE="RESTART"
Dec  2 21:21:30 phoniebox bash[356]: SECONDSWIPEPAUSE="2"
Dec  2 21:21:30 phoniebox bash[356]: SECONDSWIPEPAUSECONTROLS="ON"
Dec  2 21:21:30 phoniebox bash[356]: AUDIOIFACENAME="Master"
Dec  2 21:21:30 phoniebox bash[356]: VOLUMEMANAGER="mpd"
Dec  2 21:21:30 phoniebox bash[356]: AUDIOVOLCHANGESTEP="3"
Dec  2 21:21:30 phoniebox bash[356]: AUDIOVOLMAXLIMIT="100"
Dec  2 21:21:30 phoniebox bash[356]: AUDIOVOLMINLIMIT="1"
Dec  2 21:21:30 phoniebox bash[356]: AUDIOVOLSTARTUP="0"
Dec  2 21:21:30 phoniebox bash[356]: VOLCHANGEIDLE="TRUE"
Dec  2 21:21:30 phoniebox bash[356]: IDLETIMESHUTDOWN="0"
Dec  2 21:21:30 phoniebox bash[356]: POWEROFFCMD="sudo poweroff"
Dec  2 21:21:30 phoniebox bash[356]: SHOWCOVER="ON"
Dec  2 21:21:30 phoniebox bash[356]: READWLANIPYN="OFF"
Dec  2 21:21:30 phoniebox bash[356]: EDITION="plusSpotify"
Dec  2 21:21:30 phoniebox bash[356]: LANG="en-UK"
Dec  2 21:21:30 phoniebox bash[356]: VERSION="2.2 - 305325d - master"
Dec  2 21:21:30 phoniebox bash[356]: CMDVOLUP=""
Dec  2 21:21:30 phoniebox bash[356]: CMDVOLDOWN=""
Dec  2 21:21:30 phoniebox bash[356]: CMDNEXT=""
Dec  2 21:21:30 phoniebox bash[356]: CMDPREV=""
Dec  2 21:21:30 phoniebox bash[356]: CMDREWIND=""
Dec  2 21:21:30 phoniebox bash[356]: CMDSEEKFORW=""
Dec  2 21:21:30 phoniebox bash[356]: CMDSEEKBACK=""
Dec  2 21:21:30 phoniebox bash[356]: 0 is the mpd startup volume
Dec  2 21:21:30 phoniebox systemd[1]: Starting Save/Restore Sound Card State...
Dec  2 21:21:30 phoniebox rng-tools[367]: Starting Hardware RNG entropy gatherer daemon: rngd.
Dec  2 21:21:30 phoniebox avahi-daemon[357]: Found user 'avahi' (UID 108) and group 'avahi' (GID 113).
Dec  2 21:21:30 phoniebox systemd[1]: Starting Configure Bluetooth Modems connected by UART...
Dec  2 21:21:30 phoniebox avahi-daemon[357]: Successfully dropped root privileges.
Dec  2 21:21:30 phoniebox systemd[1]: Condition check resulted in Copy user wpa_supplicant.conf being skipped.
Dec  2 21:21:30 phoniebox avahi-daemon[357]: avahi-daemon 0.7 starting up.
Dec  2 21:21:30 phoniebox alsactl[375]: alsactl 1.1.8 daemon started
Dec  2 21:21:30 phoniebox rngd[380]: rngd 2-unofficial-mt.14 starting up...
Dec  2 21:21:30 phoniebox systemd[1]: Starting dphys-swapfile - set up, mount/unmount, and delete a swap file...
Dec  2 21:21:30 phoniebox systemd[1]: Starting triggerhappy global hotkey daemon...
Dec  2 21:21:30 phoniebox rsyslogd: imuxsock: Acquired UNIX socket '/run/systemd/journal/syslog' (fd 3) from systemd.  [v8.1901.0]
Dec  2 21:21:30 phoniebox rsyslogd:  [origin software="rsyslogd" swVersion="8.1901.0" x-pid="370" x-info="https://www.rsyslog.com"] start
Dec  2 21:21:30 phoniebox kernel: [    1.443527] usb 1-1: new high-speed USB device number 2 using xhci_hcd
Dec  2 21:21:30 phoniebox kernel: [    1.626139] usb 1-1: New USB device found, idVendor=2109, idProduct=3431, bcdDevice= 4.21
Dec  2 21:21:30 phoniebox kernel: [    1.626159] usb 1-1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
Dec  2 21:21:30 phoniebox kernel: [    1.626175] usb 1-1: Product: USB2.0 Hub
Dec  2 21:21:30 phoniebox kernel: [    1.628510] hub 1-1:1.0: USB hub found
Dec  2 21:21:30 phoniebox kernel: [    1.628802] hub 1-1:1.0: 4 ports detected
[... 406 lines cut ...]
Dec  2 21:21:43 phoniebox bash[356]: #011version 1.25.10; written and copyright by Michael Hipp and others
[... cut ...]
Dec  2 21:21:48 phoniebox systemd[1]: Started Update UTMP about System Runlevel Changes.
Dec  2 21:21:48 phoniebox systemd[1]: Startup finished in 1.727s (kernel) + 23.729s (userspace) = 25.457s.
Dec  2 21:21:52 phoniebox systemd[1]: Started Session 3 of user pi.
Dec  2 21:21:59 phoniebox systemd[1]: systemd-fsckd.service: Succeeded.
Dec  2 21:22:13 phoniebox systemd-timesyncd[306]: Synchronized to time server for the first time 213.172.105.106:123 (2.debian.pool.ntp.org).
Dec  2 21:22:21 phoniebox systemd[1]: systemd-hostnamed.service: Succeeded.

s-martin, 2020-12-02:
Please try the fix in https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/1167#issuecomment-736576390

fatdreads, 2020-12-02:
fix worked, thanks!

PR #1169 (2020-12-02): Fix volume of startup sound
Fixes #1167.

Default value was 0, so users would assume startup sound doesn't work.
commit 7ee558c6 (2020-12-02): Fix volume of startup sound (#1169)

Fixes #1167
commit 24258901 (2021-08-11): Release 2.3 (#1476)

* implemented shutdownwithreducingvolume

* implemented shutdownwithreducingvolume

* fixed type

* activation of shutdown button class in gpio_control

* update to next version number

* fixed date for 2.2 release

* go back for commit to master

* update to next release

* Minor changes

* Add version number to startup script

* Improve reading artist (#1166)

* Improve reading artist

Implement #1164

* Fix syntax

* Fix syntax

* More idiomatic check

* Feature/gpio debugging (#1165)

* change logging

* functionCall in case of invalid read of btn

* adjust behavior to [... cut]
commit abcf6e92 (2022-12-18): Release 2.4 (#1898)

* Feature/gpio debugging (#1165)

* change logging

* functionCall in case of invalid read of btn

* adjust behavior to fix strange behavior in case of button press is not stead
y

* Set Own Repo

* Fix volume of startup sound (#1169)

Fixes #1167

* missing savepos-command in "playerstopafter"

As there is a savepos-command in "playerstop" this should also be included in "playerstopafter".

* Update buster-install-default.sh (#1181)

Added spacing

* Revert "Set Own Repo"

This reverts commit 6d504e5158f8ebff98fbf90c11b5410bd406b10d.

* Add base functionality for Bluetoot [... cut]
closed 2020-12-02 by s-martin