MiczFlor/RPi-Jukebox-RFID #1170: 🚀 | Enable Phoniebox to play a random subfolder

matthias-pelger, 2020-12-01:
## Feature Description

Scenario:
There are a lot of episodes of a certain audiobook (e.g. Benjamin Blümchen) and the user does not care which specific episode is going to be played, but all chapters of that episode need to be played in ordered mode (obviously :) ).

The files are ordered in that way:
AUDIOFOLDERSPATH/Benjamin Blümchen/Episode 1/<many>.mp3
AUDIOFOLDERSPATH/Benjamin Blümchen/Episode 2/<many>.mp3

The RFID-Tag is assigned to the top folder (in this case Benjamin Blümchen) via frontend. This would normaly lead to play all subfolders in ordered mode, which is great for most cases :)

As an inspiration/solution I provide my personalized way to do this.

My solution is to put a file named "random" with no content on the top level, making this the new file layout:
AUDIOFOLDERSPATH/Benjamin Blümchen/random
AUDIOFOLDERSPATH/Benjamin Blümchen/Episode 1/<many>.mp3
AUDIOFOLDERSPATH/Benjamin Blümchen/Episode 2/<many>.mp3

And apply this patch to the "scripts/rfid_trigger_play.sh" in line 321:
```
# Look for file "random" to determine a random subfolder to be played
    if [ -f "${AUDIOFOLDERSPATH}/${FOLDER}/random" ]
    then
        RND_FOLDER=`find "${AUDIOFOLDERSPATH}/${FOLDER}/" -type d | shuf -n 1`
        FOLDER=${RND_FOLDER#"$AUDIOFOLDERSPATH/"}
    fi

```

matthias-pelger, 2020-12-03:
Investigating further, I figured that the second swipe is not working anymore with my quick&dirty enhancement.

Is there a specific reason why the secondswipe check is based upon the " Latest_Playlist_Played" as found in "rfid_triggerplay.sh"?
```
    # Check if the second swipe happened
    # - The same playlist is cued up ("$LASTPLAYLIST" == "$PLAYLISTNAME")
    if [ "$LASTPLAYLIST" == "$PLAYLISTNAME" ]
```
First guess would be to simple avoid this and aim directly for the CARDID (adding another file to store this in the same manner as playlist)

MiczFlor, 2020-12-03:
Hi @matthias-pelger 
thanks for the input. It's a great feature - and might need a little planning. Currently the check you found is needed to actually verify the *second swipe* - meaning the swipe comes up with the same playlist to be played.
I am at work, can not look into this further now.
All the best, micz

matthias-pelger, 2020-12-03:
Hi @MiczFlor,
thanks for the hint. As the project is good structured I tried to solve this on my own. This is what I came up with:
https://github.com/MiczFlor/RPi-Jukebox-RFID/compare/develop...matthias-pelger:random-subfolder-play

This just adds stuff to enable the WebUI to show an additional button for shuffle subfolders and a debug option.
In the player part, this now uses the existing logic for folder.conf with new parameter "FOLDERSHUFFLE".

Biggest change is in file "rfid_trigger_play.sh" where I changed the logic of secondswipe check from LASTPLAYLIST to LASTID. LASTID is basically the folder from CARDID, seperating the really played folder from the one assigned to the card.
Beside this, tht LASTPLAYLIST is still there and being set. I think this will not break anything and still enable a thourough secondswipe check.

I thought on creating a pull request right away, but first I wanted some input and maybe ideas on improving things or make it more acceptable for the project.

I run this version now all evening and have no bugs found so far. Happy to hear your thoughts on this!

alwashe, 2020-12-30:
This feature would be great. Is there a way to achieve this as an amateur?

matthias-pelger, 2020-12-30:
hi @alwashe,

to be honest, I'm a newbee to git. I only know of two options:
1. make the changes you see here (https://github.com/MiczFlor/RPi-Jukebox-RFID/compare/develop...matthias-pelger:random-subfolder-play) yourself 
2. checkout the branch from my repo. But I'm a bit behind compared to the master of @MiczFlor, as it was never intended as a fork for someone else

Maybe @MiczFlor or @s-martin or someone else knows of a trick here?

biohazardxxx, 2021-03-22:
This would be great to get merged. Looking forward to get this. Currently I need to reassign RFID tags each second day to a new episode

alwashe, 2023-03-03:
I think this feature is absolutly essentlial. I can't make 500 rfid cards for every episode!? or do i miss something?

hollywoot, 2023-04-14:
I've tried to edit it manually by myself but it didnt work.
So it would be great to get a step-by-step guide. :-)

MaStFAU, 2024-12-03:
This feature makes so much sense. Will it be supported in Version 3?

s-martin, 2024-12-03:
The PR #1175 is implementing this feature, but only for V2 so far