MiczFlor/RPi-Jukebox-RFID #1171: Question: Prevent SD-Card corruption

tomraithel, 2020-12-02:
Hey!
I had my PhonieBox up and running for over a year now. Suddenly it won´t start up anymore; After checking the bootscreen and the SD card, I figured, that the SD card has been corrupted and can´t be restored anymore.

A little research told me, that this is a typical problem with SD-Cards in the raspberry PI as a lot of I/O (especially writes from the logs) affects the SD card negatively over time.

I now need to setup the PhonieBox again with a new SD-Card. In order to prevent the same scenario from happening again, I wonder if others have experienced the same problem and if there are good solutions agains SD card corruption on the PI.

I´m thankful for any tips.
Tom

s-martin, 2020-12-02:
Actually this was something I thought about a couple of days ago.

As you said it's a well known problem that many write accesses (e.g. logging) can wear out a SD card.

So far we added more logs over the time than removing logs.

I think there are a couple of settings, which should be adapted to avoid excessive logging when the box runs just fine.

E.g. the Spotify edition writes a lot of MPD status logs.

At the moment I can't tell you which settings should be adapted, but places to look are MPD or Mopidy configuration.

Please post your findings in this issue.

@MiczFlor i think it would be a good idea to have a global logging switch in the web UI to either turn on more detailed logging for debugging (see #1148) or to turn off or reduce logging to a minimum to avoid wearing out SD cards.

andreasbrett, 2020-12-07:
Better solution would be zram for the logging folders and writing back logs to the sdcard at slow intervals (writing 100kb every 5min produces less wear than writing 200b every second)

tomraithel, 2020-12-07:
Today I`ve installed zram following this (german) tutorial: https://blog.gc2.at/post/zram/
I´m not sure though, if that alone fixes the issue or if I need to configure additional stuff for the logging so that it can actually make use of zram.

Additionally I configured `/var/log` as a tmpfs and disabled access timestamps on my main partition (instructions here: https://taillieu.info/index.php/internet-of-things/raspberrypi/320-stopping-sd-card-corruption-on-raspberry-pi-s-raspbian)

These are all precautions that I collected from the internet, but I´m not really sure how to verify it improved my setup other then having my Kids using the phoniebox excessively :). Is there a good way to measure the reads/writes on the SD card?

smu, 2021-02-11:
I can recommend https://github.com/azlux/log2ram. 

log2ram takes care about the tmpfs for `/var/log`, but writes the log to disk from time to time. 

azlux also provides a Debian package, which makes it very easy to install (and deinstall).