MiczFlor/RPi-Jukebox-RFID #1172: mpd doesn't work with spotify integration

1amcord, 2020-12-03:
### What I did

I've set up a fresh RPi 3B+ with the current one-liner setup.
The initial image was a fresh raspian buster installation.

Unfortunately I didn't get it running.
Do I have to change any dependencies in mpd.service or mopidy.service?

### What happened

The mpd doesn't start at boot. Instead I got this status:
```
pi@phoniebox:~ $ systemctl status mpd.service
● mpd.service - Music Player Daemon
   Loaded: loaded (/lib/systemd/system/mpd.service; disabled; vendor preset: enabled)
   Active: inactive (dead)
     Docs: man:mpd(1)
           man:mpd.conf(5)
           file:///usr/share/doc/mpd/user-manual.html
```

### Further information that might help
A manual restart via systemctl works (except the bind to 6600-error):
```

pi@phoniebox:~ $ systemctl status mpd.service
● mpd.service - Music Player Daemon
   Loaded: loaded (/lib/systemd/system/mpd.service; disabled; vendor preset: enabled)
   Active: active (running) since Thu 2020-12-03 03:59:33 CET; 42s ago
     Docs: man:mpd(1)
           man:mpd.conf(5)
           file:///usr/share/doc/mpd/user-manual.html
 Main PID: 28061 (mpd)
    Tasks: 3 (limit: 2063)
   CGroup: /system.slice/mpd.service
           └─28061 /usr/bin/mpd --no-daemon

Dec 03 03:59:32 phoniebox systemd[1]: Starting Music Player Daemon...
Dec 03 03:59:33 phoniebox mpd[28061]: exception: bind to '127.0.0.1:6600' failed (continuing anyway, because binding to '[::1]:6600' succeeded): Failed to bind socket: Address already in
Dec 03 03:59:33 phoniebox systemd[1]: Started Music Player Daemon.
Dec 03 03:59:33 phoniebox mpd[28061]: avahi: Service 'Music Player @ phoniebox' successfully established.
```
I've tried the installer-script without spotify integration as well and there the mpd starts without any errors.

Don't know if this is relevant for this Bug, but there is another problem with mpd in syslog.

```
Dec  3 03:38:43 phoniebox systemd[1]: Started Session 1 of user pi.
Dec  3 03:38:43 phoniebox systemd[10030]: Starting Music Player Daemon...
Dec  3 03:38:46 phoniebox mpd: exception: Failed to set group 0: Operation not permitted
Dec  3 03:38:46 phoniebox systemd[10030]: mpd.service: Main process exited, code=exited, status=1/FAILURE
Dec  3 03:38:46 phoniebox systemd[10030]: mpd.service: Failed with result 'exit-code'.
Dec  3 03:38:46 phoniebox systemd[10030]: Failed to start Music Player Daemon.
Dec  3 03:38:46 phoniebox systemd[10030]: Reached target Default.
Dec  3 03:38:46 phoniebox systemd[10030]: Startup finished in 3.710s.
Dec  3 03:38:47 phoniebox systemd[1]: systemd-hostnamed.service: Succeeded.
Dec  3 03:38:52 phoniebox mopidy[490]: INFO     [SpotifyBackend-6] mopidy_spotify.playlists Refreshed 46 Spotify playlists
Dec  3 03:38:52 phoniebox mopidy[490]: INFO     [MainThread] mopidy.commands Starting Mopidy core
Dec  3 03:38:52 phoniebox mopidy[490]: INFO     [MainThread] mopidy.commands Starting Mopidy frontends: HttpFrontend, IrisFrontend, MpdFrontend
Dec  3 03:38:52 phoniebox mopidy[490]: INFO     [HttpFrontend-12] mopidy.http.actor HTTP server running at [::ffff:0.0.0.0]:6680
Dec  3 03:38:52 phoniebox mopidy[490]: INFO     [IrisFrontend-13] mopidy_iris.core Starting Iris 3.45.0
Dec  3 03:38:52 phoniebox mopidy[490]: INFO     [MainThread] mopidy_mpd.actor MPD server running at [::ffff:0.0.0.0]:6600
Dec  3 03:38:52 phoniebox mopidy[490]: INFO     [MainThread] mopidy.commands Starting GLib mainloop
Dec  3 03:38:52 phoniebox mopidy[490]: INFO     [MpdSession-15] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52228
Dec  3 03:38:52 phoniebox mopidy[490]: INFO     [MpdSession-16] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52232
Dec  3 03:38:52 phoniebox mopidy[490]: INFO     [MpdSession-17] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52236
Dec  3 03:38:52 phoniebox mopidy[490]: INFO     [MpdSession-18] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52240
Dec  3 03:38:52 phoniebox mopidy[490]: INFO     [MpdSession-19] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52244
Dec  3 03:38:52 phoniebox bash[394]: OK MPD 0.19.0
```
[... 285 lines cut ...]
Dec 03 03:42:37 phoniebox mopidy[490]: INFO     [MpdSession-250] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:53164
Dec 03 03:42:37 phoniebox mopidy[490]: INFO     [MpdSession-251] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:53168
Dec 03 03:42:37 phoniebox mopidy[490]: INFO     [MpdSession-252] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:53172
Dec 03 03:42:37 phoniebox mopidy[490]: INFO     [MpdSession-253] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:53176
Dec 03 03:42:37 phoniebox mopidy[490]: INFO     [MpdSession-254] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:53180
Dec 03 03:42:37 phoniebox mopidy[490]: INFO     [MpdSession-255] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:53184
Dec 03 03:42:37 phoniebox mopidy[490]: INFO     [MpdSession-256] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:53188
Dec 03 03:42:43 phoniebox mopidy[490]: INFO     [MpdSession-257] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:53192
Dec 03 03:42:43 phoniebox mopidy[490]: INFO     [MpdSession-258] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:53196
Dec 03 03:42:43 phoniebox mopidy[490]: INFO     [MpdSession-259] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:53200
Dec 03 03:42:43 phoniebox mopidy[490]: INFO     [MpdSession-260] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:53204
Dec 03 03:42:43 phoniebox mopidy[490]: INFO     [MpdSession-261] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:53208
Dec 03 03:42:43 phoniebox mopidy[490]: INFO     [MpdSession-262] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:53212
Dec 03 03:42:43 phoniebox mopidy[490]: INFO     [MpdSession-263] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:53216
Dec 03 03:42:43 phoniebox mopidy[490]: INFO     [MpdSession-264] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:53220
Dec 03 03:42:48 phoniebox mopidy[490]: INFO     [MpdSession-265] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:53224
Dec 03 03:42:48 phoniebox mopidy[490]: INFO     [MpdSession-266] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:53228
Dec 03 03:42:48 phoniebox mopidy[490]: INFO     [MpdSession-267] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:53232
Dec 03 03:42:48 phoniebox mopidy[490]: INFO     [MpdSession-268] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:53236
Dec 03 03:42:48 phoniebox mopidy[490]: INFO     [MpdSession-269] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:53240
Dec 03 03:42:49 phoniebox mopidy[490]: INFO     [MpdSession-270] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:53244
Dec 03 03:42:49 phoniebox mopidy[490]: INFO     [MpdSession-271] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:53248
Dec 03 03:42:49 phoniebox mopidy[490]: INFO     [MpdSession-272] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:53252
pi@phoniebox:~ $ 
```

s-martin, 2020-12-03:
That's intended. Spotify uses Mopidy and not MPD.

The other error we need to look into.

1amcord, 2020-12-03:
Thank you for clarification, @s-martin 

I've tried to upload another file and got this message in syslog:
```

Dec  3 07:05:24 phoniebox lighttpd[838]: chown: changing ownership of '/home/pi/RPi-Jukebox-RFID/shared/audiofolders/up': Operation not permitted
Dec  3 07:05:36 phoniebox lighttpd[838]: chown: changing ownership of '/home/pi/RPi-Jukebox-RFID/shared/audiofolders/up/12 Wenn Der Hausmeister Kommt.mp3': Operation not permitted
```

Any ideas how this can be solved? Do I have to change file/folder rights?

This is how it looks like after folder creation/file upload via browser:

```
pi@phoniebox:~/RPi-Jukebox-RFID/shared $ ls -al audiofolders/
total 16
drwxrwxrwx 3 pi       www-data 4096 Dec  3 07:05 .
drwxrwxr-x 4 pi       www-data 4096 Dec  3 06:59 ..
-rwxrwxrwx 1 pi       www-data  111 Dec  3 07:05 folder.conf
-rwxrwxrwx 1 pi       www-data    0 Dec  3 06:57 placeholder
drwxrwxrwx 2 www-data www-data 4096 Dec  3 07:05 up
```
```
pi@phoniebox:~/RPi-Jukebox-RFID/shared $ ls -al audiofolders/
total 16
drwxrwxrwx 3 pi       www-data 4096 Dec  3 07:05 .
drwxrwxr-x 4 pi       www-data 4096 Dec  3 06:59 ..
-rwxrwxrwx 1 pi       www-data  111 Dec  3 07:05 folder.conf
-rwxrwxrwx 1 pi       www-data    0 Dec  3 06:57 placeholder
drwxrwxrwx 2 www-data www-data 4096 Dec  3 07:05 up
```

1amcord, 2020-12-03:
Hmmh, there are more errors related to mopidy, lighthttpd and mpd at the same time, but I don't know what to do/where to look:

```
Dec  3 07:05:01 phoniebox mopidy[516]: INFO     [MpdSession-136] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50114
Dec  3 07:05:01 phoniebox mopidy[516]: INFO     [MpdSession-137] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50118
Dec  3 07:05:24 phoniebox lighttpd[838]: chown: changing ownership of '/home/pi/RPi-Jukebox-RFID/shared/audiofolders/up': Operation not permitted
Dec  3 07:05:36 phoniebox lighttpd[838]: chown: changing ownership of '/home/pi/RPi-Jukebox-RFID/shared/audiofolders/up/12 Wenn Der Hausmeister Kommt.mp3': Operation not permitted
Dec  3 07:05:40 phoniebox mopidy[516]: INFO     [MpdSession-138] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50122
Dec  3 07:05:40 phoniebox mopidy[516]: INFO     [MpdSession-139] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50126
Dec  3 07:05:40 phoniebox mopidy[516]: INFO     [MpdSession-140] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50130
Dec  3 07:05:41 phoniebox mopidy[516]: INFO     [MpdSession-141] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50134
Dec  3 07:05:41 phoniebox mopidy[516]: INFO     [MpdSession-142] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50138
Dec  3 07:05:41 phoniebox mopidy[516]: INFO     [MainThread] mopidy.commands GLib mainloop got SIGTERM. Exiting...
Dec  3 07:05:41 phoniebox mopidy[516]: INFO     [MainThread] mopidy.commands Stopping Mopidy frontends
Dec  3 07:05:41 phoniebox systemd[1]: Stopping Mopidy music server...
Dec  3 07:05:41 phoniebox mopidy[516]: INFO     [IrisFrontend-13] mopidy_iris.core Stopping Iris
Dec  3 07:05:41 phoniebox mopidy[516]: INFO     [MainThread] mopidy.commands Stopping Mopidy core
Dec  3 07:05:41 phoniebox mopidy[516]: INFO     [MainThread] mopidy.commands Stopping Mopidy backends
Dec  3 07:05:41 phoniebox mopidy[516]: INFO     [MainThread] mopidy.commands Stopping Mopidy audio
Dec  3 07:05:41 phoniebox mopidy[516]: INFO     [MainThread] mopidy.commands Stopping Mopidy mixer
Dec  3 07:05:42 phoniebox systemd[1]: mopidy.service: Succeeded.
Dec  3 07:05:42 phoniebox systemd[1]: Stopped Mopidy music server.
Dec  3 07:05:42 phoniebox systemd[1]: Created slice User Slice of UID 112.
Dec  3 07:05:42 phoniebox systemd[1]: Starting User Runtime Directory /run/user/112...
Dec  3 07:05:42 phoniebox systemd[1]: Started User Runtime Directory /run/user/112.
Dec  3 07:05:42 phoniebox systemd[1]: Starting User Manager for UID 112...
Dec  3 07:05:42 phoniebox systemd[14362]: Listening on GnuPG network certificate management daemon.
Dec  3 07:05:42 phoniebox systemd[14362]: Reached target Timers.
Dec  3 07:05:42 phoniebox systemd[14362]: Listening on GnuPG cryptographic agent and passphrase cache.
Dec  3 07:05:42 phoniebox systemd[14362]: Listening on GnuPG cryptographic agent and passphrase cache (access for web browsers).
Dec  3 07:05:42 phoniebox systemd[14362]: Listening on GnuPG cryptographic agent (ssh-agent emulation).
Dec  3 07:05:42 phoniebox systemd[14362]: Starting D-Bus User Message Bus Socket.
Dec  3 07:05:42 phoniebox systemd[14362]: Listening on GnuPG cryptographic agent and passphrase cache (restricted).
Dec  3 07:05:42 phoniebox systemd[14362]: Reached target Paths.
Dec  3 07:05:42 phoniebox systemd[14362]: Listening on D-Bus User Message Bus Socket.
Dec  3 07:05:42 phoniebox systemd[14362]: Reached target Sockets.
Dec  3 07:05:42 phoniebox systemd[14362]: Reached target Basic System.
Dec  3 07:05:42 phoniebox systemd[1]: Started User Manager for UID 112.
Dec  3 07:05:42 phoniebox systemd[14362]: Starting Music Player Daemon...
Dec  3 07:05:42 phoniebox systemd[1]: Started Session c1 of user mopidy.
Dec  3 07:05:43 phoniebox lighttpd[838]: echo: write error: Broken pipe
Dec  3 07:05:43 phoniebox mpd: exception: Failed to set group 0: Operation not permitted
Dec  3 07:05:43 phoniebox systemd[14362]: mpd.service: Main process exited, code=exited, status=1/FAILURE
Dec  3 07:05:43 phoniebox systemd[14362]: mpd.service: Failed with result 'exit-code'.
Dec  3 07:05:43 phoniebox systemd[14362]: Failed to start Music Player Daemon.
Dec  3 07:05:43 phoniebox systemd[14362]: Reached target Default.
Dec  3 07:05:43 phoniebox systemd[14362]: Startup finished in 1.403s.
Dec  3 07:05:45 phoniebox systemd[14362]: Started D-Bus User Message Bus.
Dec  3 07:05:46 phoniebox systemd[1]: session-c1.scope: Succeeded.
Dec  3 07:05:47 phoniebox systemd[1]: Starting Mopidy music server...
Dec  3 07:05:47 phoniebox systemd[1]: Started Mopidy music server.
Dec  3 07:05:48 phoniebox lighttpd[838]: echo: write error: Broken pipe
Dec  3 07:05:48 phoniebox lighttpd[838]: mpd error: Connection refused
Dec  3 07:05:49 phoniebox lighttpd[838]: mpd error: Connection refused
Dec  3 07:05:49 phoniebox lighttpd[838]: mpd error: Connection refused
Dec  3 07:05:49 phoniebox lighttpd[838]: mpd error: Connection refused
Dec  3 07:05:49 phoniebox lighttpd[838]: mpd error: Connection refused
Dec  3 07:05:49 phoniebox lighttpd[838]: echo: write error: Broken pipe
Dec  3 07:05:49 phoniebox mopidy[14402]: INFO     [MainThread] mopidy.__main__ Starting Mopidy 3.0.2
Dec  3 07:05:49 phoniebox mopidy[14402]: INFO     [MainThread] mopidy.config Loading config from builtin defaults
[... cut ...]
Dec  3 07:05:57 phoniebox systemd[1]: Stopping User Runtime Directory /run/user/112...
Dec  3 07:05:57 phoniebox systemd[13660]: run-user-112.mount: Succeeded.
Dec  3 07:05:57 phoniebox systemd[1]: run-user-112.mount: Succeeded.
Dec  3 07:05:57 phoniebox systemd[1]: user-runtime-dir@112.service: Succeeded.
Dec  3 07:05:57 phoniebox systemd[1]: Stopped User Runtime Directory /run/user/112.
Dec  3 07:05:57 phoniebox systemd[1]: Removed slice User Slice of UID 112.
Dec  3 07:05:59 phoniebox lighttpd[838]: echo: write error: Broken pipe

```

s-martin, 2020-12-05:
Please check https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#i-added-new-folders-with-audiofiles-but-the-playout-does-not-work

s-martin, 2020-12-15:
@1amcord  could you check, if the following fixes your issue:

1. Replace content of requirements-spotify.txt with `Mopidy-Iris==3.54.2`
2. `sudo pip install --upgrade --force-reinstall -r requirements-spotify.txt`
3. Reboot

1amcord, 2020-12-15:
@s-martin thanks a lot that you haven't forgotten me.

I'm a bit ashamed but I switched to the non-spotify version and I guess we have no use for spotify at the moment.
Also the kids and I are glad that the box is nearly complete and I don't want to change anything because it is under heavy usage.
That is why I am unable to test your suggestions.

I hope you understand...

closed 2020-12-15 by 1amcord

s-martin, 2020-12-15:
No problem.

enjoy the box!

danielsmint, 2020-12-18:
I checked it:

I had similar problems, installed Phoniebox +Spotify from https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/INSTALL-stretch#raspbian-buster-recommended-version (One Line Install Stable Version, on a fresh RPi3+ running Buster) and i couldnt play music on Phoniebox UI (per MPD) or IRIS (Mopidy). I could't add Spotify Songs to my Playlist.

After that i installed Phoniebox Classic (MPD) and installed Spotify (and IRIS) by myself. Now i am able to play local files in th Phoniebox UI and IRIS (since MPD is included in Mobipy).

Finaly, i added Mopidy-Iris==3.54.2 to the requirements-spotify.txt and with `sudo pip3 install --upgrade --force-reinstall -r requirements-spotify.txt` (pip3!) it finaly worked! 

I am now able to add Spotify Playlists to my Phoniebox and i can play them.

Thank you for that hint!