MiczFlor/RPi-Jukebox-RFID #1173: Can't playback folders but single files

thomaswasilewski, 2020-12-04:
## Bug

Can't playback Folders

### What I did

I installed the raspberry pi with above mentioned buster image and ran the installer script

### What happened

Inside the Webinterface: When I press play on a specific file, the file plays ->  (Green Bar with File is playing; Information under the Controlls, I can hear the file playing)

When I press play on a Folder (with for examples 10 mp3s), the Green Bar says that it is playing, but no Information is displayed unter the Controlls and I can't hear nothing

(Specific example) Folder ZZZ/Counting the file  number00.mp3 is playing by pressing play next to the file, but the Folder ZZZ/Counting is not playing by pressing play next to the folder

-> Since I wanted to assign my RFID Cards to Folders, as long as folders are not playing, I can't use phoniebox.

Sidenote: ZZZ/Podcast-DLF-Kinderhoerspiele  and ZZZ/This American Life Podcast work with the playbutton and also when I assign an RFID Chip to them

also Sidenote: Spotify does not work, but that is a problem for another time

and final Sidenote: I can't remember why, but I changed the Port of mpd from 6600 to 6603 because the mpd service wouldn't start with 6600. Maybe this could be the Problem?

### I expected this to happen

To Play a Folder oder Playlist

### Further information that might help

Dec  4 00:35:24 MonieBox kernel: [    4.946605] usb 1-1.1.1: new high-speed USB device number 5 using dwc_otg
Dec  4 00:35:24 MonieBox kernel: [    4.957127] hid-generic 0003:0D8C:0014.0001: input,hidraw0: USB HID v1.00 Device [C-Media Electronics Inc. USB Audio Device] on usb-3f980000.usb-1.3/input3
Dec  4 00:35:24 MonieBox systemd[1]: Started File System Check Daemon to report status.
Dec  4 00:35:24 MonieBox systemd-fsck[131]: e2fsck 1.44.5 (15-Dec-2018)
Dec  4 00:35:24 MonieBox systemd[1]: Started udev Coldplug all Devices.
Dec  4 00:35:24 MonieBox kernel: [    5.097162] usb 1-1.1.1: New USB device found, idVendor=0424, idProduct=7800, bcdDevice= 3.00
Dec  4 00:35:24 MonieBox systemd[1]: Starting Helper to synchronize boot up for ifupdown...
Dec  4 00:35:24 MonieBox kernel: [    5.106372] usb 1-1.1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
Dec  4 00:35:24 MonieBox kernel: [    5.208988] random: systemd: uninitialized urandom read (16 bytes read)
Dec  4 00:35:24 MonieBox kernel: [    5.234118] random: systemd: uninitialized urandom read (16 bytes read)
Dec  4 00:35:24 MonieBox kernel: [    5.266930] random: systemd: uninitialized urandom read (16 bytes read)
Dec  4 00:35:24 MonieBox systemd[1]: Started Helper to synchronize boot up for ifupdown.
Dec  4 00:35:24 MonieBox kernel: [    5.381809] lan78xx 1-1.1.1:1.0 (unnamed net_device) (uninitialized): No External EEPROM. Setting MAC Speed
Dec  4 00:35:24 MonieBox kernel: [    5.383383] libphy: lan78xx-mdiobus: probed
Dec  4 00:35:24 MonieBox kernel: [    5.448295] lan78xx 1-1.1.1:1.0 (unnamed net_device) (uninitialized): int urb period 64
Dec  4 00:35:24 MonieBox kernel: [    5.478173] i2c /dev entries driver
Dec  4 00:35:24 MonieBox kernel: [    6.822168] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
Dec  4 00:35:24 MonieBox systemd-fsck[131]: rootfs: clean, 115143/7616000 files, 1119285/31194112 blocks
Dec  4 00:35:24 MonieBox kernel: [    7.539255] random: crng init done
Dec  4 00:35:24 MonieBox kernel: [    7.539270] random: 7 urandom warning(s) missed due to ratelimiting
Dec  4 00:35:24 MonieBox kernel: [    7.564558] vc_sm_cma: module is from the staging directory, the quality is unknown, you have been warned.
Dec  4 00:35:24 MonieBox systemd[1]: Started File System Check on Root Device.
Dec  4 00:35:24 MonieBox systemd[1]: Starting Remount Root and Kernel File Systems...
Dec  4 00:35:24 MonieBox systemd[1]: Started Remount Root and Kernel File Systems.
Dec  4 00:35:24 MonieBox systemd[1]: Starting Load/Save Random Seed...
Dec  4 00:35:24 MonieBox systemd[1]: Condition check resulted in Rebuild Hardware Database being skipped.
Dec  4 00:35:24 MonieBox systemd[1]: Starting Flush Journal to Persistent Storage...
Dec  4 00:35:24 MonieBox systemd[1]: Starting Create System Users...
Dec  4 00:35:24 MonieBox systemd[1]: Started Load/Save Random Seed.
Dec  4 00:35:24 MonieBox systemd[1]: Started Flush Journal to Persistent Storage.
Dec  4 00:35:24 MonieBox systemd[1]: Started Create System Users.
Dec  4 00:35:24 MonieBox systemd[1]: Starting Create Static Device Nodes in /dev...
Dec  4 00:35:24 MonieBox systemd[1]: Started Create Static Device Nodes in /dev.
Dec  4 00:35:24 MonieBox systemd[1]: Starting udev Kernel Device Manager...
Dec  4 00:35:24 MonieBox systemd[1]: Reached target Local File Systems (Pre).
Dec  4 00:35:24 MonieBox systemd[1]: Started udev Kernel Device Manager.
Dec  4 00:35:24 MonieBox systemd[1]: Found device /dev/serial1.
Dec  4 00:35:24 MonieBox mtp-probe: checking bus 1, device 4: "/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3"
Dec  4 00:35:24 MonieBox mtp-probe: bus: 1, device: 4 was not an MTP device
Dec  4 00:35:24 MonieBox systemd[1]: Found device /dev/disk/by-partuuid/05c3a4da-01.
[... 464 lines cut ...]
master

### Installscript

the normal one

## Hardware

### RaspberryPi version

3 B+

### RFID Reader

PN532

### Soundcard

Bus 001 Device 004: ID 0d8c:0014 C-Media Electronics, Inc. Audio Adapter (Unitek Y-247A)

### Other notable hardware

no

Thank you for your help!

matthias-pelger, 2020-12-04:
hi @thomaswasilewski ,
just a guess, but did you scan the local files before trying to play them?
I had a simillar issue, when the scan didn't finish in the first try and I was only able to play the albums that where scanned, but not the ones missing, even when all where showed in the WebUI but not in the Mopidy Interface (the Spotify tab). A rescan solved it (In Files and Folders).

btw.:
> and final Sidenote: I can't remember why, but I changed the Port of mpd from 6600 to 6603 because the mpd service wouldn't start with 6600. Maybe this could be the Problem?
MPD is not needed as Mopidy is doing all the work

Cheers

s-martin, 2020-12-05:
@thomaswasilewski, please try to set the port of MPD back to 6600 and test again. I’m not sure, if there are side effects, mopidy also utilizes MPD.

s-martin, 2022-12-27:
Can you check, if the issue still persists with current 2.4 version?

closed 2022-12-27 by s-martin