MiczFlor/RPi-Jukebox-RFID #1176: On phoniebox 2.2 +spotify the Volumcontroll dose not work 🐛 | BUG SUMMARY

Allstar1000GIT, 2020-12-06:
## Bug

If I install the phoniebox with spotify the player dose not play any audiofiles.
there is a pop up in green "songXY" is playing but the section on the top where it normaly shows the actuell playing song stays empty (marked it in red).

The classic instalation works without problems.
 
so i guess it dosen´t really plays anything.
[image]

[image]

### What I did

I installed the phoniebox 2.2 on an headless buster image with ssh. (one line installer)

s-martin, 2020-12-06:
@Allstar1000GIT can you post the results of `journalctl -u mopidy`?

Allstar1000GIT, 2020-12-06:
I try too post it this evening

ozke86, 2020-12-06:
Bei mir wird in der Classic Lautstärke [%] angezeigt 
Wenn ich einen HDMI Monitor anschließen wird wider 100% angezeigt und der Ton kommt über Klinke !

s-martin, 2020-12-06:
> Bei mir wird in der Classic Lautstärke [%] angezeigt !
> Wenn ich einen HDMI Monitor anschließen wird wieder 100% angezeigt und der Ton kommt über Klinke !

Could you please check, if the solution in https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#raspberry-pi-update-may-2020-broke-my-audio works for you?

ozke86, 2020-12-06:
Hat funktioniert habe auf 0 gesetzt

Allstar1000GIT, 2020-12-06:
Here is the log from  `journalctl -u mopidy`

[image]
[image]
[image]

Allstar1000GIT, 2020-12-06:
I am really sure ther is something wrong with the spotify version because i can play songs from spotify with the phoniebox without any problems, but it dosent play any files stored on direktly on the pi or from youtube

Allstar1000GIT, 2020-12-06:
I am not sure if it helps but I installed a buster version with gui and tried to play the .m3u and VLC brings an error „not able to play this file“ if I go to the mp3 file itself VLC plays it without any problems

s-martin, 2020-12-06:
Could you also post the results of `tail -n 500 /var/log/syslog`?

s-martin, 2020-12-06:
> I am not sure if it helps but I installed a buster version with gui and tried to play the .m3u file and VLC brings an error „not able to play this file“ if I go to the mp3 file itself VLC plays it without any problems

Just to clarify, does playing mp3 files or m3u playlists or both not work?

Allstar1000GIT, 2020-12-06:
m3u dosent work
mp3 itself worked

Allstar1000GIT, 2020-12-06:
Here ist the log from: tail -n 500 /var/log/syslog

Dec  6 19:55:46 raspberrypi mopidy[3737]: INFO     [MpdSession-8853] mopidy_mpd.                                                                             session New MPD connection from [::ffff:127.0.0.1]:41332
Dec  6 19:55:46 raspberrypi mopidy[3737]: INFO     [MpdSession-8854] mopidy_mpd.                                                                             session New MPD connection from [::ffff:127.0.0.1]:41336
Dec  6 19:55:46 raspberrypi mopidy[3737]: INFO     [MpdSession-8855] mopidy_mpd.                                                                             session New MPD connection from [::ffff:127.0.0.1]:41340
Dec  6 19:55:46 raspberrypi mopidy[3737]: INFO     [MpdSession-8856] mopidy_mpd.                                                                             session New MPD connection from [::ffff:127.0.0.1]:41344
Dec  6 19:55:46 raspberrypi mopidy[3737]: INFO     [MpdSession-8857] mopidy_mpd.                                                                             session New MPD connection from [::ffff:127.0.0.1]:41348
Dec  6 19:55:51 raspberrypi mopidy[3737]: INFO     [MpdSession-8858] mopidy_mpd.                                                                             session New MPD connection from [::ffff:127.0.0.1]:41352
Dec  6 19:55:51 raspberrypi mopidy[3737]: INFO     [MpdSession-8859] mopidy_mpd.                                                                             session New MPD connection from [::ffff:127.0.0.1]:41356
Dec  6 19:55:51 raspberrypi mopidy[3737]: INFO     [MpdSession-8860] mopidy_mpd.                                                                             session New MPD connection from [::ffff:127.0.0.1]:41360
Dec  6 19:55:51 raspberrypi mopidy[3737]: INFO     [MpdSession-8861] mopidy_mpd.                                                                             session New MPD connection from [::ffff:127.0.0.1]:41364
Dec  6 19:55:51 raspberrypi mopidy[3737]: INFO     [MpdSession-8862] mopidy_mpd.                                                                             session New MPD connection from [::ffff:127.0.0.1]:41368
Dec  6 19:55:51 raspberrypi mopidy[3737]: INFO     [MpdSession-8863] mopidy_mpd.                                                                             session New MPD connection from [::ffff:127.0.0.1]:41372
Dec  6 19:55:52 raspberrypi mopidy[3737]: INFO     [MpdSession-8864] mopidy_mpd.                                                                             session New MPD connection from [::ffff:127.0.0.1]:41376
Dec  6 19:55:52 raspberrypi mopidy[3737]: INFO     [MpdSession-8865] mopidy_mpd.                                                                             session New MPD connection from [::ffff:127.0.0.1]:41380
Dec  6 19:55:57 raspberrypi mopidy[3737]: INFO     [MpdSession-8866] mopidy_mpd.                                                                             session New MPD connection from [::ffff:127.0.0.1]:41384
Dec  6 19:55:57 raspberrypi mopidy[3737]: INFO     [MpdSession-8867] mopidy_mpd.                                                                             session New MPD connection from [::ffff:127.0.0.1]:41388
Dec  6 19:55:57 raspberrypi mopidy[3737]: INFO     [MpdSession-8868] mopidy_mpd.                                                                             session New MPD connection from [::ffff:127.0.0.1]:41392
Dec  6 19:55:57 raspberrypi mopidy[3737]: INFO     [MpdSession-8869] mopidy_mpd.                                                                             session New MPD connection from [::ffff:127.0.0.1]:41396
Dec  6 19:55:57 raspberrypi mopidy[3737]: INFO     [MpdSession-8870] mopidy_mpd.                                                                             session New MPD connection from [::ffff:127.0.0.1]:41400
Dec  6 19:55:57 raspberrypi mopidy[3737]: INFO     [MpdSession-8871] mopidy_mpd.                                                                             session New MPD connection from [::ffff:127.0.0.1]:41404
Dec  6 19:55:57 raspberrypi mopidy[3737]: INFO     [MpdSession-8872] mopidy_mpd.                                                                             session New MPD connection from [::ffff:127.0.0.1]:41408
Dec  6 19:55:57 raspberrypi mopidy[3737]: INFO     [MpdSession-8873] mopidy_mpd.                                                                             session New MPD connection from [::ffff:127.0.0.1]:41412
Dec  6 19:56:02 raspberrypi mopidy[3737]: INFO     [MpdSession-8874] mopidy_mpd.                                                                             session New MPD connection from [::ffff:127.0.0.1]:41416
Dec  6 19:56:02 raspberrypi mopidy[3737]: INFO     [MpdSession-8875] mopidy_mpd.                                                                             session New MPD connection from [::ffff:127.0.0.1]:41420
Dec  6 19:56:02 raspberrypi mopidy[3737]: INFO     [MpdSession-8876] mopidy_mpd.                                                                             session New MPD connection from [::ffff:127.0.0.1]:41424
Dec  6 19:56:02 raspberrypi mopidy[3737]: INFO     [MpdSession-8877] mopidy_mpd.                                                                             session New MPD connection from [::ffff:127.0.0.1]:41428
[... cut ...]
Dec  6 20:00:29 raspberrypi mopidy[3737]: INFO     [MpdSession-9253] mopidy_mpd.                                                                             session New MPD connection from [::ffff:127.0.0.1]:42932
Dec  6 20:00:29 raspberrypi mopidy[3737]: INFO     [MpdSession-9254] mopidy_mpd.                                                                             session New MPD connection from [::ffff:127.0.0.1]:42936
Dec  6 20:00:29 raspberrypi mopidy[3737]: INFO     [MpdSession-9255] mopidy_mpd.                                                                             session New MPD connection from [::ffff:127.0.0.1]:42940

s-martin, 2020-12-06:
Ok, so regular local mp3s are played, but m3u Playlists are not.

Are the m3u playlists located in`/home/pi/RPi-Jukebox-RFID/playlists`? If not, do the m3u playlists work, if you put them   in this directory?

Allstar1000GIT, 2020-12-06:
I opend the m3u with an editor and ther was nothing in it, so I went to the web interface and hit  the scan database button under folder and files. So now it is working. THX for your fast help

s-martin, 2020-12-06:
If this works for you, please close this issue.

closed 2020-12-06 by Allstar1000GIT