MiczFlor/RPi-Jukebox-RFID #1178: 🐛 | Startup sound did not work

scharc, 2020-12-06:
## Bug

### What I did

Upgraded from 2.0 to 2.2 with the one line installer

### What happened

startup sound stopped working

### I expected this to happen

startup sound working 

### Further information that might help

syslog:

Dec  6 22:32:10 phoniebox bash[952]: High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
Dec  6 22:32:10 phoniebox bash[952]: #011version 1.25.10; written and copyright by Michael Hipp and others
Dec  6 22:32:10 phoniebox bash[952]: #011free software (LGPL) without any warranty but with best wishes
Dec  6 22:32:10 phoniebox bash[952]: Cannot connect to server socket err = No such file or directory
Dec  6 22:32:10 phoniebox bash[952]: Cannot connect to server request channel
Dec  6 22:32:10 phoniebox bash[952]: jack server is not running or cannot be started
Dec  6 22:32:10 phoniebox bash[952]: JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Dec  6 22:32:10 phoniebox bash[952]: JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Dec  6 22:32:10 phoniebox bash[952]: /home/pi/RPi-Jukebox-RFID/scripts/startup-scripts.sh: line 42:  1054 Segmentation fault      /usr/bin/mpg123 -f -${mpgvolume} /home/pi/RPi-Jukebox-RFID/shared/startupsound.mp3
Dec  6 22:32:10 phoniebox bash[952]: Updating DB (#1) ...
Dec  6 22:32:10 phoniebox bash[952]: volume: 15%   repeat: off   random: off   single: off   consume: off

I run the startup script manually from the command line and got the same error...

When I run the script as root it works, so my fix is just prefixing the mpg123 line with sudo in startup-scripts.sh

I can give more details if needed.

fatdreads, 2020-12-07:
maybe this helps:
https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/1167

s-martin, 2020-12-07:
Can you provide more of syslog before that snippet? I’m missing a couple of log entries from` startup-script.sh` here.
Looks like the variable `{mpgvolume}` was not expanded.

scharc, 2020-12-07:
Sure: This is the second boot. In my first bug report I copied a log file where I forgot to raise the startvolume from 0 to something useful. On this boot I corrected my failure:

`Dec  6 22:39:11 phoniebox kernel: [    0.000000] Booting Linux on physical CPU 0x0
Dec  6 22:39:11 phoniebox kernel: [    0.000000] Linux version 5.4.79-v7+ (dom@buildbot) (gcc version 8.4.0 (Ubuntu/Linaro 8.4.0-3ubuntu1)) #1373 SMP Mon Nov 23 13:22:33 GMT 2020
Dec  6 22:39:11 phoniebox kernel: [    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
Dec  6 22:39:11 phoniebox kernel: [    0.000000] CPU: div instructions available: patching division code
Dec  6 22:39:11 phoniebox kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
Dec  6 22:39:11 phoniebox kernel: [    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
Dec  6 22:39:11 phoniebox kernel: [    0.000000] Memory policy: Data cache writealloc
Dec  6 22:39:11 phoniebox kernel: [    0.000000] Reserved memory: created CMA memory pool at 0x37400000, size 64 MiB
Dec  6 22:39:11 phoniebox kernel: [    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
Dec  6 22:39:11 phoniebox kernel: [    0.000000] On node 0 totalpages: 242688
Dec  6 22:39:11 phoniebox kernel: [    0.000000]   Normal zone: 2133 pages used for memmap
Dec  6 22:39:11 phoniebox kernel: [    0.000000]   Normal zone: 0 pages reserved
Dec  6 22:39:11 phoniebox kernel: [    0.000000]   Normal zone: 242688 pages, LIFO batch:63
Dec  6 22:39:11 phoniebox kernel: [    0.000000] percpu: Embedded 20 pages/cpu s49740 r8192 d23988 u81920
Dec  6 22:39:11 phoniebox kernel: [    0.000000] pcpu-alloc: s49740 r8192 d23988 u81920 alloc=20*4096
Dec  6 22:39:11 phoniebox kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
Dec  6 22:39:11 phoniebox kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 240555
Dec  6 22:39:11 phoniebox kernel: [    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=7d4d7400-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
Dec  6 22:39:11 phoniebox kernel: [    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
Dec  6 22:39:11 phoniebox kernel: [    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
Dec  6 22:39:11 phoniebox kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Dec  6 22:39:11 phoniebox kernel: [    0.000000] Memory: 880476K/970752K available (9216K kernel code, 698K rwdata, 2608K rodata, 1024K init, 827K bss, 24740K reserved, 65536K cma-reserved)
Dec  6 22:39:11 phoniebox kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Dec  6 22:39:11 phoniebox kernel: [    0.000000] ftrace: allocating 29201 entries in 58 pages
Dec  6 22:39:11 phoniebox kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Dec  6 22:39:11 phoniebox kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Dec  6 22:39:11 phoniebox kernel: [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Dec  6 22:39:11 phoniebox kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x324/0x4f8 with crng_init=0
Dec  6 22:39:11 phoniebox kernel: [    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
Dec  6 22:39:11 phoniebox kernel: [    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
Dec  6 22:39:11 phoniebox kernel: [    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
Dec  6 22:39:11 phoniebox kernel: [    0.000022] Switching to timer-based delay loop, resolution 52ns
Dec  6 22:39:11 phoniebox kernel: [    0.000287] Console: colour dummy device 80x30
Dec  6 22:39:11 phoniebox kernel: [    0.000881] printk: console [tty1] enabled
Dec  6 22:39:11 phoniebox kernel: [    0.000946] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
Dec  6 22:39:11 phoniebox kernel: [    0.000994] pid_max: default: 32768 minimum: 301
Dec  6 22:39:11 phoniebox kernel: [    0.001202] LSM: Security Framework initializing
Dec  6 22:39:11 phoniebox kernel: [    0.001438] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
Dec  6 22:39:11 phoniebox kernel: [    0.001484] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
Dec  6 22:39:11 phoniebox kernel: [    0.002952] Disabling memory control group subsystem
Dec  6 22:39:11 phoniebox kernel: [    0.003092] CPU: Testing write buffer coherency: ok
Dec  6 22:39:11 phoniebox kernel: [    0.003679] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
Dec  6 22:39:11 phoniebox kernel: [    0.004640] Setting up static identity map for 0x100000 - 0x10003c
Dec  6 22:39:11 phoniebox kernel: [    0.004858] rcu: Hierarchical SRCU implementation.
Dec  6 22:39:11 phoniebox kernel: [    0.005554] smp: Bringing up secondary CPUs ...
Dec  6 22:39:11 phoniebox kernel: [    0.006699] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[... cut ...]
Dec  6 22:40:17 phoniebox systemd-timesyncd[308]: Synchronized to time server for the first time 79.133.44.136:123 (2.debian.pool.ntp.org).
Dec  6 22:40:25 phoniebox systemd[1]: systemd-hostnamed.service: Succeeded.
Dec  6 22:41:02 phoniebox idle-watchdog.sh[576]: warning: commands will be executed using /bin/sh
Dec  6 22:41:02 phoniebox idle-watchdog.sh[576]: job 63 at Sun Dec  6 22:56:00 2020`

Just to clear things up: When I run the startup-script.sh from the commandline I get the same error if I am not running as root

s-martin, 2022-12-20:
Closing as there was no activity for a very long time.

Please check the current release 2.4 or our new implementation 3.2, if this issue still persists.

Feel free to open a new issue, if necessary.

closed 2022-12-20 by s-martin

dvalob, 2024-03-08:
Hi! I am having the same issue using the 2.5 version. I have also the segmentation fault when calling the startup script when trying to play the startup sound. Please find here my logs (volume is correctly set to 25 as per previous releases).

"
Mar  8 20:43:56 raspberrypi bash[913]: /home/pi/RPi-Jukebox-RFID/scripts/startup-scripts.sh: line 53:  1032 **Segmentation fault**      /usr/bin/mpg123 -f -${mpgvolume} /home/pi/RPi-Jukebox-RFID/shared/startupsound.mp3
Mar  8 20:43:56 raspberrypi bash[913]: Updating DB (#1) ...
Mar  8 20:43:56 raspberrypi bash[913]: volume: 25%   repeat: off   random: off   single: off   consume: off_
"

Full log:
ar  8 20:43:42 raspberrypi fake-hwclock[116]: Fri  8 Mar 20:43:38 UTC 2024
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] Booting Linux on physical CPU 0x0
Mar  8 20:43:42 raspberrypi systemd-fsck[135]: e2fsck 1.44.5 (15-Dec-2018)
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] Linux version 5.10.103-v7+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1529 SMP Tue Mar 8 12:21:37 GMT 2022
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] CPU: div instructions available: patching division code
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Rev 1.2
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] random: fast init done
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] Memory policy: Data cache writealloc
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] Reserved memory: created CMA memory pool at 0x37400000, size 64 MiB
Mar  8 20:43:42 raspberrypi systemd[1]: Started udev Coldplug all Devices.
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] Zone ranges:
Mar  8 20:43:42 raspberrypi kernel: [    0.000000]   DMA      [mem 0x0000000000000000-0x000000003b3fffff]
Mar  8 20:43:42 raspberrypi kernel: [    0.000000]   Normal   empty
Mar  8 20:43:42 raspberrypi systemd[1]: Starting Helper to synchronize boot up for ifupdown...
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] Movable zone start for each node
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] Early memory node ranges
Mar  8 20:43:42 raspberrypi kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000003b3fffff]
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] On node 0 totalpages: 242688
Mar  8 20:43:42 raspberrypi kernel: [    0.000000]   DMA zone: 2133 pages used for memmap
Mar  8 20:43:42 raspberrypi systemd[1]: Started Helper to synchronize boot up for ifupdown.
Mar  8 20:43:42 raspberrypi kernel: [    0.000000]   DMA zone: 0 pages reserved
Mar  8 20:43:42 raspberrypi kernel: [    0.000000]   DMA zone: 242688 pages, LIFO batch:63
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] percpu: Embedded 20 pages/cpu s50828 r8192 d22900 u81920
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] pcpu-alloc: s50828 r8192 d22900 u81920 alloc=20*4096
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 240555
Mar  8 20:43:42 raspberrypi systemd-fsck[135]: rootfs: clean, 114270/1797408 files, 943027/7748608 blocks
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=2ad10d96-02 rootfstype=ext4 fsck.repair=yes rootwait
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Mar  8 20:43:42 raspberrypi systemd[1]: Started File System Check on Root Device.
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] Memory: 878804K/970752K available (10240K kernel code, 1312K rwdata, 2952K rodata, 1024K init, 862K bss, 26412K reserved, 65536K cma-reserved)
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] ftrace: allocating 32081 entries in 95 pages
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] ftrace: allocated 94 pages with 5 groups
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] rcu: Hierarchical RCU implementation.
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] 	Rude variant of Tasks RCU enabled.
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] 	Tracing variant of Tasks RCU enabled.
Mar  8 20:43:42 raspberrypi kernel: [    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[... cut ...]
Mar  8 20:44:06 raspberrypi mpd: update: updating Cantajuegos/Cantajuego - 05 - El Arca De Noe.mp3
Mar  8 20:44:10 raspberrypi systemd[1]: systemd-fsckd.service: Succeeded.
Mar  8 20:44:38 raspberrypi systemd-timesyncd[345]: Synchronized to time server for the first time [2001:418:3ff::53]:123 (2.debian.pool.ntp.org).
Mar  8 20:44:39 raspberrypi systemd[1]: systemd-hostnamed.service: Succeeded.
Mar  8 20:45:16 raspberrypi systemd[1]: phoniebox-idle-watchdog.service: Succeeded.
Mar  8 20:50:40 raspberrypi systemd[1]: Started Session 4 of user pi.