MiczFlor/RPi-Jukebox-RFID #1180: How to switch on and off your powerbank powerd Phoniebox?

MrReSc, 2020-12-07:
I have just ordered all the missing material to build a Phoniebox.
Among other things a powerbank with two USB-A outputs and a Micro USB for charging as well as the USB speakers that are recommended.

I still wonder how you solved the switching on and off. With a GPIO button the box can be shut down. How do I turn the box back on? I wanted to connect the loudspeakers directly to the powerbank, because the RPi can't supply that much power. This means that the loudspeakers always draw a small current.

How did you solve this with your powerbank powerd Phoniebox?

domu83, 2020-12-07:
Hi @MrReSc
I use this one: https://shop.pimoroni.com/products/onoff-shim but connect the speakers to the Pi.
Good read recommondation is this @splitti's blog here: https://splittscheid.de/selfmade-phoniebox/

Cheers

s-martin, 2020-12-07:
> Hi @MrReSc
> I use this one: https://shop.pimoroni.com/products/onoff-shim but connect the speakers to the Pi.
> Good read recommondation is this @splitti's blog here: https://splittscheid.de/selfmade-phoniebox/
> 
> Cheers

Yes, I can recommend that, too.

MrReSc, 2020-12-09:
The OnOff SHIM works perfectly with my Powerbank. Thanks.

closed 2020-12-09 by MrReSc

Sheepiely, 2021-09-30:
@MrReSc which power bank do you use?

MrReSc, 2021-09-30:
> @MrReSc which power bank do you use?

BOM: https://github.com/MrReSc/PhonieboxSetup

Sheepiely, 2021-09-30:
@MrReSc  Thanks for the speedy response. The onoff shim can't turn the pi zero back on when my Mi power bank 3 goes into deep sleep. I'm going to order the Adata powerbank now.