MiczFlor/RPi-Jukebox-RFID #1182: 🚀 | Second-Swipe to skip to next subfolder

Burnum-Burnum, 2020-12-07:
## Feature Description
First off, I'm quite new to Phoniebox and raspberry-pi in general, but my Phoniebox is up an running quite smoothly :_)
So it is totally possible that this feature is already implemented and I'm just not aware of it. If so, please push me in the right direction!

### What functionality would you like to see in your phoniebox?

Option to have second swipe skip to the next subfolder. A lot of audio books have a new track every minute or so, so skipping to the next book would be usefull

### How do you envision the feature to work from a users perspective?

Over the last weeks I accumulated quite a lot audio books for my son. They are organized in Folders as followed:

- audio_books
   - author_1
       - book_1
           - track_01.mp3
           - track_02.mp3
           - ...
           - track_40.mp3
        - book_2
           - track_01.mp3
           - track_02.mp3
           - ...
           - track_40.mp3
        - ...
        - book_50

and so on. Since each book has about 40 trackes, skipping forward to the next book is quite time conuming using the second swipe method. 
Would be awesome, if one could link a card to the author_1 folder and it would start playing book_1. By swiping a second time, it would skip to the next subfolder of author_1

### Further information that might help

alwashe, 2020-12-27:
This would be great. Is there a way to do this now?

Burnum-Burnum, 2020-12-28:
I've tried some ways to get this to work, but nothing did it...
My best guess was creating a m3u playlist not with single mp3 but with m3u's for the single books. But that didn't work either :_)

matthias-pelger, 2020-12-31:
hey there...
as I really like this feature, I started working on this, together with my "random subfolder play" feature. But there are some changes needed to include this properly. 
The idea is to place this as a switch in folder config, so that second swipe can stay. Planning to add a pull request for this in early January.