MiczFlor/RPi-Jukebox-RFID #1183: Läuft nach Update nicht mehr

Eisselpudel, 2020-12-08:
Ich habe mich endlich mal wieder an die Phoniebox gesetzt und alles lief super. Wollte jetzt den Bluetoothlautsprecher mal aktivieren und habe mein RaspberryPi nach einem Jahr geupdatet. Danach läuft der Player nicht mehr. Ich denke, es liegt am Mopidy-Server. Der ist nicht verbunden. Das steht nämlich beim Player, der Mopidy-Service läuft.
Bin gerade verzweifelt.

```
************************************
*** PHONIEBOX INFO
*** version: 2.2 - 305325d - master
*** edition: plusSpotify
*** Audio_iFace_Name: PCM
*** Audio_Folders_Path: /home/pi/RPi-Jukebox-RFID/shared/audiofolders
*** Audio_Volume_Change_Step: 3
*** Max_Volume_Limit: 100
*** Idle_Time_Before_Shutdown: 0
*** Second_Swipe: PAUSE
*** Playlists_Folders_Path: /home/pi/RPi-Jukebox-RFID/playlists
*** ShowCover: OFF
************************************
*** CONF FILES DEFAULT
 
*** /etc/samba/smb.conf
-rw-r--r-- 1 root root 8.2K Feb 24  2020 /etc/samba/smb.conf
   path=/home/pi/RPi-Jukebox-RFID/shared
   path=/home/pi/RPi-Jukebox-RFID/shared/audiofolders
*** /etc/lighttpd/lighttpd.conf
-rw-r--r-- 1 root root 2.1K Feb 24  2020 /etc/lighttpd/lighttpd.conf
*** /etc/lighttpd/conf-available/15-fastcgi-php.conf
-rw-r--r-- 1 root root 575 Feb 24  2020 /etc/lighttpd/conf-available/15-fastcgi-php.conf
*** /etc/php/7.0/fpm/php.ini
ls: cannot access '/etc/php/7.0/fpm/php.ini': No such file or directory
*** /etc/sudoers
-r--r----- 1 root root 784 Feb 24  2020 /etc/sudoers
*** /etc/systemd/system/phoniebox*
-rw-r--r-- 1 root root 314 Feb 24  2020 /etc/systemd/system/phoniebox-rfid-reader.service
-rw-r--r-- 1 root root 285 Feb 24  2020 /etc/systemd/system/phoniebox-startup-sound.service
-rw-r--r-- 1 root root 289 Feb 24  2020 /etc/systemd/system/phoniebox-gpio-buttons.service
-rw-r--r-- 1 root root 291 Feb 24  2020 /etc/systemd/system/phoniebox-idle-watchdog.service
*** /etc/mpd.conf
-rw-r----- 1 mpd audio 14K Feb 24  2020 /etc/mpd.conf
music_directory		"/home/pi/RPi-Jukebox-RFID/shared/audiofolders"
# music_directory are changed.
# symbolic links outside of the configured music_directory.
# symbolic links inside of the configured music_directory.
	mixer_control	"PCM"		# optional
#	mixer_control	"PCM"		# optional
*** /etc/dhcpcd.conf
-rw-rw-r-- 1 root netdev 1.8K Nov 13  2019 /etc/dhcpcd.conf
#static ip_address=192.168.0.10/24
#static ip_address=192.168.1.23/24
#static routers=192.168.0.1
#static routers=192.168.1.1
option domain_name_servers, domain_name, domain_search, host_name
#static domain_name_servers=192.168.0.1 8.8.8.8 fd51:42f8:caae:d92e::1
#static domain_name_servers=192.168.1.1
*** /etc/wpa_supplicant/wpa_supplicant.conf
-rw-r--r-- 1 root root 160 Feb 13  2020 /etc/wpa_supplicant/wpa_supplicant.conf
country=DE
************************************
*** +Spotify Edition
 
*** /etc/locale.gen
-rw-r--r-- 1 root root 8.9K Feb 24  2020 /etc/locale.gen
*** /etc/locale.gen
-rw-r--r-- 1 root root 8.9K Feb 24  2020 /etc/locale.gen
*** /etc/mopidy/mopidy.conf
-rw-r----- 1 mopidy root 1.1K Feb 24  2020 /etc/mopidy/mopidy.conf
username = bremenpudel
*** ~/.config/mopidy/mopidy.conf
-rw-r--r-- 1 root root 3.3K Dec  8 15:16 /home/pi/.config/mopidy/mopidy.conf
#username = 
username = bremenpudel

```

malzeit, 2020-12-08:
Ich habe das gleiche Problem.
Was mir bisher aufgefallen ist, dass wenn ich die Phoniebox-GUI öffne und in der Maske "Ordner & Dateien" die Datenbank erneuere (Musikbibliothek scannen) und dann die Karte 2-3 mal auflegen, dann läuft es mit allen Karten auch mit Spotify. Davor laufen nur lokal abgelegte Musik.

s-martin, 2020-12-08:
> Ich habe das gleiche Problem.
> Was mir bisher aufgefallen ist, dass wenn ich die Phoniebox-GUI öffne und in der Maske "Ordner & Dateien" die Datenbank erneuere (Musikbibliothek scannen) und dann die [...]

Please try:

https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#spotify-wont-play-but-after-scan-of-library-it-works

malzeit, 2020-12-08:
> Please try:
>
> https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#spotify-wont-play-but-after-scan-of-library-it-works

thank you very much! Now it runs.

now I just have to solve the audio problem. It only wants to play via HDMI, not via headphone input, although this was selected during installation. But that is another story.

s-martin, 2020-12-08:
> now I just have to solve the audio problem. It only wants to play via HDMI, not via headphone input, although this was selected during installation. But that is another story.

There you go:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#raspberry-pi-update-may-2020-broke-my-audio

Eisselpudel, 2020-12-09:
That doesn't work with my phoniebox.
I have to start the mopidy-server everytime with "sudo service mpd restart", then I scan the library but nothing happens, even the local music doesn't work.

s-martin, 2020-12-12:
Please post the result of `journalctl -u mopidy`.

Eisselpudel, 2020-12-12:
Sorry, forget to write:
I re-installed phoniebox yesterday and now everything is running. :-)
Thank you for your answers.

closed 2020-12-12 by s-martin