MiczFlor/RPi-Jukebox-RFID #1184: Audio not working

TimiRose, 2020-12-09:
Hi! 
I am new to Raspberry Pi and Command line. I just followed all the steps with the one-line install and installation seemed to work. 
I can upload files to the web app, but most files don´t play at all and the ones that play are very quite even at 100% volume.
Spotify scans libraries forever, but nothing happens.
None of the sample content for testing does anything. Just stays silent.

what version are you on? Type in terminal cat /home/pi/settings/version 
_I don´t have a settings folder inside the pi directory. the command doesn´t work_

what soundcard are you using?
_headphone output_

does your Raspberry Pi play any sound?
_yes, a small amount of the files, but very quietly_

did the audio work and then you connected a sound card and now it doesn't?
_see above_

Thanks for helping out!
Tim

s-martin, 2020-12-09:
> Hi!
> I am new to Raspberry Pi and Command line. I just followed all the steps with the one-line install and installation seemed to work.
> I can upload files to the web app, but most files don´t play [...]

Please use `cat /home/pi/RPi-Jukebox-RFID/settings/version`

> what soundcard are you using?
> _headphone output_
> 
> does your Raspberry Pi play any sound?
> _yes, a small amount of the files, but very quietly_
> 
> did the audio work and then you connected a sound card and now it doesn't?
> _see above_
> 
> Thanks for helping out!
> Tim

Please try the hints in section a) of that section in the Troubleshooting FAQ

TimiRose, 2020-12-09:
what version are you on?
_2.2 - 305325d - master_

Since I´m not using an external soundcard I understood that section a) doesn´t occur to me. 

The lines at the end of the file are this
defaults.ctl.card 0
defaults.pcm.card 0

and that should be right, no?

s-martin, 2020-12-10:
Could you check this section, too:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#raspberry-pi-update-may-2020-broke-my-audio

In May 2020 some Raspberry changes were done and I think that's not fully reflected in the installer script, if you use on board sound.

TimiRose, 2020-12-10:
Okay it worked with typing alsamixer and then selecting the Headphone output as card. There the volume was somehow set to very low by default. Thanks!

closed 2020-12-10 by TimiRose