MiczFlor/RPi-Jukebox-RFID #1188: 🐛 | Raspberry OS 2020-12-02 with Script v2.2 => error while installing

0x0lli, 2020-12-10:
## Bug
### What I did
installed the raspberry pi with above mentioned buster image and ran the installer script

### What happened
Installer breaks at this point : 
'''
Successfully built multidict
Installing collected packages: multidict, typing-extensions, async-timeout, yarl                 , attrs, aiohttp, backoff, spotify
Successfully installed aiohttp-3.7.3 async-timeout-3.0.1 attrs-20.3.0 backoff-1.                 10.0 multidict-5.1.0 spotify-0.10.2 typing-extensions-3.7.4.3 yarl-1.6.3
./stretch-install-default.sh: line 488: git: command not found
./stretch-install-default.sh: line 489: cd: mopidy-spotify: No such file or directory
'''

## Software
### Base image and version
2020-12-02-raspios-buster-armhf-lite

### Installscript
scripts/installscripts/buster-install-default.sh

## Hardware
### RaspberryPi version
3 A+

### RFID Reader
RFID-RC522

### Soundcard
hifiberry MiniAMP

0x0lli, 2020-12-10:
just tried it using 2020-08-20-raspios-buster-armhf-lite:
samesame.

This is the first time installing spotify with it, I've not tested it yet on any system I've setup so far.

Took a look and ### **git wasnt installed** with the lite versions apparently.
Therefore, installed git 
The installation worked then, except for one thing: 
### **The RFID-options aka device-list to choose from is not displayed**:

```
If you are using an USB RFID reader, connect it to your RPi.
(In case your RFID reader required soldering, consult the manual.)
Have you connected your USB Reader? [Y/n] Y
Choose the reader from list
Device Number: 
```

s-martin, 2020-12-10:
Looks like you used the installer script for Raspian Stretch (old release).

Please use buster-install-default.sh for Raspian Buster.

0x0lli, 2020-12-10:
Thanks Martin, looks like I always find the biggest Fettnäpfchen :)

closed 2020-12-10 by 0x0lli