MiczFlor/RPi-Jukebox-RFID #1189: OnOff SHIM music does resume automatically

MrReSc, 2020-12-10:
I used the OnOff SHIM to switch the Phoniebox on and off. If something is playing and I switch the box off with the OnOff SHIM without stopping the music, and then start it again, the music is automatically played again. Is it possible to change this behaviour?

If music is playing and I restart from the Web UI, the behaviour is different. The music does not resume automatically.

BerniPi, 2020-12-11:
Alternative solution:
Add the line
/home/pi/RPi-Jukebox-RFID/scripts/playout_controls.sh -c=shutdown

instead of the two mentioned lines at the same place.

Advantage: the shutdown routine gets updated with phoniebox updates.

s-martin, 2020-12-11:
Duplicates #951

MrReSc, 2020-12-12:
> Alternative solution:
> Add the line
> `/home/pi/RPi-Jukebox-RFID/scripts/playout_controls.sh -c=shutdown`
> 
> instead of the two mentioned lines at the same place.
> 
> Advantage: the shutdown routine gets updated with phoniebox updates.

I have just found the solution myself. I did it exactly like this.

closed 2020-12-12 by MrReSc