MiczFlor/RPi-Jukebox-RFID #1191: Spotify doesn´t connect,

TimiRose, 2020-12-10:
Hi, I am new to Phoniebox and just set up the latest fresh install. 
I am connected to Spotify, through "Browse" I can acess and play my library in a very weird way (some of my tracks show up, but no idea how they are put together). All other links, like Playlists, Artists etc just load forever and say "Could not load library artists"
If I click on "Tracks" it says "Enable Spotify to browse tracks".
Upgrading to Iris 3.55.2. takes forever but nothing happens.

Mp3s I upload on to the Jukebox only properly play through the Spotify app, not the webplayer.

Thanks for helping!
Tim

s-martin, 2020-12-10:
Please post results of` journalctl -u mopidy`.

You could also check here https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#spotify-wont-play-but-after-scan-of-library-it-works

TimiRose, 2020-12-10:
Here are the results.

-- Logs begin at Thu 2020-12-10 21:23:38 CET, end at Thu 2020-12-10 23:32:24 CET. --
Dec 10 21:23:43 raspberrypi systemd[1]: Starting Mopidy music server...
Dec 10 21:23:43 raspberrypi systemd[1]: Started Mopidy music server.
Dec 10 21:23:53 raspberrypi mopidy[500]: INFO     [MainThread] mopidy.__main__ Starting Mopidy 3.0.2
Dec 10 21:23:53 raspberrypi mopidy[500]: INFO     [MainThread] mopidy.config Loading config from builtin defaults
Dec 10 21:23:53 raspberrypi mopidy[500]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mop
Dec 10 21:23:53 raspberrypi mopidy[500]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mop
Dec 10 21:23:53 raspberrypi mopidy[500]: INFO     [MainThread] mopidy.config Loading config from file:///etc/mopidy/mo
Dec 10 21:23:53 raspberrypi mopidy[500]: INFO     [MainThread] mopidy.config Loading config from command line options
Dec 10 21:23:53 raspberrypi mopidy[500]: ERROR    [MainThread] mopidy.internal.log Loading logging config '/etc/mopidy
Dec 10 21:23:55 raspberrypi mopidy[500]: INFO     [MainThread] mopidy.__main__ Enabled extensions: softwaremixer, http
Dec 10 21:23:55 raspberrypi mopidy[500]: INFO     [MainThread] mopidy.__main__ Disabled extensions: none
Dec 10 21:23:56 raspberrypi mopidy[500]: INFO     [MainThread] mopidy.commands Starting Mopidy mixer: SoftwareMixer
Dec 10 21:23:56 raspberrypi mopidy[500]: INFO     [MainThread] mopidy.commands Mixer volume set to 30
Dec 10 21:23:56 raspberrypi mopidy[500]: INFO     [MainThread] mopidy.commands Starting Mopidy audio
Dec 10 21:23:56 raspberrypi mopidy[500]: INFO     [MainThread] mopidy.commands Starting Mopidy backends: FileBackend, 
Dec 10 21:23:56 raspberrypi mopidy[500]: INFO     [Audio-2] mopidy.audio.actor Audio output set to "alsasink"
Dec 10 21:23:57 raspberrypi mopidy[500]: INFO     [SpotifyEventLoop] mopidy_spotify.backend Logged in to Spotify in on
Dec 10 21:23:58 raspberrypi mopidy[500]: INFO     [SpotifyBackend-6] mopidy_spotify.web Logged into Spotify Web API as
Dec 10 21:24:27 raspberrypi mopidy[500]: INFO     [SpotifyBackend-6] mopidy_spotify.playlists Refreshed 45 Spotify pla
Dec 10 21:24:27 raspberrypi mopidy[500]: INFO     [MainThread] mopidy.commands Starting Mopidy core
Dec 10 21:24:27 raspberrypi mopidy[500]: INFO     [MainThread] mopidy.commands Starting Mopidy frontends: HttpFrontend
Dec 10 21:24:27 raspberrypi mopidy[500]: INFO     [HttpFrontend-12] mopidy.http.actor HTTP server running at [::ffff:0
Dec 10 21:24:27 raspberrypi mopidy[500]: INFO     [IrisFrontend-13] mopidy_iris.core Starting Iris 3.45.0
Dec 10 21:24:27 raspberrypi mopidy[500]: INFO     [MainThread] mopidy_mpd.actor MPD server running at [::ffff:0.0.0.0]
Dec 10 21:24:27 raspberrypi mopidy[500]: INFO     [MainThread] mopidy.commands Starting GLib mainloop
Dec 10 21:24:27 raspberrypi mopidy[500]: INFO     [MpdSession-15] mopidy_mpd.session New MPD connection from [::ffff:1

Will check the link tomorrow.
Thanks!

s-martin, 2020-12-10:
Log looks ok, but I think the given link may help....

s-martin, 2023-12-05:
Closing as no interaction for a couple of months.
Feel free to reopen the issue, if the problem still occurs.

closed 2023-12-05 by s-martin