MiczFlor/RPi-Jukebox-RFID #1192: 🚀 | Add support for toggling between speakers and bluetooth headphones

ChisSoc, 2020-12-11:
## Feature Description

### What functionality would you like to see in your phoniebox?

If users have regular speakers and Bluetooth headphones connected to their Phoniebox, provide a mechanism through all the usual user interfaces (i.e. GPIO, Card Swipe, Webinterface) to toggle between the two audio sinks. 

### How do you envision the feature to work from a users perspective?

When a user powers on their Bluetooth headphones, they connect automatically to the Phoniebox (as described in the Wiki). At the swtich of the button (or card swipe, etc) the (already running) audio playback is transferred from the speakers to the headphones. This happens almost seamlessly. Parents feel an instant wave of relief at not having to listen to the 500th iteration of this month favourite song. The small user feels instantly proud at having working headphones much like the dad always uses while doing home-office online meetings. If no Bluetooth headphones are connected, the audio sink toggle request defaults to speakers. An LED indicates the currently active audio sink.

### Further information that might help

The feature is already coded and runs on my Phoniebox. And I'd like to share it :-) 

The sink toggle feature uses the mpd output feature to toggle between the outputs. But around that a bit of code is needed for integration into the Phoniebox. It checks for Bluetooth connectivity first and also sets an (optional) LED for visual status control. 

The web interface extension has also already been coded. 

Now, I'd like to hear your thoughts on how to integrate the feature properly.

There is a few things to consider
- to follow the API thought, the calling of the bt-switch.py script should happen through playout_controls.sh, which would need a new command
- rfid_trigger_play.sh should also get a new entry to enable this feature with a card swipe
- in the web interface, the feature should only be visible, if the user has actually set up two audio sinks. This could be checked  from the 'mpc outputs' command. But this is a little unspecific, because two audio sinks in the mpd.conf does not guarantee that the second output is the Bluetooth device. Alternatively, a new configuration setting could be introduced. What are your thoughts?
- after boot the Phoniebox should default back to speaker output, to make sure that playback can always be heard preventing the user from thinking the Box would be broken. This scenario is relevant, if the Phoniebox is shut down with headphone sink, but booted up another day, without headphones connected. This is easily done by adding a line to the startup.sh script.
- a install script would be helpful for new users, when enabling this feature. Probably a wiki article as well.

[image]

s-martin, 2020-12-15:
Cool! As you already have an implementation, it’s would probably be the easiest to open a PR.

ChisSoc, 2020-12-18:
I have done the clean-up and testing. Almost there to create the pull request. But I have one question, I'd like to clarify beforehand:

For enabling/disabeling this feature, I am checking the file ./settings/BtHeadphoneIntegration for existence and content. I have not integrated that into ./settings/global.conf and ./scripts/inc.writeGlobalconfig.sh

The question is: should I do that or not?
- **yes**, to make all feature-controlling settings behave in the same manner
- **no**, because this feature will only be used by a percentace of Phoniebox users.

If you want to take a look at the code so far, the current state is pushed. Main dir is here: https://github.com/ChisSoc/RPi-Jukebox-RFID/tree/bt-headphones/components/bt-headphones , but I also had to modify a set of other files for full integration into rfid_trigger_play and web interface.

closed 2021-02-26 by MiczFlor