MiczFlor/RPi-Jukebox-RFID #1193: 🚀 | Control Phonieboxh through buttons on Bluetooth headset

ChisSoc, 2020-12-11:
## Feature Description

### What functionality would you like to see in your phoniebox?

Many Bluetooth headsets have buttons to control play/pause, next track, previous track. Let's make them usable to control the Phoniebox like GPIO control buttons (if a headset is connected via Bluetooth)

### How do you envision the feature to work from a users perspective?

The user can connect the Bluetooth headset to the Phoniebox and without further interaction, use the Buttons on the Headset to control the playback.

### Further information that might help

A corresponding script is already coded and works flawlessly on my Phoniebox. Now, I'd like to share it. It runs as service, checking for the Bluetooth device connection. Upon button press of connected BT device, it reacts on the event and calls playout_controls.sh. 

During installation, a little bit of customization to each specific headset is most likely necessary by the user (for mapping key codes to desired actions). With a corresponding readme, that is no problem and is in the best spirit of enabling people to tinker with things. An install script can also be helpful to set up the service.

I'd love to hear your thoughts. 

Together with my other feature proposal https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/1192, I could put both into a folder 'components/bt-headphones'

s-martin, 2020-12-15:
I think many users would love that.

Please open a PR with your working implementation.

I would suggest to use the path `components/control/bt-headphones`

ChisSoc, 2020-12-18:
Happy to oblige. However, I am currently not seeing this folder structure, but ./components/gpio_control. Is this something you are going to change?

s-martin, 2020-12-18:
Please add the subdirectory control, doesn’t exist yet.

ChisSoc, 2021-01-04:
The real testing only started after my daughter unwrapped her Christmas present. I had to fix a few small issues. Now, I will go ahead and prepare the pull request. Sorry for the delay.

closed 2021-02-26 by MiczFlor