MiczFlor/RPi-Jukebox-RFID #1194: Neither onOff Shim nor shutdown -h 0 powers Raspi off

Barzille, 2020-12-11:
## Bug

### What I did

Powering down using onoff shim or using "sudo shutdown -h 0" or using ./scripts/playout_controls.sh -c=shutdown helps.

### What happened
- Network is dropping
- The Raspi seems to go down but the red light keeps shining

### I expected this to happen

The Raspi is disconnected and shutdown completely so I can restart it using the shim

### Further information that might help
```

pi@raspberrypi:~/RPi-Jukebox-RFID $ ./scripts/playout_controls.sh -c=shutdown
lsof: WARNING: can't stat() fuse.gvfsd-fuse file system /run/user/1000/gvfs
      Output information may be incomplete.
lsof: WARNING: can't stat() fuse.gvfsd-fuse file system /run/user/1000/gvfs
      Output information may be incomplete.
lsof: WARNING: can't stat() fuse.gvfsd-fuse file system /run/user/1000/gvfs
      Output information may be incomplete.
volume: n/a   repeat: off   random: off   single: off   consume: off
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.25.10; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /home/pi/RPi-Jukebox-RFID/scripts/../shared/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: shutdownsound.mp3 ...

MPEG 1.0 L III cbr64 44100 mono

[0:02] Decoding of shutdownsound.mp3 finished.
Shutdown scheduled for Fri 2020-12-11 13:07:46 GMT, use 'shutdown -c' to cancel.
```

```
pi@raspberrypi:~ $ tail -n 500 /var/log/syslog
Dec 11 13:12:51 raspberrypi kernel: [    7.491154] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
Dec 11 13:12:51 raspberrypi kernel: [    7.492002] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Oct 22 2019 01:59:28 version 7.45.98.94 (r723000 CY) FWID 01-3b33decd
Dec 11 13:12:51 raspberrypi kernel: [    8.716817] Under-voltage detected! (0x00050000)
Dec 11 13:12:51 raspberrypi systemd[1]: Started System Logging Service.
Dec 11 13:12:51 raspberrypi systemd[1]: Started Check for Raspberry Pi EEPROM updates.
Dec 11 13:12:51 raspberrypi rngd[440]: rngd 2-unofficial-mt.14 starting up...
Dec 11 13:12:51 raspberrypi kernel: [    9.877558] uart-pl011 3f201000.serial: no DMA platform data
Dec 11 13:12:51 raspberrypi rng-tools[424]: Starting Hardware RNG entropy gatherer daemon: rngd.
Dec 11 13:12:51 raspberrypi systemd[1]: Started Deferred execution scheduler.
Dec 11 13:12:51 raspberrypi systemd[1]: Started rng-tools.service.
Dec 11 13:12:51 raspberrypi systemd[1]: Started Save/Restore Sound Card State.
Dec 11 13:12:51 raspberrypi thd[435]: Found socket passed from systemd
Dec 11 13:12:51 raspberrypi kernel: [    9.886786] 8021q: 802.1Q VLAN Support v1.8
Dec 11 13:12:51 raspberrypi rngd[440]: entropy feed to the kernel ready
Dec 11 13:12:51 raspberrypi dhcpcd[377]: `' is not a valid IP address
Dec 11 13:12:51 raspberrypi dhcpcd[377]: `' is not a valid IP address
Dec 11 13:12:51 raspberrypi systemd[1]: Started Check for v3d driver.
Dec 11 13:12:51 raspberrypi kernel: [    9.917324] random: crng init done
Dec 11 13:12:51 raspberrypi kernel: [    9.917343] random: 7 urandom warning(s) missed due to ratelimiting
Dec 11 13:12:51 raspberrypi systemd[1]: Started triggerhappy global hotkey daemon.
Dec 11 13:12:51 raspberrypi dhcpcd[377]: forked to background, child pid 445
Dec 11 13:12:51 raspberrypi systemd[1]: Started dhcpcd on all interfaces.
Dec 11 13:12:51 raspberrypi dphys-swapfile[395]: want /var/swap=100MByte, checking existing: keeping it
Dec 11 13:12:51 raspberrypi udisksd[431]: udisks daemon version 2.8.1 starting
Dec 11 13:12:51 raspberrypi systemd[1]: Reached target Sound Card.
Dec 11 13:12:51 raspberrypi dbus-daemon[409]: [system] Successfully activated service 'org.freedesktop.systemd1'
Dec 11 13:12:51 raspberrypi systemd[1]: Started Login Service.
[... 481 lines cut ...]
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

### Branch / Release

master

### Installscript

## Hardware

### RaspberryPi version

Raspberry Pi 3 Model B Rev 1.2

### Other notable hardware

s-martin, 2020-12-11:
Did you test, if the RPi can be started again with the on off shim after you shut down?

Barzille, 2020-12-11:
Yes. That’s not working as it is still somehow powered

s-martin, 2020-12-11:
Do you use a gpio button for shutdown command or the button on the onoff shim?

Barzille, 2020-12-11:
At first the onoff shim button. Now I connected a push button to the shim -> same behaviour 

But this is only the trigger as I mentioned above the shutdown command itself results in the same outcome

Barzille, 2020-12-15:
When I power the RPi via direct current connection and not via the onOff Shim it has the same behavior (red light stays on), so it seems to me that the shim is not cutting the power connection, maybe it's defect. I have ordered a new one and will let you know once I checked that.

Barzille, 2020-12-15:
And btw is this my clean shutdown config for the shim:
```
# Config for cleanshutd
# Commented out values will be reverted to defaults,
# and may not work on any given board.

# OnOff SHIM uses trigger 17 and poweroff 4
# Zero Lipo uses trigger 4 and poweroff off
# pHAT BEAT uses trigger 12 and powerof off
# Default values are trigger 4 and poweroff off

daemon_active=1
trigger_pin=17
led_pin=17
poweroff_pin=4
hold_time=2
shutdown_delay=0
polling_rate=1
```

s-martin, 2020-12-15:
Unfortunately my Phoniebox w OnOffShim has a hardware defect, so I can’t check. Will get back to you as soon as I tested it.

Barzille, 2020-12-18:
I haven’t received the new one yet

Barzille, 2020-12-20:
I just tried with a new shim, same behavior.
Is there anything I could check or look for in special?

Barzille, 2020-12-20:
btw this is my gpio_setting.ini

```
[DEFAULT]
enabled: True
[VolumeControl]
enabled: True
Type: TwoButtonControl ;or RotaryEncoder
PinUp: 5
PinDown: 6
pull_up: True
hold_time: 0.3
hold_repeat: True
timeBase: 0.1 ;only for RotaryEncoder
functionCallDown: functionCallVolD
functionCallUp: functionCallVolU
functionCallTwoButtons: functionCallVol0 ;only for TwoButtonControl

[PrevNextControl]
enabled: False
Type: TwoButtonControl
Pin1: 22
Pin2: 23
functionCall1: functionCallPlayerPrev
functionCall2: functionCallPlayerNext
functionCallTwoButtons: None
pull_up: True
hold_time: 0.3
hold_repeat: False

[PlayPause]
enabled: True
Type: Button
Pin: 27
pull_up: True
hold_time: 0.3
functionCall: functionCallPlayerPause

[Shutdown]
enabled: False
Type:  Button
Pin: 3
pull_up: True
hold_time: 2
functionCall: functionCallShutdown

[Volume0]
enabled: False
Type:  Button
Pin: 17
pull_up: True
hold_time: 0.3
functionCall: functionCallVol0

[VolumeUp]
enabled: True
Type:  Button
Pin: 24
pull_up: True
hold_time: 0.3
hold_repeat: True
functionCall: functionCallVolU

[VolumeDown]
enabled: True
Type:  Button
Pin: 16
pull_up: True
hold_time: 0.3
hold_repeat: True
functionCall: functionCallVolD

[NextSong]
enabled: True
Type:  Button
Pin: 23
pull_up: True
hold_time: 0.3
functionCall: functionCallPlayerNext

[PrevSong]
enabled: True
Type:  Button
Pin: 13
pull_up: True
hold_time: 0.3
functionCall: functionCallPlayerPrev

[Halt]
enabled: False
Type:  Button
Pin: 21
pull_up: True
hold_time: 0.3
functionCall: functionCallPlayerPauseForce
```

Barzille, 2020-12-20:
And this is the syslog after pushing the onOff Shim Button:

```
pi@raspberrypi:~/RPi-Jukebox-RFID/settings $ sudo tail -f /var/log/syslog
Dec 20 17:23:05 raspberrypi systemd[1]: man-db.service: Succeeded.
Dec 20 17:23:05 raspberrypi systemd[1]: Started Daily man-db regeneration.
Dec 20 17:23:06 raspberrypi systemd[1]: apt-daily.service: Succeeded.
Dec 20 17:23:06 raspberrypi systemd[1]: Started Daily apt download activities.
Dec 20 17:23:06 raspberrypi systemd[1]: Starting Daily apt upgrade and clean activities...
Dec 20 17:23:07 raspberrypi systemd[1]: systemd-hostnamed.service: Succeeded.
Dec 20 17:23:08 raspberrypi systemd[1]: apt-daily-upgrade.service: Succeeded.
Dec 20 17:23:08 raspberrypi systemd[1]: Started Daily apt upgrade and clean activities.
Dec 20 17:23:19 raspberrypi systemd[1]: Started Session 4 of user pi.
Dec 20 17:23:33 raspberrypi systemd[1]: phoniebox-idle-watchdog.service: Succeeded.

Broadcast message from root@raspberrypi (somewhere) (Sun Dec 20 17:29:37 2020):

BCM 17 held low, system shutdown in 0 minutes

Dec 20 17:29:37 raspberrypi systemd[1]: systemd-rfkill.socket: Succeeded.
Dec 20 17:29:37 raspberrypi systemd[1]: Closed Load/Save RF Kill Switch Status /dev/rfkill Watch.
Dec 20 17:29:37 raspberrypi systemd[1]: Stopped target Timers.
Dec 20 17:29:37 raspberrypi systemd[1]: Stopped target Graphical Interface.
Dec 20 17:29:37 raspberrypi systemd[1]: Stopping Session 3 of user pi.
Dec 20 17:29:37 raspberrypi systemd[1]: Stopped target Multi-User System.
Dec 20 17:29:37 raspberrypi systemd[1]: Stopping Samba SMB Daemon...
Dec 20 17:29:37 raspberrypi systemd[1]: Stopping Lighttpd Daemon...
Dec 20 17:29:37 raspberrypi systemd[1]: Stopping LSB: exim Mail Transport Agent...
Dec 20 17:29:37 raspberrypi systemd[1]: gldriver-test.service: Succeeded.
Dec 20 17:29:37 raspberrypi systemd[1]: Stopped Check for v3d driver.
Dec 20 17:29:37 raspberrypi systemd[1]: Stopping Regular background program processing daemon...
Dec 20 17:29:37 raspberrypi systemd[1]: Stopping Configure Bluetooth Modems connected by UART...
Dec 20 17:29:37 raspberrypi systemd[1]: Stopping Authorization Manager...
Dec 20 17:29:37 raspberrypi systemd[1]: Removed slice system-bthelper.slice.
Connection to raspberrypi closed by remote host.
Connection to raspberrypi closed.
```

chaskitb, 2020-12-23:
I have a similar issue. With the update to 2.2 the shutdown over GPIO does not work. As soon as the part in the gpio-settings.ini the section [shutdown] is enabled with "enabled : True" the other buttons don't work anymore, and most interestingly, the sound does not work anymore as well.

s-martin, 2020-12-23:
> I have a similar issue. With the update to 2.2 the shutdown over GPIO does not work. As soon as the part in the gpio-settings.ini the section [shutdown] is enabled with "enabled : True" the other buttons don't work anymore, and most interestingly, the sound does not work anymore as well.

Could you please show your gpio_settings.ini?

chaskitb, 2020-12-23:
[VolumeControl]
enabled: True
Type: TwoButtonControl
PinUp: 6
PinDown: 5
pull_up: False
hold_time: 0.3
hold_repeat: True
timeBase: 0.1
functionCallDown: functionCallVolD
functionCallUp: functionCallVolU
functionCallTwoButtons: None

[PrevNextControl]
enabled: True
Type: TwoButtonControl
Pin1: 26
Pin2: 13
functionCall1: functionCallPlayerPrev
functionCall2: functionCallPlayerNext
functionCallTwoButtons: None
pull_up: False
hold_time: 0.3
hold_repeat: False

[PlayPause]
enabled: True
Type: Button
Pin: 12
pull_up: False
hold_time: 0.3
functionCall: functionCallPlayerPause

[Shutdown]
enabled: False
Type: Button
Pin: 3
pull_up: False
hold_time: 2
functionCall: functionCallShutdown

[Volume0]
enabled: False
Type:  Button
Pin: 17
pull_up: True
hold_time: 0.3
functionCall: functionCallVol0

[VolumeUp]
enabled: False
Type:  Button
Pin: 5
pull_up: True
hold_time: 0.3
hold_repeat: True
functionCall: functionCallVolU

[VolumeDown]
enabled: False
Type:  Button
Pin: 6
pull_up: True
hold_time: 0.3
hold_repeat: True
functionCall: functionCallVolD

[NextSong]
enabled: False
Type:  Button
Pin: 26
pull_up: True
hold_time: 0.3
functionCall: functionCallPlayerNext

[PrevSong]
enabled: False
Type:  Button
Pin: 13
pull_up: True
hold_time: 0.3
functionCall: functionCallPlayerPrev

[Halt]
enabled: False
Type:  Button
Pin: 21
pull_up: True
hold_time: 0.3
functionCall: functionCallPlayerPauseForce

Barzille, 2020-12-23:
Is the „Halt“ or „Shutdown“ section in any way related to the onOff Shim? Meaning: should I enable any of it to have the shim working?

matthias-pelger, 2020-12-29:
If you are using the button or button connector on the shim itself, the GPIO settings are irrelevant.

If you experience a "restart" with your onoffshim, it could also be, that you have a capacitor attached to +5v or +3,3  pins and GND.
It is normally suggested to use this capacitor to enable a short powersupply if your powerbank switches from or to charging and would normally loose power. Other areas are you are a fool like myself and use this capacitor for your selfmade amplifier and don't see this problem :)

Barzille, 2020-12-29:
I see a power cut off when I attach powercord to the powerbank, but that's ok for me.
And as mentioned before my main problem is that the shim as well as the system shutdown command powers down the raspi, but is not dropping the power supply so the raspi keeps on with a red light. But it's not possible to start up the device again without removing the power connection from the powerbank to the raspi.

What do you mean with the last sentence? German is fine too ;) @matthias-pelger

matthias-pelger, 2020-12-29:
Sorry for bad english, still in holiday mode :)

Also auf Deutsch:
Ich hatte das gleiche Problem, als ich einen Verstärker selber gelötet habe und zur spannungsstabilisierung des Verstärkers einen größeren Elko zwischen +5V und GND gehängt habe. Dieser verursachte genau das Phänomen: Raspi ging aus und sofort wieder an. Das liegt wohl irgendwie daran, dass der OnOffShim es nicht mag wenn er nach dem Ausschalten noch mit Spannnung "von hinten" versorgt wird. Bei mir hat eine einfache Diode, die verhindert das Strom vom Elko zurück zu Raspi fließt gelöst.

Daher der Vorschlag: prüfe mal alles was da an +5V und +3.3V an den Pins vom OnOffShim hängt ob da irgendwo noch ein Stromimpuls kommen könnte der den Shim wieder einschaltet.

Barzille, 2020-12-30:
Danke 👍 
Das hatte ich auch vermutet und alle anderen Pins, ich nutze das Extension Bord, auf andere gesteckt, so dass jetzt nur noch der shim und der hifiberry miniamp sich Pins teilen. 
meine Button LEDs werden zwar auch über 5V Dauerstrom versorgt, aber über andere Pins :(

matthias-pelger, 2020-12-31:
Hmm, wäre es nicht möglich, dass der Shim auf alle 5V Pins gleich reagiert? Würde vermuten, dass die geteilt ist und nicht getrennt wird.

Barzille, 2020-12-31:
Hm, du meinst ich sollte mal alle 5V Pins abkoppeln?

matthias-pelger, 2020-12-31:
genau, also testweise und dann einzeln wieder dran bis der fehler wieder auftritt.

Barzille, 2021-02-13:
Ich habe jetzt auch mal das Extension Board abgenommen und den OnOff Shim direkt auf den Pi gestöpselt, aber leider immer noch das gleiche Problem. Es sind auch keine weiteren anderen PINs angeschlossen/belegt.

Barzille, 2021-02-13:
Ich habe hier eine ähnliche Beschreibung gefunden, das werde ich mal überprüfen: https://forum-raspberrypi.de/forum/thread/44762-phoniebox-onoffshim-schaltet-den-rpi-nicht-aus/

Barzille, 2021-02-13:
Hat leider auch nicht geholfen, ich habe jetzt mal einen RPI4 bestellt, mal gucken ob das da besser läuft und ggf. checke ich noch mal die Lötstellen.

s-martin, 2022-12-24:
Closing for now as there was no activity for a long time.
Feel free to reopen a new issue, if the issue persists with current version 2.4 (or beta 3.2).

closed 2022-12-24 by s-martin