MiczFlor/RPi-Jukebox-RFID #1196: Different startup time

ralle211, 2020-12-12:
I have a rfid card to toggle Wi-Fi on off. 
If Wi-Fi is on, the startup time is much shorter that if the Wi-Fi is off!?

I use the latest release of Phoniebox 2.2 and the buster image (updated). 

I have a raspberry 3B. 

I have a fix IP address, disabled ipv6,...
I did all startup performance tweaks I found here:
https://splittscheid.de/faqs-zu-meiner-phoniebox/

Why is the startup time LONGER without Wi-Fi is on?

s-martin, 2020-12-12:
Please post the results of `tail -n 500 /var/log/syslog` to see what’s happening during startup. 
I guess RPi is trying to fetch something from the internet, e.g. current time which runs into a timeout, when no WiFi is available.

Do you have any network drives mounted to the Pi?

Do you use classic or Spotify edition?

ralle211, 2020-12-12:
Ok I will sometime today. 

Classic Edition 

No Network Folders etc. only local SD Card as source for mp3.