MiczFlor/RPi-Jukebox-RFID #1197: RFID-RC522 on pi zero not working after fresh install

NicoBleh, 2020-12-12:
## Bug

### What I did

Setup a pi zero w with I2S PCM5102 DAC and RFID-RC522 and Rasperian Pi QS Lite.
Used default install script for Buster to get the Phoniebox running.

### What happened

When placing a card on the RFID reader it does not read the card.

### I expected this to happen

When placing a card on the RFID reader it reads out the card and shows it at last read card.

### Further information that might help

```
Dec 12 20:07:42 raspberrypi rsyslogd:  [origin software="rsyslogd" swVersion="8.1901.0" x-pid="278" x-info="https://www.rsyslog.com"] rsyslogd was HUPed
Dec 12 20:07:42 raspberrypi systemd[1]: phoniebox-gpio-control.service: Main process exited, code=exited, status=1/FAILURE
Dec 12 20:07:42 raspberrypi systemd[1]: phoniebox-gpio-control.service: Failed with result 'exit-code'.
Dec 12 20:07:42 raspberrypi systemd[1]: logrotate.service: Succeeded.
Dec 12 20:07:43 raspberrypi systemd[1]: Started Rotate log files.
Dec 12 20:07:44 raspberrypi systemd[1]: Started Session 3 of user pi.
Dec 12 20:07:46 raspberrypi systemd[1]: phoniebox-gpio-control.service: Service RestartSec=3s expired, scheduling restart.
Dec 12 20:07:46 raspberrypi systemd[1]: phoniebox-gpio-control.service: Scheduled restart job, restart counter is at 2.
Dec 12 20:07:46 raspberrypi systemd[1]: phpsessionclean.service: Succeeded.
Dec 12 20:07:46 raspberrypi systemd[1]: Started Clean php session files.
Dec 12 20:07:46 raspberrypi systemd[1]: Stopped Phoniebox GPIO Control Service.
Dec 12 20:07:46 raspberrypi systemd[1]: Started Phoniebox GPIO Control Service.
Dec 12 20:07:52 raspberrypi PhonieboxGPIOControl[1110]: INFO:root:['VolumeControl', 'PrevNextControl', 'PlayPause', 'Shutdown', 'Volume0', 'VolumeUp', 'VolumeDown', 'NextSong', 'PrevSong', 'Halt']
Dec 12 20:07:52 raspberrypi PhonieboxGPIOControl[1110]: INFO:root:adding GPIO-Device, VolumeControl
Dec 12 20:07:52 raspberrypi PhonieboxGPIOControl[1110]: INFO:gpio_control:VolumeControl as TwoButtonControl
Dec 12 20:07:52 raspberrypi PhonieboxGPIOControl[1110]: INFO:GPIODevices.two_button_control:adding new action
Dec 12 20:07:52 raspberrypi PhonieboxGPIOControl[1110]: INFO:GPIODevices.simple_button:{}: set when_pressed
Dec 12 20:07:52 raspberrypi PhonieboxGPIOControl[1110]: INFO:GPIODevices.simple_button:add new action
Dec 12 20:07:53 raspberrypi PhonieboxGPIOControl[1110]: INFO:GPIODevices.simple_button:{}: set when_pressed
Dec 12 20:07:53 raspberrypi PhonieboxGPIOControl[1110]: INFO:GPIODevices.simple_button:add new action
Dec 12 20:07:53 raspberrypi PhonieboxGPIOControl[1110]: INFO:root:adding GPIO-Device, PrevNextControl
Dec 12 20:07:53 raspberrypi PhonieboxGPIOControl[1110]: INFO:root:adding TwoButtonControl
Dec 12 20:07:53 raspberrypi PhonieboxGPIOControl[1110]: INFO:GPIODevices.two_button_control:adding new action
Dec 12 20:07:53 raspberrypi PhonieboxGPIOControl[1110]: INFO:GPIODevices.simple_button:{}: set when_pressed
Dec 12 20:07:53 raspberrypi PhonieboxGPIOControl[1110]: INFO:GPIODevices.simple_button:add new action
Dec 12 20:07:54 raspberrypi mpd: zeroconf: No global port, disabling zeroconf
Dec 12 20:07:54 raspberrypi systemd[1]: Started Music Player Daemon.
Dec 12 20:07:54 raspberrypi systemd[1]: Starting Phoniebox Startup...
Dec 12 20:07:54 raspberrypi systemd[1]: man-db.service: Succeeded.
Dec 12 20:07:54 raspberrypi systemd[1]: Started Daily man-db regeneration.
Dec 12 20:07:55 raspberrypi bash[1204]: AUDIOFOLDERSPATH="/home/pi/RPi-Jukebox-RFID/shared/audiofolders"
Dec 12 20:07:55 raspberrypi bash[1204]: PLAYLISTSFOLDERPATH="/home/pi/RPi-Jukebox-RFID/playlists"
Dec 12 20:07:55 raspberrypi bash[1204]: SECONDSWIPE="RESTART"
Dec 12 20:07:55 raspberrypi bash[1204]: SECONDSWIPEPAUSE="2"
Dec 12 20:07:55 raspberrypi bash[1204]: SECONDSWIPEPAUSECONTROLS="ON"
Dec 12 20:07:55 raspberrypi bash[1204]: AUDIOIFACENAME="0"
Dec 12 20:07:55 raspberrypi bash[1204]: VOLUMEMANAGER="mpd"
Dec 12 20:07:55 raspberrypi bash[1204]: AUDIOVOLCHANGESTEP="3"
Dec 12 20:07:55 raspberrypi bash[1204]: AUDIOVOLMAXLIMIT="75"
Dec 12 20:07:55 raspberrypi bash[1204]: AUDIOVOLMINLIMIT="1"
Dec 12 20:07:55 raspberrypi bash[1204]: AUDIOVOLSTARTUP="75"
Dec 12 20:07:55 raspberrypi bash[1204]: VOLCHANGEIDLE="TRUE"
Dec 12 20:07:55 raspberrypi bash[1204]: IDLETIMESHUTDOWN="0"
Dec 12 20:07:55 raspberrypi bash[1204]: POWEROFFCMD="sudo poweroff"
Dec 12 20:07:55 raspberrypi bash[1204]: SHOWCOVER="ON"
Dec 12 20:07:55 raspberrypi bash[1204]: READWLANIPYN="OFF"
Dec 12 20:07:55 raspberrypi bash[1204]: EDITION="classic"
Dec 12 20:07:55 raspberrypi bash[1204]: LANG="de-DE"
Dec 12 20:07:55 raspberrypi bash[1204]: VERSION="2.2 - 305325d - master"
Dec 12 20:07:55 raspberrypi bash[1204]: CMDVOLUP=""
Dec 12 20:07:55 raspberrypi bash[1204]: CMDVOLDOWN=""
Dec 12 20:07:55 raspberrypi bash[1204]: CMDNEXT=""
[... 97 lines cut ...]
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

### Branch / Release

```
Refresh index: 100% (392/392), done.
On branch master
Your branch is up to date with 'origin/master'.
```
## Hardware

### RaspberryPi version

Raspberry Pi Zero W Rev 1.1

### RFID Reader

AZDelivery RFID Kit RC522

### Soundcard

iHaospace Interface I2S PCM5102 DAC Decoder GY-PCM5102 I2S Player Module for Raspberry Pi

[image]

MrReSc, 2020-12-14:
I use different hardware and can't tell you if the info in the wiki is correct, but have you read the page in the wiki about [special reders](https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/RFID-Reader-Special)? Maybe you can find help there.

NicoBleh, 2020-12-14:
The reader is the recommended one from the Readme.
https://amzn.to/2C7YZCZ
I had a look at the wiki about special readers and also followed the suggestions but no luck.
What i find interesting is that the provided wiring from the merchant differs from the wiki where the IRQ Pin is not connected.
https://cdn.shopify.com/s/files/1/1509/1638/files/RFID_RC522_Set_mit_Reader_Chip_und_Card_13_56MHz_Raspberry_Pi_Schematic.pdf?8907146477526484569
What now got my attention is when reading the logs:
```
Dec 14 21:28:18 raspberrypi systemd[1]: Stopping Phoniebox RFID-Reader Service...
Dec 14 21:28:18 raspberrypi systemd[1]: phoniebox-rfid-reader.service: Main process exited, code=killed, status=15/TERM
Dec 14 21:28:18 raspberrypi systemd[1]: phoniebox-rfid-reader.service: Succeeded.
Dec 14 21:28:18 raspberrypi systemd[1]: Stopped Phoniebox RFID-Reader Service.
Dec 14 21:28:26 raspberrypi systemd[1]: Started Phoniebox RFID-Reader Service.
Dec 14 21:28:29 raspberrypi daemon_rfid_reader.py[3961]: /usr/local/lib/python3.7/dist-packages/pirc522/rfid.py:90: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
```

seagullki, 2020-12-14:
the same with Pi4
[RF1D-RC522 keine Funktion #1204](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/1204)

seagullki, 2020-12-14:
with a Kabel from RC522 "IRQ" to PIN 18 (GPIO 24) it works!

NicoBleh, 2020-12-15:
@seagullki sadly not working with me.

Here is my wiring:

Color | RPi | GPIO | RC522 | Pin
------|---|-------|---------|---- 
red |	17|	Vcc|	Vcc|	    1	
green	|18	|GPIO24|	    IRQ	|    4	
violet	|19	|MOSI	   | MOSI	|6	
yellow	|20|	GND|	        GND	|    3	
blue	|21	|MISO	    |MISO|	5	
orange	|22	|GPIO25	|    RST	|    2	
grey	|23	|SCLK	   | SCK	   | 7	
white|	24	|GPIO8/CE0	|SDA |    8

Toqqi, 2020-12-16:
@NicoBleh 

Here's my wiring for RC522:

RPi | GPIO | RC522 | Pin
-- | -- | -- | --
**1** | Vcc | Vcc | 1
18 | GPIO24 | IRQ | 4
19 | MOSI | MOSI | 6
**6** | GND | GND | 3
21 | MISO | MISO | 5
22 | GPIO25 | RST | 2
23 | SCLK | SCK | 7
24 | GPIO8/CE0 | SDA | 8

NicoBleh, 2020-12-16:
@Toqqi ok that is the same except of vcc and ground which I take from rpi pin 17 and 20.
Have you done any special configuration?

s-martin, 2020-12-16:
Can you try anyway, if @Toqqi's config works for you, too?

NicoBleh, 2020-12-16:
@s-martin changed to @Toqqi wiring with the same result.

Toqqi, 2020-12-16:
@NicoBleh Which pins are used by the soundcard?

NicoBleh, 2020-12-16:
Ok so this is now what i tried.

- I disconnected the soundcard and only have the cardreader connected with @Toqqi 's pinout.
- Erased the SD-card and put Raspberry PI OSLite (32-bit) on it.
- Startup, enter raspi-config and activate spi
- installed `python-dev python-pip git`
- then `pip install spidev` and `pip install pi-rc522`
- `git clone https://github.com/ondryaso/pi-rc522.git`
- reboot
- executed `python /home/pi/pi-rc522/examples/Read.py`

It is showing 'Starting' but when I swipe a card or the keychain there is nothing read.

Not sure if I miss something or the reader is actually broken.
It does have a red led lit.

Any ideas?

NicoBleh, 2020-12-23:
Update:
After trying back and forth I bought the RFID Reader again, connected it with my initial wiring and it works out of the box.