MiczFlor/RPi-Jukebox-RFID #1198: Collect good, working configurations for ALSA equalizer

ChisSoc, 2020-12-12:
Setting up the ALSA Equalizer for the Raspberry is not straight forward and has some pitfalls (see https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Equalizer_systemwide_based_on_plugequal_for_mpd). Furthermore, there might different configurations needed for different hardware setups. I propose to collect some good working setups here. These can then be transferred into the wiki. This is a follow up of https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/862, which had the equalizer working in terminal, but not with MPD.

Please state 
- Exact audio hardware, how it is connected, and if board audio is disabled
- /etc/asound.conf
- /etc/mpd.conf - only the relevant audio_output section

Do read the wiki article, if trying to make a setup work.

ChisSoc, 2020-12-12:
I'll make a start. I have tested my configuration tonight on a **fresh install** of the Raspberry Pi + Phoniebox. All tests as described in the wiki are working. Here we go:

### Hardware
Hifiberry MiniAmp, via GPIO Header, on-board sound disabled

### /etc/asound.conf

```
pcm.hifiberry {
    type softvol
    slave.pcm equal
    control.name "Master"
    control.card 0
}

pcm.!default {
   type asym
   playback.pcm "hifiberry"
}

ctl.!default {
  type hw card 0
}

ctl.equal {
  type equal;
}

pcm.equalcore {
  type equal;
  slave.pcm "plughw:0,0";
}

pcm.equal {
  type plug;
  slave.pcm equalcore;
}
```
### /etc/mpd.conf
```
audio_output {
        type            "alsa"
        name            "HifiBerry Speakers"
        device          "hifiberry"
        auto_resample   "no"
        mixer_control   "Master"
}
```
### Notes
The HifiBerry MiniAmp does no have hardware volume control. That is why I have the Software Volume Manager (type softvol) in my asound.conf. If only playing via mpd, it is not strictly necessary as mpd also supports built-in software volume control. If using a sound card with hardware volume control, also not necessary. On the other hand, it also does not do any harm in these cases. So, might as well keep it for first experiments.

ChisSoc, 2020-12-13:
Here is a bit of background to the mpd.conf I am using above. Hope that helps in understanding it. 

```
audio_output {
        type            "alsa"
        name            "HifiBerry Speakers"
        # This device is used for playing music through
        #   Device is an ALSA device or a true hardware device (e.g. hw:0,0)
        #   Our new ALSA equalizer must go here, or the front-mode plug of the asound.conf 
        #   In the example configuration, the chain is "softvol" -> "equal". Meaning the name of the softvol, aptly named "hifiberry" goes here
        #   In the example configuration, pcm.!default is properly configured and points to "hifiberry" meaning, that "default" and "hifiberry" are identical
        #   Don't put hw:0,0 here, as that means MPD uses the sound card directly bypassing the ALSA configuration
        device          "hifiberry"
        # Prevent ALSA from doing any resampling if necessary. This should be done by MPD
        auto_resample   "no"
        # Mixer is how mpd handles volume control
        #   Your settings may vary depending on your hardware and there is multiple working configurations for identical hardware
        # mixer_type sets if volume control is handled by ALSA ("hardware"), or by MPD ("software")
        #   By default, the hardware mixer is used for devices which support it, and none for the others
        #   Attention: "none for others" means by default you have no volume control, unless you have a "softvol" setup in your ASLA config
        #   If you sound card does not have a hardware volume control (e.g. HifiBerry MiniAmp), you can set up a softvol pcm 
        #   in your asound.conf. In this case it acts like a hardware control. I prefer this option over MPD software control, as 
        #   is enables system-wide volume control (i.e. also for mpeg123, speaker-test, aplay - some of which are used for start-up sounds etc or during development)
        #   In the example configuration, we have an ALSA-based pcm (type softvol) for volume control -> Leave mixer_type at default
        # mixer_type    "hardware"
        # The name of the ALSA  mixer control 
        #   If you have a sound card with hardware volume control, choose the appropriate control
        #   Type amixer scontrols to get a list of available mixer controls and choose the right one ;-)
        #   Atttention: defaulting to PCM, so must be properly configured unless MPD handles volume control!
        #   In the example configuration, the ALSA-based pcm (type softvol) is named "Master"
        mixer_control   "Master"
        # Sets the ALSA mixer device name, defaulting to default which lets ALSA pick a value.
        #   In default, you should have a properly configured ctl.!default entry in your asound.conf
        #   In the example configuration ctl.!default is properly configured making it identical system-wide
        # mixer_device "hw:0,0"
}
```

s-martin, 2020-12-15:
Good idea!

Maybe it would make sense to move this to https://github.com/MiczFlor/RPi-Jukebox-RFID/discussions?discussions_q=category%3AIdeas

ChisSoc, 2020-12-15:
I now had some time  to test a setup a fresh Phoniebox install with a sound card that has hardware volume control. This time I tested with the **Spotify Edition.**

### Hardware

Hifiberry Amp2 (yes, the big one :-), via GPIO Header, on-board sound disabled
Amp2 has hardware volume control. This control is termed 'Digital'

### /etc/asound.conf
```
pcm.!default {
  type asym
  playback.pcm hifiberry
}

ctl.!default {
  type hw card 0
}

ctl.equal {
  type equal;
}

pcm.equalcore {
  type equal;
  slave.pcm "plughw:0,0";
}

pcm.hifiberry {
  type plug;
  slave.pcm equalcore
}
```
### /etc/mpd.conf
Not used as the Spotify Edition runs with mopidy. (If using the classic edition take the mpd.conf from the previous config)

### /etc/mopidy/mopidy.conf
No changes necessary

### Notes

- Mopidy runs under user mopidy. So, in order to changes equalizer settings, run `sudo -u mopidy alsamixer -D equal`
- Mopidy chooses the default ASLA device. So be sure to set up pcm.!default
- Mopidy in the default configuration supports only software volume control. Which means volume control of mopidy and `alsamixer -D default` are multiplied

ChisSoc, 2020-12-15:
Having done a few more experiments, I can confirm that the configuration from the Wiki works well with the Classic Edition, but needs modifications for the Spotify Edition. If there are no objections I will update the Wiki article to a version that works for both, Classic and Spotify Edition.

s-martin, 2020-12-15:
Yes, please update the wiki. Thanks for your effort!

0x0lli, 2020-12-23:
@ChisSoc  
Good idea. 
have not setup an equalizer yet but am about to with the AMP2 setup.
have you got working equalizer setups for the miniamp ?
Am wondering whether an Infineon merus 12070p would work aswell and how to configure that in asound.conf as I am not sure about whether its soft/hardware mixer.

What have you set as iFace "name"? I didnt find "Master" within your configuration- or is it irrelevant?

ChisSoc, 2020-12-24:
- For the AMP2, the iFace Name is "**Digital**" (this is defined by the hardware driver from HifiBerry)
- For the MiniAmp, the iFace Name ist "**Master**" (the newly created volume control)

I don't know about other sound cards. Try `$ amixer scontrols` for a complete list and pick the right one.