MiczFlor/RPi-Jukebox-RFID #1199: Music playback is interrupted for a few seconds every 20-30 seconds

SchreibAl, 2020-12-13:
Music playback is interrupted for a few seconds every 20-30 seconds. However, I have not changed the system since the last time I used it. I have also checked all the sound settings that I know of. Seems like it could be buffer problem. Can anyone help or give tips? 

Thanks in advance

s-martin, 2020-12-13:
Do you use classic or Spotify edition?

Are you playing local files or streams?

SchreibAl, 2020-12-13:
Thank you for your quick response.
-> I'm using the Spotify Edition playing streams.

s-martin, 2020-12-13:
Could you post results of `journalctl -u mopidy`?

Are you experiencing any issues, if you play spotify with another player from your network?

SchreibAl, 2020-12-13:
Oh no, just found the problem... my smartphone had connected to the bluetooth speaker I use for the phoniebox and kept getting in the way. Stupid mistake... but might help someone else....
Thank you anyway for your quick responses to my issue.

closed 2020-12-13 by SchreibAl