MiczFlor/RPi-Jukebox-RFID #1201: Spotify is connected - and nothing happens anymore

StefWick, 2020-12-13:
## Bug
Spotify does not do anything (playlist, artists, etc)

### What I did
I installed phoniebox on a fresh system. 

### What happened
When I access to the webinterface from the box, I can see everything. The status ist "connected", I see other Computers (if I logged in twice), but I can't search anything. For example, if I click on "update" for version 3.55.4, there is a grey sign on the right side, but it do nothing. There is online a circle turning his rounds - and nothing more. 

### I expected this to happen
Play music, find existing playlists, etc.

### Further information that might help
-- Logs begin at Sun 2020-12-13 22:28:15 CET, end at Sun 2020-12-13 23:24:34 CET. --
Dez 13 22:37:08 PunkRockBox systemd[1]: Starting Mopidy music server...
Dez 13 22:37:08 PunkRockBox systemd[1]: Started Mopidy music server.
Dez 13 22:37:26 PunkRockBox mopidy[563]: INFO     [MainThread] mopidy.__main__ Starting Mopidy 3.0.2
Dez 13 22:37:26 PunkRockBox mopidy[563]: INFO     [MainThread] mopidy.config Loading config from builtin defaults
Dez 13 22:37:26 PunkRockBox mopidy[563]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy.conf
Dez 13 22:37:26 PunkRockBox mopidy[563]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy-local.conf
Dez 13 22:37:26 PunkRockBox mopidy[563]: INFO     [MainThread] mopidy.config Loading config from file:///etc/mopidy/mopidy.conf
Dez 13 22:37:26 PunkRockBox mopidy[563]: INFO     [MainThread] mopidy.config Loading config from command line options
Dez 13 22:37:26 PunkRockBox mopidy[563]: ERROR    [MainThread] mopidy.internal.log Loading logging config '/etc/mopidy/logging.conf' failed. 'formatters'
Dez 13 22:37:27 PunkRockBox mopidy[563]: INFO     [MainThread] mopidy.__main__ Enabled extensions: local, stream, softwaremixer, iris, m3u, mpd, file, spotify, http
Dez 13 22:37:27 PunkRockBox mopidy[563]: INFO     [MainThread] mopidy.__main__ Disabled extensions: none
Dez 13 22:37:28 PunkRockBox mopidy[563]: INFO     [MainThread] mopidy.commands Starting Mopidy mixer: SoftwareMixer
Dez 13 22:37:28 PunkRockBox mopidy[563]: INFO     [MainThread] mopidy.commands Mixer volume set to 30
Dez 13 22:37:28 PunkRockBox mopidy[563]: INFO     [MainThread] mopidy.commands Starting Mopidy audio
Dez 13 22:37:28 PunkRockBox mopidy[563]: INFO     [MainThread] mopidy.commands Starting Mopidy backends: FileBackend, M3UBackend, StreamBackend, SpotifyBackend, LocalBackend
Dez 13 22:37:29 PunkRockBox mopidy[563]: INFO     [Audio-2] mopidy.audio.actor Audio output set to "alsasink"
Dez 13 22:37:30 PunkRockBox mopidy[563]: INFO     [SpotifyEventLoop] mopidy_spotify.backend Logged in to Spotify in online mode
Dez 13 22:37:34 PunkRockBox mopidy[563]: ERROR    [SpotifyBackend-6] mopidy_spotify.web Authorization failed, not attempting Spotify API request. Please get new credentials from https://www.mopidy.com/authenticate and/or restart Mopidy to
Dez 13 22:37:34 PunkRockBox mopidy[563]: ERROR    [SpotifyBackend-6] mopidy_spotify.web OAuth token refresh failed: invalid_client Client not known.
Dez 13 22:37:34 PunkRockBox mopidy[563]: ERROR    [SpotifyBackend-6] mopidy_spotify.web Failed to load Spotify user profile
Dez 13 22:37:34 PunkRockBox mopidy[563]: INFO     [MainThread] mopidy.commands Starting Mopidy core
Dez 13 22:37:35 PunkRockBox mopidy[563]: INFO     [MainThread] mopidy.commands Starting Mopidy frontends: HttpFrontend, IrisFrontend, MpdFrontend
Dez 13 22:37:35 PunkRockBox mopidy[563]: INFO     [HttpFrontend-12] mopidy.http.actor HTTP server running at [::ffff:0.0.0.0]:6680
Dez 13 22:37:35 PunkRockBox mopidy[563]: INFO     [IrisFrontend-13] mopidy_iris.core Starting Iris 3.45.0
Dez 13 22:37:35 PunkRockBox mopidy[563]: INFO     [MainThread] mopidy_mpd.actor MPD server running at [::ffff:0.0.0.0]:6600
Dez 13 22:37:35 PunkRockBox mopidy[563]: INFO     [MainThread] mopidy.commands Starting GLib mainloop
Dez 13 22:37:35 PunkRockBox mopidy[563]: INFO     [MpdSession-15] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36222
Dez 13 22:37:35 PunkRockBox mopidy[563]: INFO     [MpdSession-16] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36226
Dez 13 22:37:35 PunkRockBox mopidy[563]: INFO     [MpdSession-17] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36230
Dez 13 22:37:35 PunkRockBox mopidy[563]: INFO     [MpdSession-18] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36234
Dez 13 22:37:35 PunkRockBox mopidy[563]: INFO     [MpdSession-19] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36238
Dez 13 22:37:35 PunkRockBox mopidy[563]: INFO     [MpdSession-20] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36242
Dez 13 22:37:38 PunkRockBox mopidy[563]: INFO     [MpdSession-21] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36246
Dez 13 22:37:38 PunkRockBox mopidy[563]: INFO     [MpdSession-22] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36250
Dez 13 22:37:38 PunkRockBox mopidy[563]: INFO     [MpdSession-23] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36254
Dez 13 22:37:38 PunkRockBox mopidy[563]: INFO     [MpdSession-24] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36258
Dez 13 22:37:38 PunkRockBox mopidy[563]: INFO     [MpdSession-25] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36262
Dez 13 22:37:39 PunkRockBox mopidy[563]: INFO     [MpdSession-26] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36266
Dez 13 22:37:39 PunkRockBox mopidy[563]: INFO     [MpdSession-27] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36270
Dez 13 22:37:39 PunkRockBox mopidy[563]: INFO     [MpdSession-28] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36274
Dez 13 22:37:39 PunkRockBox mopidy[563]: INFO     [MpdSession-29] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36278
Dez 13 22:49:57 PunkRockBox mopidy[563]: INFO     [MpdSession-30] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36286
Dez 13 22:49:57 PunkRockBox mopidy[563]: INFO     [MpdSession-31] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36290
Dez 13 22:49:57 PunkRockBox mopidy[563]: INFO     [MpdSession-32] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36294
Dez 13 22:49:57 PunkRockBox mopidy[563]: INFO     [MpdSession-33] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36298
Dez 13 22:49:57 PunkRockBox mopidy[563]: INFO     [MpdSession-34] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36302
Dez 13 22:49:57 PunkRockBox mopidy[563]: INFO     [MpdSession-35] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36306
Dez 13 22:49:58 PunkRockBox mopidy[563]: INFO     [MpdSession-36] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36310
Dez 13 22:49:58 PunkRockBox mopidy[563]: INFO     [MpdSession-37] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36314
Dez 13 22:49:58 PunkRockBox mopidy[563]: INFO     [Thread-38] mopidy_iris.system Running system action 'upgrade'
Dez 13 22:49:58 PunkRockBox mopidy[563]: ERROR    [Thread-38] mopidy_iris.system Password-less access to file:///usr/local/lib/python3.7/dist-packages/mopidy_iris/system.sh was refused. Check your /etc/sudoers file.
Dez 13 22:49:58 PunkRockBox mopidy[563]: ERROR    [Thread-38] mopidy_iris.system Password-less access to file:///usr/local/lib/python3.7/dist-packages/mopidy_iris/system.sh was refused. Check your /etc/sudoers file.
Dez 13 22:50:00 PunkRockBox mopidy[563]: ERROR    [HttpServer] mopidy_iris.handlers Timeout during request
Dez 13 22:50:02 PunkRockBox mopidy[563]: ERROR    [HttpServer] mopidy_iris.handlers HTTP 401: UNAUTHORIZED
Dez 13 22:50:02 PunkRockBox mopidy[563]: ERROR    [LocalBackend-7] mopidy_local.library Error browsing local:directory?type=artist&role=albumartist: no such table: artist
Dez 13 22:50:03 PunkRockBox mopidy[563]: ERROR    [LocalBackend-7] mopidy_local.library Error browsing local:directory?type=album: no such table: album
[... 66 lines cut ...]

### Base image and version
Raspbian GNU/Linux 10 (buster)

### Branch / Release
Auf Branch master

### Installscript

normal

## Hardware

### RaspberryPi version

Raspberry Pi 3 Model B Rev 1.2pi

### RFID Reader

USB
### Soundcard

Miniamp

### Other notable hardware

s-martin, 2020-12-13:
Please check, if this fixes your issue:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#spotify-wont-play-but-after-scan-of-library-it-works

StefWick, 2020-12-13:
No. But: we can close this issue. After a lot of reading and searching - I forgot to authorized my account... :-(

Update is not working, playing music is working... Thanks!

0x0lli, 2020-12-14:
Authorization is what you do in order to get Spotify to work with MPD on the mopidy site that ultimately gets yourself a "client_id" and "client_secret", right ? I did that. then, I installed all three versions of  "2020-12-02-raspios-buster-armhf" (lite, regular and full) followed by script v2.2 and not one of them works for me. In Spotify edition, after I configured it the way I have successfully done before using the non-Spotify-version.
I can see the message "* is being played" when I try to playback local files in web-interface but can hear nothing 
In Iris it shows me every functionality Spotify offers via Iris but on hitting "play": "Mopidy:failed to add some tracks".
Any ideas ?

s-martin, 2020-12-14:
Did you check the Troubleshooting link in    https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/1201#issuecomment-744080207
if this works for you?

TheRealNic, 2020-12-16:
I actually have the same problem, the library scan did not resolve the problem. It actually worked with the previous version of the phoniebox...
What additional information should I provide?

TheRealNic, 2020-12-16:
BTW: I'm using the Spotify version with a hifiberry miniamp

s-martin, 2020-12-16:
Please post the result of `journalctl -u mopidy`

TheRealNic, 2020-12-16:
Log:
[log_mopidy.txt](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/5705876/log_mopidy.txt)

TheRealNic, 2020-12-16:
Ok, I just got the phoniebox to play spotify content: In the Iris settings under services I selected Spotify and needed to Login there (Pop up window opened up, there I could login). I'm not sure why this was necessary, but now it works

TheRealNic, 2020-12-16:
Ok, new Update: Spotify itself is working (so I am able to play Spotify via the Iris web interface) but not in the phoniebox webapp itself. Was anything changed when registering a Spotify Link?

0x0lli, 2020-12-17:
update from my side:
after "Dateien und Ordner" - "Musikbibliothek scannen"
it works for local files, which is what I am happy about in the first place.
I wanted to give away a phoniebox for my dad with spotify. Which is authorized but still gives me a:
"Mopidy: Failed to add some tracks" for whatever track I searched in the very nice interface.
Logfile is here, any help highly appreciated

[log.txt](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/5709549/log.txt)

0x0lli, 2020-12-17:
just did a manual update of  iris via:

sudo python3 -m pip install --upgrade Mopidy-Iris

afterwards, iris told me mopidy and pusher were inactive.

sudo systemctl enable mpd

kjeine konflikte in iris werden angezeigt, aber immer noch dieselbe Nachricht:
"Mopidy: Failed to add some tracks"

s-martin, 2020-12-18:
@0x0lli 
Could you please try
sudo pip3 install --upgrade --force-reinstall Mopidy-Iris and reboot?

0x0lli, 2020-12-18:
did that, looked a bit more promising - but nothing changes.
"mopidy-failed to add some tracks"
It does show me, thee default server

Tried to see if snapcast would work and it did in fact connect to my snapcast server can show basically everything but control ist not working the way it is supposed to, I can control another speaker in my network but am using the volume knob of the iris snapcast-client

In phoniebox-info it shows:
Mopidy Server Status:
Mopidy.Server: Connected
Mopidy.Service: active (running) since Fri 2020-12-18 17:34:30 CET; 3min 22s ago

thiyyas, 2020-12-18:
I face some similar issue and want to share what I found out up to now. Maybe this additional Informationen will help me and all others to get to a solution for the problem.

I don't have my Phoniebox here for some days (I'm getting help with the box I plan), so I cannot provide log files but still trying to find the reason for my problem. 

I found out, that I have this similar problem only in some networks.  
In my home wifi everything is fine, its also working properly in my parents wifi. 

But if connected with an android hotspot and another home Network (which is an sub-net (Vlan) of the mentioned Network from my parents) I cannot play spotify Tracks. 
When opening spotify interface via Phoniebox UI I also get  "mopidy-failed to add some tracks" error. 
In some log (sorry cannot remeber without having a look on it where exactly it was) i found a message that spotify login is not possible. But this only happens in some network, in other networks, with exact same setup of Phoniebox everything is working fine. 
I did not find any different settings for the network in the router. The raspberry is also allowed to use mopidy port. 

@0x0lli: maybe you can check if your spotify on the phoniebox will also work in other networks? 

@s-martin: do you have an idea, what could cause the problem with spotify in some networks?

s-martin, 2020-12-18:
@thiyyas sounds like this issue
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#spotify-wont-play-but-after-scan-of-library-it-works

The problem is that Mopidy connects only once after start to Spotify. If there’s no internet available at that time, this may fail.

please check the link above, if this fixes your issue.

thiyyas, 2020-12-18:
@s-martin: Badly that does not work either.  
I tried this a couple times, but scan of library did not help.
As soon as i get my Box back i will try the forced reinstall of mopidy-iris. Maybe this will help. 
I just hoped this observation on my System will lead to a solution for 0xOlli and later on for me too.

0x0lli, 2020-12-19:
@thiyyas: As I am using a RPI3A+, I use it with the WLAN of my router, that is less than a meter away and has never let me down for any application. Apart from that, I configured the Pi with sudo raspi-config to wait for LAN already anyway. I think that point is off the table but will have a look at it if the other attempts fail. there is still some 3B+ and 4's laying around but I didnt wanna use those.

If that is of any help to others: I am using an extension board that works and has the following components connected to it:, 
-a hifiberry-miniamp,
- two KY-040-rotary encoder (for rwd/fwd and vol_up/down), 
- one gpio button (for play/pause), 
- a SH-1106 0,96" OLED and 
- an onoffshim (I did not software install it until everything else is running)

0x0lli, 2020-12-19:
setup 2020-08-20-raspios-buster-armhf-lite
setup wait for network
installed script v2.2
config'ed everything to working condition.
then, ran suggested comment within:
https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/1201#issuecomment-748052260
sudo pip3 install --upgrade --force-reinstall Mopidy-Iris

then, scanned the library within phniebox web-interface
nothing changes

then, scanned the library within iris 
nothing changes, it appears to loop indefinite, just like the update function before the shell-based forced update

s-martin, 2020-12-19:
@0x0lli: just to clarify: did it work until the pip upgrade?

did you always try the lite version of Raspian?

0x0lli, 2020-12-21:
OS- // configged WiFi in - // updatedOS // updated Iris // Iris working

20201202(regular)- // yes // yes // yes // no
20201202full- // yes // yes // yes // no
20201202lite- // yes // yes // yes // no
20200820lite // yes // yes // no // no
20200820lite // yes // yes // yes // no
20200820full // yes // no// yes // no
20200820full // no // no // no // no
20201202full // no // no // no // no

I did: 
configure config.txt for miniamp / wlan_supplicant / ssh

configure WLAN in script, (tried also not)
set Audio-Interface to "Master" (because of hifiberry miniamp)
give script my requested spotify data
install MPD (tried also not)
install RFID 522

Console log:
```
"systemctl status mpd.socket":
-- The job identifier is 460.
Dec 19 16:34:18 pi3A-phoniebox mpd[21621]: exception: bind to '127.0.0.1:6600' failed (continuing anyway, because binding to '[::1]:6600' succeeded): Failed to bind socket: Address already in use
Dec 19 16:34:18 pi3A-phoniebox mpd[21621]: exception: Failed to set group 0: Operation not permitted
Dec 19 16:34:18 pi3A-phoniebox systemd[21610]: mpd.service: Main process exited, code=exited, status=1/FAILURE
```

Iris author wrote, the problem might be, Mopidy is not able to connect its service (bind socket).
I might have set the 0.0.0.0 to 127.0.0.1 but the problem was still the same.

is there **ANY** positive feedback of anyone running iris? 
I sure would love to know about their configuration
Iris' settings show me its connected and authorized, I can scan through the playlists that are connected.

Is that of any useful information ?

s-martin, 2020-12-21:
If you already have a working WiFi (e.g. when you ssh into the Pi) you don't need to configure WiFi during the installation script.

0x0lli, 2020-12-21:
I thought that maybe some values get pushed to other services

0x0lli, 2020-12-21:
I thought that maybe some values get pushed to other services, therefore tried it. As mentioned, to no extend.
Is there a feedback of a phoniebox-spotify user with a running system?
what hardware are they using ? (soundcard, RFID, GPIO...)

TheRealNic, 2020-12-21:
I am building 3 boxes at the moment and on 2 spotify via iris is working and on one it isn't... They all have the same configuration (PI 3B+, OnOff shim, HifiBerry miniamp)... I needed to grant access inside iris to spotify (as described in comment https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/1201#issuecomment-747091422), but on the third box this isn't working

0x0lli, 2020-12-21:
Thanks a lot for the info.
Hiw do you proceed with the installation? 
I configure config.txt for miniamp, wpa, ssh
Start 1st time, raspi-config for locale, hostname and passwd, wait for boot option, update, upgrade, install script. (no WiFi config there, install mpd, master as interface, rfid 522, leave path as set) 
Then I edit asound.conf, gpio_control.ini and install oled via splitti.

Even Virgin boxes without update or configuration don't work with iris for me. 
@TheRealNic  What raspian image do you use?

TheRealNic, 2020-12-21:
I am running 2020-08-20-raspios-buster-armhf-lite on all 3 boxes...

0x0lli, 2020-12-21:
I would be very interested in your install routine.

TheRealNic, 2020-12-21:
Pretty much the same as yours, I just don't install any oled since I am not using a display... But I will also try to play around with using different browsers in private sessions to access iris

s-martin, 2020-12-21:
What’s the output of `journalctl -u mopidy `?

0x0lli, 2020-12-21:
I dont understand why Iris interface shows me:
server;connected
Spotify:authenticated
but in those logs, it states, its not connected.
doesnt make the slightest bit sense to me.

went to get another authentification with spotify, logged off and then on with the new pair of keys.
stil:nothing.

journalctl -u mopidy:
```
-- Logs begin at Mon 2020-12-21 13:09:25 CET, end at Mon 2020-12-21 14:42:34 CET. --
Dec 21 13:09:43 pi3A128 systemd[1]: Starting Mopidy music server...
Dec 21 13:09:43 pi3A128 systemd[1]: Started Mopidy music server.
Dec 21 13:09:53 pi3A128 mopidy[4954]: INFO     [MainThread] mopidy.__main__ Starting Mopidy 3.0.2
Dec 21 13:09:53 pi3A128 mopidy[4954]: INFO     [MainThread] mopidy.config Loading config from builtin defaults
Dec 21 13:09:53 pi3A128 mopidy[4954]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy-local.conf
Dec 21 13:09:53 pi3A128 mopidy[4954]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy.conf
Dec 21 13:09:53 pi3A128 mopidy[4954]: INFO     [MainThread] mopidy.config Loading config from file:///etc/mopidy/mopidy.conf
Dec 21 13:09:53 pi3A128 mopidy[4954]: INFO     [MainThread] mopidy.config Loading config from command line options
Dec 21 13:09:53 pi3A128 mopidy[4954]: ERROR    [MainThread] mopidy.internal.log Loading logging config '/etc/mopidy/logging.conf' failed. 'formatters'
Dec 21 13:09:55 pi3A128 mopidy[4954]: INFO     [MainThread] mopidy.__main__ Enabled extensions: mpd, local, stream, softwaremixer, iris, file, m3u, http, spotify
Dec 21 13:09:55 pi3A128 mopidy[4954]: INFO     [MainThread] mopidy.__main__ Disabled extensions: none
Dec 21 13:09:57 pi3A128 mopidy[4954]: INFO     [MainThread] mopidy.commands Starting Mopidy mixer: SoftwareMixer
Dec 21 13:09:57 pi3A128 mopidy[4954]: INFO     [MainThread] mopidy.commands Mixer volume set to 30
Dec 21 13:09:57 pi3A128 mopidy[4954]: INFO     [MainThread] mopidy.commands Starting Mopidy audio
Dec 21 13:09:57 pi3A128 mopidy[4954]: INFO     [MainThread] mopidy.commands Starting Mopidy backends: FileBackend, M3UBackend, StreamBackend, SpotifyBackend, LocalBackend
Dec 21 13:09:57 pi3A128 mopidy[4954]: INFO     [Audio-2] mopidy.audio.actor Audio output set to "alsasink"
Dec 21 13:09:58 pi3A128 mopidy[4954]: ERROR    [SpotifyEventLoop] spotify.session Spotify login error: <ErrorType.BAD_USERNAME_OR_PASSWORD: 6>
Dec 21 13:09:58 pi3A128 mopidy[4954]: ERROR    [SpotifyBackend-6] mopidy_spotify.web Authorization failed, not attempting Spotify API request. Please get new credentials from https://www.mopidy.com/authenticate and/or restart Mopidy to resolve this problem.
Dec 21 13:09:58 pi3A128 mopidy[4954]: ERROR    [SpotifyBackend-6] mopidy_spotify.web OAuth token refresh failed: invalid_client client_id and client_secret set to same value.
Dec 21 13:09:58 pi3A128 mopidy[4954]: ERROR    [SpotifyBackend-6] mopidy_spotify.web Failed to load Spotify user profile
Dec 21 13:09:58 pi3A128 mopidy[4954]: INFO     [MainThread] mopidy.commands Starting Mopidy core
Dec 21 13:09:58 pi3A128 mopidy[4954]: INFO     [MainThread] mopidy.commands Starting Mopidy frontends: HttpFrontend, MpdFrontend, IrisFrontend
Dec 21 13:09:58 pi3A128 mopidy[4954]: INFO     [HttpFrontend-12] mopidy.http.actor HTTP server running at [::ffff:0.0.0.0]:6680
Dec 21 13:09:58 pi3A128 mopidy[4954]: INFO     [MainThread] mopidy_mpd.actor MPD server running at [::ffff:0.0.0.0]:6600
Dec 21 13:09:58 pi3A128 mopidy[4954]: INFO     [IrisFrontend-14] mopidy_iris.core Starting Iris 3.55.4
Dec 21 13:09:58 pi3A128 mopidy[4954]: INFO     [MainThread] mopidy.commands Starting GLib mainloop
Dec 21 13:09:58 pi3A128 mopidy[4954]: INFO     [IrisFrontend-14] mopidy.internal.path Creating dir file:///var/lib/mopidy/iris
Dec 21 13:09:58 pi3A128 mopidy[4954]: INFO     [MpdSession-15] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:58918
Dec 21 13:09:58 pi3A128 mopidy[4954]: INFO     [MpdSession-16] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:58922
Dec 21 13:09:58 pi3A128 mopidy[4954]: INFO     [MpdSession-17] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:58926
.
.
.
Dec 21 13:12:25 pi3A128 mopidy[4954]: INFO     [MpdSession-301] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:60062
Dec 21 13:12:25 pi3A128 mopidy[4954]: INFO     [MpdSession-302] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:60066
Dec 21 13:12:25 pi3A128 systemd[1]: Stopping Mopidy music server...
Dec 21 13:12:25 pi3A128 mopidy[4954]: INFO     [MainThread] mopidy.commands GLib mainloop got SIGTERM. Exiting...
Dec 21 13:12:25 pi3A128 mopidy[4954]: INFO     [MainThread] mopidy.commands Stopping Mopidy frontends
Dec 21 13:12:25 pi3A128 mopidy[4954]: INFO     [IrisFrontend-14] mopidy_iris.core Stopping Iris
Dec 21 13:12:25 pi3A128 mopidy[4954]: INFO     [MainThread] mopidy.commands Stopping Mopidy core
Dec 21 13:12:25 pi3A128 mopidy[4954]: INFO     [MainThread] mopidy.commands Stopping Mopidy backends
Dec 21 13:12:25 pi3A128 mopidy[4954]: INFO     [MainThread] mopidy.commands Stopping Mopidy audio
Dec 21 13:12:25 pi3A128 mopidy[4954]: INFO     [MainThread] mopidy.commands Stopping Mopidy mixer
Dec 21 13:12:25 pi3A128 systemd[1]: mopidy.service: Succeeded.
Dec 21 13:12:25 pi3A128 systemd[1]: Stopped Mopidy music server.
Dec 21 13:12:31 pi3A128 systemd[1]: Starting Mopidy music server...
Dec 21 13:12:31 pi3A128 systemd[1]: Started Mopidy music server.
[... cut ...]
Dec 21 13:17:01 pi3A128 mopidy[11177]: INFO     [SpotifyBackend-6] mopidy_spotify.lookup Failed to lookup 'spotify:track:4qzYESfn1UhuX0FJ5WpikT': Session must be logged in and online to load objects: <ConnectionState.LOGGED_OUT: 0>
Dec 21 13:17:03 pi3A128 mopidy[11177]: INFO     [MpdSession-281] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:32946
.
.
.
Dec 21 13:17:04 pi3A128 mopidy[11177]: INFO     [MpdSession-282] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:32950
Dec 21 13:17:04 pi3A128 mopidy[11177]: INFO     [MpdSession-283] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:32954
.
.
.

``

s-martin, 2020-12-21:
I also don’t know why Iris interface shows connected, but the errors

```
Dec 21 13:12:35 pi3A128 mopidy[11177]: ERROR    [SpotifyBackend-6] mopidy_spotify.web Authorization failed, not attempting Spotify API request. Please get new credentials from https://www.mopidy.com/authenticate and/or restart Mopidy to resolve this problem.
Dec 21 13:12:35 pi3A128 mopidy[11177]: ERROR    [SpotifyBackend-6] mopidy_spotify.web OAuth token refresh failed: invalid_client client_id and client_secret set to same value.
```

are pretty clear.

it also says client_id and client_secret have same value which is wrong.

0x0lli, 2020-12-22:
Thanks @s-martin  for the troubleshooting!!!

edited all three files that were loaded - this one contained the login information:
/etc/mopidy/mopidy.conf

still, the error showed, it was not connected:
Dec 22 16:16:07 pi3A128 mopidy[4183]: ERROR    [SpotifyEventLoop] spotify.session Spotify login error: <ErrorType.BAD_USERNAME_OR_PASSWORD: 6>

this gave the final hint:
It is rather strange that I can use my email-address or regular Spotify Username to log into the web interface.
Then, the cryptic username Spotify gives to its users is the only one requested to connect another (which?) module to Spotify and playback audio there.

One can find the requested login name in the logfile output: 
[SpotifyBackend-6] mopidy_spotify.web Logged into Spotify Web API as *****************************

I think there should be a note into the installer, as I might not be the last one that uses the "wrong" login credentials. Also, it could be my fault to not have double checked the pasted key (client_id == client_secret) but I was fairy certain, all went fine. After so many trials it is nice to have everything working.
Thanks you very much @s-martin !
My dad is going to be a happy man on christmas.

```
-- Logs begin at Tue 2020-12-22 16:15:34 CET, end at Tue 2020-12-22 16:18:01 CET. --
Dec 22 16:15:52 pi3A128 systemd[1]: Starting Mopidy music server...
Dec 22 16:15:52 pi3A128 systemd[1]: Started Mopidy music server.
Dec 22 16:16:03 pi3A128 mopidy[4183]: INFO     [MainThread] mopidy.__main__ Starting Mopidy 3.0.2
Dec 22 16:16:03 pi3A128 mopidy[4183]: INFO     [MainThread] mopidy.config Loading config from builtin defaults
Dec 22 16:16:03 pi3A128 mopidy[4183]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy-local.conf
Dec 22 16:16:03 pi3A128 mopidy[4183]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy.conf
Dec 22 16:16:03 pi3A128 mopidy[4183]: INFO     [MainThread] mopidy.config Loading config from file:///etc/mopidy/mopidy.conf
Dec 22 16:16:03 pi3A128 mopidy[4183]: INFO     [MainThread] mopidy.config Loading config from command line options
Dec 22 16:16:03 pi3A128 mopidy[4183]: ERROR    [MainThread] mopidy.internal.log Loading logging config '/etc/mopidy/logging.conf' failed. 'formatters'
Dec 22 16:16:05 pi3A128 mopidy[4183]: INFO     [MainThread] mopidy.__main__ Enabled extensions: stream, iris, softwaremixer, mpd, spotify, local, file, m3u, http
Dec 22 16:16:05 pi3A128 mopidy[4183]: INFO     [MainThread] mopidy.__main__ Disabled extensions: none
Dec 22 16:16:07 pi3A128 mopidy[4183]: INFO     [MainThread] mopidy.commands Starting Mopidy mixer: SoftwareMixer
Dec 22 16:16:07 pi3A128 mopidy[4183]: INFO     [MainThread] mopidy.commands Mixer volume set to 30
Dec 22 16:16:07 pi3A128 mopidy[4183]: INFO     [MainThread] mopidy.commands Starting Mopidy audio
Dec 22 16:16:07 pi3A128 mopidy[4183]: INFO     [MainThread] mopidy.commands Starting Mopidy backends: FileBackend, M3UBackend, StreamBackend, SpotifyBackend, LocalBackend
Dec 22 16:16:07 pi3A128 mopidy[4183]: INFO     [Audio-2] mopidy.audio.actor Audio output set to "alsasink"
Dec 22 16:16:07 pi3A128 mopidy[4183]: ERROR    [SpotifyEventLoop] spotify.session Spotify login error: <ErrorType.BAD_USERNAME_OR_PASSWORD: 6>
Dec 22 16:16:21 pi3A128 mopidy[4183]: INFO     [SpotifyBackend-6] mopidy_spotify.web Logged into Spotify Web API as        #  edited, correct signature
Dec 22 16:16:58 pi3A128 mopidy[4183]: INFO     [SpotifyBackend-6] mopidy_spotify.playlists Refreshed 69 Spotify playlists
Dec 22 16:16:58 pi3A128 mopidy[4183]: INFO     [MainThread] mopidy.commands Starting Mopidy core
Dec 22 16:16:58 pi3A128 mopidy[4183]: INFO     [MainThread] mopidy.commands Starting Mopidy frontends: HttpFrontend, MpdFrontend, IrisFrontend
Dec 22 16:16:58 pi3A128 mopidy[4183]: INFO     [HttpFrontend-12] mopidy.http.actor HTTP server running at [::ffff:0.0.0.0]:6680
Dec 22 16:16:58 pi3A128 mopidy[4183]: INFO     [MainThread] mopidy_mpd.actor MPD server running at [::ffff:0.0.0.0]:6600
Dec 22 16:16:58 pi3A128 mopidy[4183]: INFO     [IrisFrontend-14] mopidy_iris.core Starting Iris 3.55.4
Dec 22 16:16:58 pi3A128 mopidy[4183]: INFO     [MainThread] mopidy.commands Starting GLib mainloop
Dec 22 16:16:58 pi3A128 mopidy[4183]: INFO     [MpdSession-15] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52570

```

s-martin, 2020-12-22:
So I guess for everyone asking in this particular issue it is now working, @TheRealNic opened a separate issue. So I close this issue.

please feel free to open a new issue or a question in Discussions, if you still have problems.

closed 2020-12-22 by s-martin