MiczFlor/RPi-Jukebox-RFID #1202: 🐛 | BUG SUMMARY

Akeman1, 2020-12-14:
## Bug: Sound does Not Work after first Installation.

### What I did: I tried all solution for Sound Problems, but i can't fix the Error. I also tried different Options during the Installation...
I use a Soundcard:
https://smile.amazon.de/dp/B00C7LXUDY/ref=cm_sw_r_wa_apa_fabc_DN31FbK9VMHRK?_encoding=UTF8&psc=1
And don't know how to configure it... correct.

I use this speakers: https://smile.amazon.de/dp/B01HDR5EIK/ref=cm_sw_r_wa_apa_fabc_JI51Fb5X6HJRT?_encoding=UTF8&psc=1

With an external powerbank.

### What happened

### I expected this to happen
Sound works within the VLC Media Player, but Not within the phoniebox

### Further information that might help

## Software

### Base image and version

### Branch / Release

### Installscript

## Hardware

### RaspberryPi version

### RFID Reader

### Soundcard

### Other notable hardware

s-martin, 2020-12-14:
Do you use classic or Spotify edition?

Have you tried https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#raspberry-pi-update-may-2020-broke-my-audio

What's the output of `tail -n 500 /var/log/syslog`?

Akeman1, 2020-12-14:
How can i edit entries in the Terminal? Like defaults.ctl.card 0 to defaults.ctl.card 1? Or can i find the entries within the File Manager?

That's the output of tail -n 500 /var/log/syslog:

session New MPD connection from [::ffff:127.0.0.1]:50806
Dec 14 18:40:02 raspberrypi mopidy[536]: INFO     [MpdSession-15633] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50810
Dec 14 18:40:02 raspberrypi mopidy[536]: INFO     [MpdSession-15634] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50814
Dec 14 18:40:02 raspberrypi mopidy[536]: INFO     [MpdSession-15635] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50818
Dec 14 18:40:07 raspberrypi mopidy[536]: INFO     [MpdSession-15636] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50822
Dec 14 18:40:07 raspberrypi mopidy[536]: INFO     [MpdSession-15637] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50826
Dec 14 18:40:07 raspberrypi mopidy[536]: INFO     [MpdSession-15638] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50830
Dec 14 18:40:07 raspberrypi mopidy[536]: INFO     [MpdSession-15639] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50834
Dec 14 18:40:07 raspberrypi mopidy[536]: INFO     [MpdSession-15640] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50838
Dec 14 18:40:08 raspberrypi mopidy[536]: INFO     [MpdSession-15641] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50842
Dec 14 18:40:08 raspberrypi mopidy[536]: INFO     [MpdSession-15642] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50846
Dec 14 18:40:08 raspberrypi mopidy[536]: INFO     [MpdSession-15643] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50850
Dec 14 18:40:13 raspberrypi mopidy[536]: INFO     [MpdSession-15644] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50856
Dec 14 18:40:13 raspberrypi mopidy[536]: INFO     [MpdSession-15645] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50860
Dec 14 18:40:13 raspberrypi mopidy[536]: INFO     [MpdSession-15646] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50864
Dec 14 18:40:13 raspberrypi mopidy[536]: INFO     [MpdSession-15647] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50868
Dec 14 18:40:13 raspberrypi mopidy[536]: INFO     [MpdSession-15648] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50872
Dec 14 18:40:13 raspberrypi mopidy[536]: INFO     [MpdSession-15649] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50876
Dec 14 18:40:13 raspberrypi mopidy[536]: INFO     [MpdSession-15650] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50880
Dec 14 18:40:13 raspberrypi mopidy[536]: INFO     [MpdSession-15651] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50884
Dec 14 18:40:18 raspberrypi mopidy[536]: INFO     [MpdSession-15652] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50888
Dec 14 18:40:18 raspberrypi mopidy[536]: INFO     [MpdSession-15653] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50892
Dec 14 18:40:18 raspberrypi mopidy[536]: INFO     [MpdSession-15654] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50896
Dec 14 18:40:18 raspberrypi mopidy[536]: INFO     [MpdSession-15655] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50900
Dec 14 18:40:19 raspberrypi mopidy[536]: INFO     [MpdSession-15656] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50904
Dec 14 18:40:19 raspberrypi mopidy[536]: INFO     [MpdSession-15657] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50908
Dec 14 18:40:19 raspberrypi mopidy[536]: INFO     [MpdSession-15658] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50912
Dec 14 18:40:19 raspberrypi mopidy[536]: INFO     [MpdSession-15659] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50916
Dec 14 18:40:24 raspberrypi mopidy[536]: INFO     [MpdSession-15660] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50922
Dec 14 18:40:24 raspberrypi mopidy[536]: INFO     [MpdSession-15661] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50926
Dec 14 18:40:24 raspberrypi mopidy[536]: INFO     [MpdSession-15662] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50930
Dec 14 18:40:24 raspberrypi mopidy[536]: INFO     [MpdSession-15663] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50934
Dec 14 18:40:24 raspberrypi mopidy[536]: INFO     [MpdSession-15664] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50938
Dec 14 18:40:24 raspberrypi mopidy[536]: INFO     [MpdSession-15665] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50942
Dec 14 18:40:24 raspberrypi mopidy[536]: INFO     [MpdSession-15666] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50946
Dec 14 18:40:24 raspberrypi mopidy[536]: INFO     [MpdSession-15667] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50950
Dec 14 18:40:29 raspberrypi mopidy[536]: INFO     [MpdSession-15668] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50954
Dec 14 18:40:29 raspberrypi mopidy[536]: INFO     [MpdSession-15669] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50958
Dec 14 18:40:30 raspberrypi mopidy[536]: INFO     [MpdSession-15670] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50962
[... cut ...]
Dec 14 18:44:54 raspberrypi mopidy[536]: INFO     [MpdSession-16128] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52888
Dec 14 18:44:54 raspberrypi mopidy[536]: INFO     [MpdSession-16129] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52892
Dec 14 18:44:54 raspberrypi mopidy[536]: INFO     [MpdSession-16130] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52896
Dec 14 18:44:54 raspberrypi mopidy[536]: INFO     [MpdSession-16131] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:52900
pi@raspberrypi:~ $

s-martin, 2020-12-14:
Please edit with `sudo nano /etc/asound.conf` and add the entries there.

Save: `CTRL+O`
Exit: `CTRL+X`

Please check https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#audio-is-not-working

Section a)

Please show also the output of `journalctl -u mopidy`

Akeman1, 2020-12-14:
cat /proc/asound/modules (ouput): 

pi@raspberrypi:~ $ cat /proc/asound/modules
 0 snd_bcm2835
 1 snd_bcm2835
 2 snd_usb_audio

So I follwed https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#audio-is-not-working  
Section a) and replaced for my case 1 with 2.

I also did the second part and changed everything from 1 to 2? Is this correct?
Still not working...
Thanks for your help!! 

Output of journalctl -u mopidy:

pi@raspberrypi:~ $ journalctl -u mopidy
-- Logs begin at Mon 2020-12-14 20:52:51 CET, end at Mon 2020-12-14 20:58:31 CET
Dec 14 20:52:56 raspberrypi systemd[1]: Starting Mopidy music server...
Dec 14 20:52:57 raspberrypi systemd[1]: Started Mopidy music server.
Dec 14 20:53:09 raspberrypi mopidy[531]: INFO     [MainThread] mopidy.__main__ S
Dec 14 20:53:09 raspberrypi mopidy[531]: INFO     [MainThread] mopidy.config Loa
Dec 14 20:53:09 raspberrypi mopidy[531]: INFO     [MainThread] mopidy.config Loa
Dec 14 20:53:09 raspberrypi mopidy[531]: INFO     [MainThread] mopidy.config Loa
Dec 14 20:53:09 raspberrypi mopidy[531]: INFO     [MainThread] mopidy.config Loa
Dec 14 20:53:09 raspberrypi mopidy[531]: INFO     [MainThread] mopidy.config Loa
Dec 14 20:53:09 raspberrypi mopidy[531]: ERROR    [MainThread] mopidy.internal.l
Dec 14 20:53:11 raspberrypi mopidy[531]: INFO     [MainThread] mopidy.__main__ E
Dec 14 20:53:11 raspberrypi mopidy[531]: INFO     [MainThread] mopidy.__main__ D
Dec 14 20:53:13 raspberrypi mopidy[531]: INFO     [MainThread] mopidy.commands S
Dec 14 20:53:13 raspberrypi mopidy[531]: INFO     [MainThread] mopidy.commands M
Dec 14 20:53:13 raspberrypi mopidy[531]: INFO     [MainThread] mopidy.commands S
Dec 14 20:53:13 raspberrypi mopidy[531]: INFO     [MainThread] mopidy.commands S
Dec 14 20:53:13 raspberrypi mopidy[531]: INFO     [Audio-2] mopidy.audio.actor A
Dec 14 20:53:13 raspberrypi mopidy[531]: ERROR    [SpotifyEventLoop] spotify.ses
Dec 14 20:53:14 raspberrypi mopidy[531]: INFO     [SpotifyBackend-6] mopidy_spot
Dec 14 20:53:14 raspberrypi mopidy[531]: INFO     [SpotifyBackend-6] mopidy_spot
Dec 14 20:53:14 raspberrypi mopidy[531]: INFO     [MainThread] mopidy.commands S
Dec 14 20:53:14 raspberrypi mopidy[531]: INFO     [MainThread] mopidy.commands S
Dec 14 20:53:14 raspberrypi mopidy[531]: INFO     [HttpFrontend-12] mopidy.http.
lines 1-23...skipping...
-- Logs begin at Mon 2020-12-14 20:52:51 CET, end at Mon 2020-12-14 20:58:31 CET. --
Dec 14 20:52:56 raspberrypi systemd[1]: Starting Mopidy music server...
Dec 14 20:52:57 raspberrypi systemd[1]: Started Mopidy music server.
Dec 14 20:53:09 raspberrypi mopidy[531]: INFO     [MainThread] mopidy.__main__ Starting Mopidy 3.0.2
Dec 14 20:53:09 raspberrypi mopidy[531]: INFO     [MainThread] mopidy.config Loading config from builtin defaults
Dec 14 20:53:09 raspberrypi mopidy[531]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy.conf
Dec 14 20:53:09 raspberrypi mopidy[531]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy-local.conf
Dec 14 20:53:09 raspberrypi mopidy[531]: INFO     [MainThread] mopidy.config Loading config from file:///etc/mopidy/mopidy.conf
Dec 14 20:53:09 raspberrypi mopidy[531]: INFO     [MainThread] mopidy.config Loading config from command line options
Dec 14 20:53:09 raspberrypi mopidy[531]: ERROR    [MainThread] mopidy.internal.log Loading logging config '/etc/mopidy/logging.conf' failed. 'formatters'
Dec 14 20:53:11 raspberrypi mopidy[531]: INFO     [MainThread] mopidy.__main__ Enabled extensions: iris, spotify, http, local, stream, mpd, m3u, file, softwaremixer
Dec 14 20:53:11 raspberrypi mopidy[531]: INFO     [MainThread] mopidy.__main__ Disabled extensions: none
Dec 14 20:53:13 raspberrypi mopidy[531]: INFO     [MainThread] mopidy.commands Starting Mopidy mixer: SoftwareMixer
Dec 14 20:53:13 raspberrypi mopidy[531]: INFO     [MainThread] mopidy.commands Mixer volume set to 30
Dec 14 20:53:13 raspberrypi mopidy[531]: INFO     [MainThread] mopidy.commands Starting Mopidy audio
Dec 14 20:53:13 raspberrypi mopidy[531]: INFO     [MainThread] mopidy.commands Starting Mopidy backends: FileBackend, M3UBackend, StreamBackend, SpotifyBackend, LocalBackend
Dec 14 20:53:13 raspberrypi mopidy[531]: INFO     [Audio-2] mopidy.audio.actor Audio output set to "alsasink"
Dec 14 20:53:13 raspberrypi mopidy[531]: ERROR    [SpotifyEventLoop] spotify.session Spotify login error: <ErrorType.USER_NEEDS_PREMIUM: 15>
Dec 14 20:53:14 raspberrypi mopidy[531]: INFO     [SpotifyBackend-6] mopidy_spotify.web Logged into Spotify Web API as gr5ekdyzl4dkl0jasmo77sfvp
Dec 14 20:53:14 raspberrypi mopidy[531]: INFO     [SpotifyBackend-6] mopidy_spotify.playlists Refreshed 1 Spotify playlists
Dec 14 20:53:14 raspberrypi mopidy[531]: INFO     [MainThread] mopidy.commands Starting Mopidy core
Dec 14 20:53:14 raspberrypi mopidy[531]: INFO     [MainThread] mopidy.commands Starting Mopidy frontends: HttpFrontend, IrisFrontend, MpdFrontend
Dec 14 20:53:14 raspberrypi mopidy[531]: INFO     [HttpFrontend-12] mopidy.http.actor HTTP server running at [::ffff:0.0.0.0]:6680
Dec 14 20:53:14 raspberrypi mopidy[531]: INFO     [IrisFrontend-13] mopidy_iris.core Starting Iris 3.45.0
Dec 14 20:53:14 raspberrypi mopidy[531]: INFO     [MainThread] mopidy_mpd.actor MPD server running at [::ffff:0.0.0.0]:6600
Dec 14 20:53:14 raspberrypi mopidy[531]: INFO     [MainThread] mopidy.commands Starting GLib mainloop
[... cut ...]
Dec 14 20:53:42 raspberrypi mopidy[531]: INFO     [MpdSession-39] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:35342
Dec 14 20:53:42 raspberrypi mopidy[531]: INFO     [MpdSession-40] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:35346
Dec 14 20:53:42 raspberrypi mopidy[531]: INFO     [MpdSession-41] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:35350
Dec 14 20:53:42 raspberrypi mopidy[531]: INFO     [MpdSession-42] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:35354

s-martin, 2020-12-14:
Replacing to 2 should work, maybe you need a reboot.

But I saw that you probably don’t have a premium Spotify account that’s why it may not work, too.

s-martin, 2020-12-14:
Did you try playing local files or streaming?

Local files should work, I guess for streaming you need a premium Spotify account.

BenutzerKeyna, 2020-12-18:
Hi Akeman,

did you solve your problem? I'm in the complete same situation.
@s-martin: I tried all the tips. Thanks a lot!

Version 2.2; incl Spotify
I did the one line installation 5 times now; 2 times sd flash

s-martin, 2020-12-18:
@BenutzerKeyna please post the results of `journalctl` -u mopidy`

BenutzerKeyna, 2020-12-18:
results of journalctl:
pi@raspberrypi:~ $ journalctl -u mopidy
-- Logs begin at Fri 2020-12-18 18:02:07 CET, end at Fri 2020-12-18 21:57:18 CET
Dec 18 18:02:13 raspberrypi systemd[1]: Starting Mopidy music server...
Dec 18 18:02:13 raspberrypi systemd[1]: Started Mopidy music server.
Dec 18 18:02:31 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.__main__ S
Dec 18 18:02:31 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.config Loa
Dec 18 18:02:31 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.config Loa
Dec 18 18:02:31 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.config Loa
Dec 18 18:02:31 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.config Loa
Dec 18 18:02:31 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.config Loa
Dec 18 18:02:31 raspberrypi mopidy[524]: ERROR    [MainThread] mopidy.internal.l
Dec 18 18:02:33 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.__main__ E
Dec 18 18:02:33 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.__main__ D
Dec 18 18:02:34 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.commands S
Dec 18 18:02:34 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.commands M
Dec 18 18:02:34 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.commands S
Dec 18 18:02:34 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.commands S
Dec 18 18:02:35 raspberrypi mopidy[524]: INFO     [Audio-2] mopidy.audio.actor A
Dec 18 18:02:36 raspberrypi mopidy[524]: INFO     [SpotifyEventLoop] mopidy_spot
Dec 18 18:02:36 raspberrypi mopidy[524]: INFO     [SpotifyBackend-6] mopidy_spot
Dec 18 18:02:36 raspberrypi mopidy[524]: INFO     [SpotifyBackend-6] mopidy_spot
Dec 18 18:02:36 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.commands S
Dec 18 18:02:36 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.commands S
Dec 18 18:02:36 raspberrypi mopidy[524]: INFO     [HttpFrontend-13] mopidy.http.
lines 1-23...skipping...
-- Logs begin at Fri 2020-12-18 18:02:07 CET, end at Fri 2020-12-18 21:57:18 CET
Dec 18 18:02:13 raspberrypi systemd[1]: Starting Mopidy music server...
Dec 18 18:02:13 raspberrypi systemd[1]: Started Mopidy music server.
Dec 18 18:02:31 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.__main__ S
Dec 18 18:02:31 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.config Loa
Dec 18 18:02:31 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.config Loa
Dec 18 18:02:31 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.config Loa
Dec 18 18:02:31 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.config Loa
Dec 18 18:02:31 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.config Loa
Dec 18 18:02:31 raspberrypi mopidy[524]: ERROR    [MainThread] mopidy.internal.l
Dec 18 18:02:33 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.__main__ E
Dec 18 18:02:33 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.__main__ D
Dec 18 18:02:34 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.commands S
Dec 18 18:02:34 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.commands M
Dec 18 18:02:34 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.commands S
Dec 18 18:02:34 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.commands S
Dec 18 18:02:35 raspberrypi mopidy[524]: INFO     [Audio-2] mopidy.audio.actor A
Dec 18 18:02:36 raspberrypi mopidy[524]: INFO     [SpotifyEventLoop] mopidy_spot
Dec 18 18:02:36 raspberrypi mopidy[524]: INFO     [SpotifyBackend-6] mopidy_spot
Dec 18 18:02:36 raspberrypi mopidy[524]: INFO     [SpotifyBackend-6] mopidy_spot
Dec 18 18:02:36 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.commands S
Dec 18 18:02:36 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.commands S
Dec 18 18:02:36 raspberrypi mopidy[524]: INFO     [HttpFrontend-13] mopidy.http.
Dec 18 18:02:36 raspberrypi mopidy[524]: INFO     [IrisFrontend-14] mopidy_iris.
Dec 18 18:02:36 raspberrypi mopidy[524]: INFO     [MainThread] mopidy_mpd.actor 
Dec 18 18:02:36 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.commands S
Dec 18 18:02:36 raspberrypi mopidy[524]: INFO     [MpdSession-16] mopidy_mpd.ses
Dec 18 18:02:36 raspberrypi mopidy[524]: INFO     [MpdSession-17] mopidy_mpd.ses
Dec 18 18:02:36 raspberrypi mopidy[524]: INFO     [MpdSession-18] mopidy_mpd.ses
Dec 18 18:02:36 raspberrypi mopidy[524]: INFO     [MpdSession-19] mopidy_mpd.ses
Dec 18 18:02:41 raspberrypi mopidy[524]: INFO     [MpdSession-20] mopidy_mpd.ses
lines 1-31/31 (END)
-- Logs begin at Fri 2020-12-18 18:02:07 CET, end at Fri 2020-12-18 21:57:18 CET. --
Dec 18 18:02:13 raspberrypi systemd[1]: Starting Mopidy music server...
Dec 18 18:02:13 raspberrypi systemd[1]: Started Mopidy music server.
Dec 18 18:02:31 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.__main__ Starting Mopidy 3.0.2
Dec 18 18:02:31 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.config Loading config from builtin defaults
Dec 18 18:02:31 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy.conf
Dec 18 18:02:31 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy-local.conf
Dec 18 18:02:31 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.config Loading config from file:///etc/mopidy/mopidy.conf
Dec 18 18:02:31 raspberrypi mopidy[524]: INFO     [MainThread] mopidy.config Loading config from command line options
[... cut ...]
Dec 18 18:02:36 raspberrypi mopidy[524]: INFO     [MpdSession-17] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:55298
Dec 18 18:02:36 raspberrypi mopidy[524]: INFO     [MpdSession-18] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:55302
Dec 18 18:02:36 raspberrypi mopidy[524]: INFO     [MpdSession-19] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:55306
Dec 18 18:02:41 raspberrypi mopidy[524]: INFO     [MpdSession-20] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:55312

Thanks in advance
br

s-martin, 2020-12-18:
Looks good. Strange?

s-martin, 2020-12-18:
@BenutzerKeyna 
Could you please try
`sudo pip3 install --upgrade --force-reinstall Mopidy-Iris` and reboot?

BenutzerKeyna, 2020-12-18:
@s-martin 
Done,  but no difference, sorry

s-martin, 2020-12-18:
Could you explain a little more, what works and what doesn’t?

BenutzerKeyna, 2020-12-18:
that works very well:
Web interface, Spotify and switch over to play (I see the time of the song is running)

Audio Output: 
usb interface 
over Phonie: sounds nothing 
over Browser/Youtube: works very well 

Rfid reader: 
over Phonie-survace only "piep", no number, no chip read in
if I select the browser line, the chip code will read

s-martin, 2020-12-18:
Could you check, if this FAQ helps?
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#audio-is-not-working

BenutzerKeyna, 2020-12-20:
it works now! Thanks a lot @s-martin 

I did the instruction link two times before. But I understood this script wrong... I thought the change to #1 change my port, not the position...
my config was:
   $ cat /proc/asound/modules
      0 snd_bcm2835
      1 snd_bcm2835
      2 snd_usb_audio
-> so i have to do the number #2 in the alsa.config
my mistake :(

than I can start with my second problem... the card reader read can read the number but not in the phonie application...
christmas can come now *jui

BenutzerKeyna, 2020-12-20:
ok, I spoke to soon... @s-martin 
Its works with my HDMI monitor, only.
with Monitor, the Sounds come over USB out of the Raspberry. If I plug off the Monitor, after reboot no sound. 
reboot already done in both directions. 
the control LED on the usb soundcard is flashing with HDMI monitor; without monitor the LED is always on

s-martin, 2020-12-20:
Please post the result of `cat /usr/share/alsa/alsa.conf`.

could please post the output of `cat /proc/asound/modules` when hdmi is connected and also when it’s not connected? Maybe with reboot in between?

BenutzerKeyna, 2020-12-21:
great supposition! Running very well
the code poste isn't necessary, i found the difference:
with HDMI 
$ cat /proc/asound/modules
0 snd_bcm2835
1 snd_bcm2835
2 snd_usb_audio
without HDMI
$ cat /proc/asound/modules
0 snd_bcm2835
1 snd_usb_audio

Change as described in the troubleshooting 
-> $ sudo nano /usr/share/alsa/alsa.conf
-> $ sudo nano /etc/asound.conf

thanks a lot!

Andreas-Roth-27, 2021-01-02:
> it works now! Thanks a lot @s-martin
>
> I did the instruction link two times before. But I understood this script wrong... I thought the change to #1 change my port, not the position...
> my config was: [...]

Same for me... card reader works ... in console it displays the id, but with an error message bash: "id": command not found

Andreas-Roth-27, 2021-01-02:
> ok, I spoke to soon... @s-martin
> Its works with my HDMI monitor, only.
> with Monitor, the Sounds come over USB out of the Raspberry. If I plug off the Monitor, after reboot no sound.
> reboot already [...]

Similar. Sound works. But if I unplug HDMI monitor it is silent

s-martin, 2021-01-02:
> Similar. Sound works. But if I unplug HDMI monitor it is silent

Yes, the problem is that when you plug in a HDMI monitor (which is also a speaker) an additional output device is added and the numbering is changed.
So if you normally use the phoniebox without a monitor you need to unplug the monitor before you call `cat /proc/asound/modules` to determine the correct output number.

Andreas-Roth-27, 2021-01-03:
With the HDMI monitor plugged in I can see
0 snd_bcm2835
1 snd_bcm2835
without the HDMI monitor I get
0 snd_bcm2835 
only. So far so good. How do I use that for my setup? Which file would I have to edit knowing about this?

s-martin, 2021-01-03:
I assume you will have no HDMI display attached, when you use the Phoniebox operational.

So please do the following:

Open the file with `sudo nano /etc/asound.conf`

Add the following to the file:
```
pcm.!default {
    type hw
    card 0
}
ctl.!default {
    type hw
    card 0
}
```

Press `CTRL+O` to save and `CTRL+X` to exit the editor.

Then reboot with `sudo reboot`.

Andreas-Roth-27, 2021-01-03:
Thank you! It worked! So that would close my Issue #1220

closed 2021-01-03 by s-martin