MiczFlor/RPi-Jukebox-RFID #1205: After Update from 2.0rv7 to 2.2. GPIO / MQTT is not working

Klaus-Guenther, 2020-12-15:
**Here is what i use:**
Raspberry Pi Zero W
HifiBerry Miniamp
Neuftech RFID Reader
Some LEDs, some push buttons

So before the update, the LED blinks at startup, and i could controll with the push buttons VolUp/Dow, Play/Pause, Shutdown (via Shim) and Prev/Next song. Also i can controll the whole phoniebox via mqtt.

**So, here is what i did:**
Update from 2.0rc7 to 2.2. with the on-line script
First: To keep all songs, folders.... FAIL. 
And yes I clicked YES on the question: Do you want to take over old settings, so also on the questions after that where it was about the RFID settings etc..
I have read the instructions for the new GPIO procedure that I have to redo the settings, here I have entered my GPIO pins. So also in the MQTT settings file username and password for MQTT.

**What happend now?**
So you can start the box, the LEDs flash, then when the software is started, the LEDs go off, was not so before. You can then start music via the web interface, which is also played, volume up and down can also be done via the software, but exactly nothing happens via the buttons.

MQTT does not connect either.

So, what log do I need to get so we can figure out if this is an error between the keyboard and the chair or in the system?

s-martin, 2020-12-15:
Please provide the output of `tail -n 500 /var/log/syslog`.

What's the output of `cat ~/RPi-Jukebox-RFID/settings/gpio_settings.ini`?

Klaus-Guenther, 2020-12-15:
Vielleicht auch noch mal als Hinweis, die LED sind mit Pullup Wiederständen am Pi angeschlossen.
Vor dem Update lief auch alles anstandslos....

**`tail -n 500 /var/log/syslog.`**

Dec 15 11:52:24 phoniebox kernel: [    0.100884] audit: type=2000 audit(0.090:1)                                                                                                                                                             : state=initialized audit_enabled=0 res=1
Dec 15 11:52:24 phoniebox kernel: [    0.120832] raspberrypi-firmware soc:firmwa                                                                                                                                                             re: Attached to firmware from 2020-11-30 22:13, variant start
Dec 15 11:52:24 phoniebox kernel: [    0.130857] raspberrypi-firmware soc:firmwa                                                                                                                                                             re: Firmware hash is ab1181cc0cb6df52bfae3b1d3fef0ce7c325166c
Dec 15 11:52:24 phoniebox kernel: [    0.195741] bcm2835-dma 20007000.dma: DMA l                                                                                                                                                             egacy API manager, dmachans=0x1
Dec 15 11:52:24 phoniebox kernel: [    0.199020] SCSI subsystem initialized
Dec 15 11:52:24 phoniebox kernel: [    0.199350] usbcore: registered new interfa                                                                                                                                                             ce driver usbfs
Dec 15 11:52:24 phoniebox kernel: [    0.199508] usbcore: registered new interfa                                                                                                                                                             ce driver hub
Dec 15 11:52:24 phoniebox kernel: [    0.199815] usbcore: registered new device                                                                                                                                                              driver usb
Dec 15 11:52:24 phoniebox kernel: [    0.202741] clocksource: Switched to clocks                                                                                                                                                             ource timer
Dec 15 11:52:24 phoniebox kernel: [    1.682326] VFS: Disk quotas dquot_6.6.0
Dec 15 11:52:24 phoniebox kernel: [    1.682554] VFS: Dquot-cache hash table ent                                                                                                                                                             ries: 1024 (order 0, 4096 bytes)
Dec 15 11:52:24 phoniebox kernel: [    1.683088] FS-Cache: Loaded
Dec 15 11:52:24 phoniebox kernel: [    1.683517] CacheFiles: Loaded
Dec 15 11:52:24 phoniebox kernel: [    1.704918] thermal_sys: Registered thermal                                                                                                                                                              governor 'step_wise'
Dec 15 11:52:24 phoniebox kernel: [    1.705646] NET: Registered protocol family                                                                                                                                                              2
Dec 15 11:52:24 phoniebox kernel: [    1.707151] tcp_listen_portaddr_hash hash t                                                                                                                                                             able entries: 512 (order: 0, 4096 bytes, linear)
Dec 15 11:52:24 phoniebox kernel: [    1.707282] TCP established hash table entr                                                                                                                                                             ies: 4096 (order: 2, 16384 bytes, linear)
Dec 15 11:52:24 phoniebox kernel: [    1.707403] TCP bind hash table entries: 40                                                                                                                                                             96 (order: 2, 16384 bytes, linear)
Dec 15 11:52:24 phoniebox kernel: [    1.707530] TCP: Hash tables configured (es                                                                                                                                                             tablished 4096 bind 4096)
Dec 15 11:52:24 phoniebox kernel: [    1.707757] UDP hash table entries: 256 (or                                                                                                                                                             der: 0, 4096 bytes, linear)
Dec 15 11:52:24 phoniebox kernel: [    1.707837] UDP-Lite hash table entries: 25                                                                                                                                                             6 (order: 0, 4096 bytes, linear)
Dec 15 11:52:24 phoniebox kernel: [    1.708394] NET: Registered protocol family                                                                                                                                                              1
Dec 15 11:52:24 phoniebox kernel: [    1.709807] RPC: Registered named UNIX sock                                                                                                                                                             et  [... cut ...]
Dec 15 11:52:24 phoniebox kernel: [    2.820053] Non-periodic Split Transactions
Dec 15 11:52:24 phoniebox kernel: [    2.820053] Periodic Split Transactions
[... 506 lines cut ...]
functionCall: functionCallVolD

[NextSong]
enabled: False
Type:  Button
Pin: 21
pull_up: True
hold_time: 0.3
functionCall: functionCallPlayerNext

[PrevSong]
enabled: False
Type:  Button
Pin: 20
pull_up: True
hold_time: 0.3
functionCall: functionCallPlayerPrev

[Halt]
enabled: False
Type:  Button
Pin: 21
pull_up: True
hold_time: 0.3
functionCall: functionCallPlayerPauseForce

s-martin, 2020-12-15:
```
Dec 15 11:52:52 phoniebox daemon_mqtt_client.py[561]: Traceback (most recent call last):
Dec 15 11:52:52 phoniebox daemon_mqtt_client.py[561]:   File "/home/pi/RPi-Jukebox-RFID/scripts/daemon_mqtt_client.py", line 363, in <module>
Dec 15 11:52:52 phoniebox daemon_mqtt_client.py[561]:     client.tls_set(ca_certs=mqttCA, certfile=mqttCert, keyfile=mqttKey)
Dec 15 11:52:52 phoniebox daemon_mqtt_client.py[561]:   File "/home/pi/.local/lib/python3.7/site-packages/paho/mqtt/client.py", line 819, in tls_set
Dec 15 11:52:52 phoniebox daemon_mqtt_client.py[561]:     context.load_cert_chain(certfile, keyfile)
Dec 15 11:52:52 phoniebox daemon_mqtt_client.py[561]: FileNotFoundError: [Errno2] No such file or directory
```

Looks like a file is missing

s-martin, 2020-12-15:
Please make sure that all needed buttons are enabled and you need to use the **GPIO number!** for `Pin`.

s-martin, 2020-12-15:
Please check also https://github.com/MiczFlor/RPi-Jukebox-RFID/discussions/865?converting=1#discussioncomment-211905 for MQTT

phlowsec, 2020-12-18:
Hi, 
I had a similar issue, after upgrading. For me it helped to change in the components/gpio_control/gpio_control.py the location of the gpio_settings.init (should be line 101) to ~/RPi-Jukebox-RFID/settings/gpio_settings.ini.
Perhaps it helps.

s-martin, 2020-12-18:
> Hi,
> I had a similar issue, after upgrading. For me it helped to change in the components/gpio_control/gpio_control.py the location of the gpio_settings.init (should be line 101) to ~/RPi-Jukebox-RFID/settings/gpio_settings.ini.
> Perhaps it helps.

This should already befinde since 2.2

s-martin, 2022-05-29:
Closing, please reopen if the issue still exists.

closed 2022-05-29 by s-martin