MiczFlor/RPi-Jukebox-RFID #1206: Spotify does not work.🐛 | BUG SUMMARY

tfischer90, 2020-12-15:
## Bug

### What I did
I installed RPi-Jukebox RFID with the one line installer. I use a fresh Raspbian Buster install on an Raspberry pi Zero W.
My installation only worked with the process discribed by FloMrFrei in Issue #718 .

### What happened

I got the interface running. When I press on Spotify a new window opens but I dont see any music. I can click the buttons on the left but nothing happens.
In the installation process I inserted my Spoitfy User name, pwd and the other required IDs.

s-martin, 2020-12-15:
Please check, if
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#spotify-wont-play-but-after-scan-of-library-it-works

helps you.

s-martin, 2020-12-18:
Could you please try
`sudo pip3 install --upgrade --force-reinstall Mopidy-Iris` and reboot?

tfischer90, 2020-12-18:
Hi S-Martin,
Thanks for the help so far! I reinstalled Mopid-Iris which resulted in the following pop-up message when I go to the web interface, click on Spotify, than on the IPadress::6680/iris/settings i get "SyntaxError: Unexpected end of JSON input" and "Could not load your profile Unexpected end of JSON input"

The journalctl results the following messages:

> -- Logs begin at Fri 2020-12-18 17:27:56 GMT, end at Fri 2020-12-18 17:30:38 GMT. --
Dec 18 17:28:33 raspberrypi systemd[1]: Starting Mopidy music server...
Dec 18 17:28:35 raspberrypi systemd[1]: Started Mopidy music server.
Dec 18 17:29:40 raspberrypi mopidy[901]: INFO     [MainThread] mopidy.__main__ Starting Mopidy 3.0.2
Dec 18 17:29:40 raspberrypi mopidy[901]: INFO     [MainThread] mopidy.config Loading config from builtin defaults
Dec 18 17:29:40 raspberrypi mopidy[901]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy-local.conf
Dec 18 17:29:40 raspberrypi mopidy[901]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy.conf
Dec 18 17:29:40 raspberrypi mopidy[901]: INFO     [MainThread] mopidy.config Loading config from file:///etc/mopidy/mopidy.conf
Dec 18 17:29:40 raspberrypi mopidy[901]: INFO     [MainThread] mopidy.config Loading config from command line options
Dec 18 17:29:40 raspberrypi mopidy[901]: ERROR    [MainThread] mopidy.internal.log Loading logging config '/etc/mopidy/logging.conf' failed. 'formatters'
Dec 18 17:29:50 raspberrypi mopidy[901]: INFO     [MainThread] mopidy.__main__ Enabled extensions: spotify, mpd, http, iris, local, m3u, softwaremixer, file, stream
Dec 18 17:29:50 raspberrypi mopidy[901]: INFO     [MainThread] mopidy.__main__ Disabled extensions: none
Dec 18 17:29:55 raspberrypi mopidy[901]: INFO     [MainThread] mopidy.commands Starting Mopidy mixer: SoftwareMixer
Dec 18 17:29:55 raspberrypi mopidy[901]: INFO     [MainThread] mopidy.commands Mixer volume set to 30
Dec 18 17:29:55 raspberrypi mopidy[901]: INFO     [MainThread] mopidy.commands Starting Mopidy audio
Dec 18 17:29:56 raspberrypi mopidy[901]: INFO     [MainThread] mopidy.commands Starting Mopidy backends: FileBackend, M3UBackend, StreamBackend, SpotifyBackend, LocalBa
Dec 18 17:29:56 raspberrypi mopidy[901]: INFO     [Audio-2] mopidy.audio.actor Audio output set to "alsasink"
Dec 18 17:30:00 raspberrypi mopidy[901]: ERROR    [SpotifyBackend-6] mopidy_spotify.web Authorization failed, not attempting Spotify API request. Please get new credent
Dec 18 17:30:00 raspberrypi mopidy[901]: ERROR    [SpotifyBackend-6] mopidy_spotify.web OAuth token refresh failed: invalid_client Client not known.
Dec 18 17:30:00 raspberrypi mopidy[901]: ERROR    [SpotifyBackend-6] mopidy_spotify.web Failed to load Spotify user profile
Dec 18 17:30:00 raspberrypi mopidy[901]: INFO     [MainThread] mopidy.commands Starting Mopidy core
Dec 18 17:30:01 raspberrypi mopidy[901]: INFO     [SpotifyEventLoop] mopidy_spotify.backend Logged in to Spotify in online mode
Dec 18 17:30:01 raspberrypi mopidy[901]: INFO     [MainThread] mopidy.commands Starting Mopidy frontends: HttpFrontend, MpdFrontend, IrisFrontend
Dec 18 17:30:01 raspberrypi mopidy[901]: INFO     [HttpFrontend-12] mopidy.http.actor HTTP server running at [::ffff:0.0.0.0]:6680
Dec 18 17:30:01 raspberrypi mopidy[901]: INFO     [MainThread] mopidy_mpd.actor MPD server running at [::ffff:0.0.0.0]:6600
Dec 18 17:30:01 raspberrypi mopidy[901]: INFO     [IrisFrontend-14] mopidy_iris.core Starting Iris 3.55.4
Dec 18 17:30:01 raspberrypi mopidy[901]: INFO     [MainThread] mopidy.commands Starting GLib mainloop
Dec 18 17:30:01 raspberrypi mopidy[901]: INFO     [MpdSession-15] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:34860
Dec 18 17:30:02 raspberrypi mopidy[901]: INFO     [MpdSession-16] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:34864
Dec 18 17:30:03 raspberrypi mopidy[901]: INFO     [MpdSession-17] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:34868
Dec 18 17:30:03 raspberrypi mopidy[901]: INFO     [MpdSession-18] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:34872
Dec 18 17:30:03 raspberrypi mopidy[901]: INFO     [MpdSession-19] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:34876
Dec 18 17:30:08 raspberrypi mopidy[901]: INFO     [MpdSession-20] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:34884
Dec 18 17:30:08 raspberrypi mopidy[901]: INFO     [MpdSession-21] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:34888
Dec 18 17:30:13 raspberrypi mopidy[901]: INFO     [MpdSession-22] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:34892
Dec 18 17:30:18 raspberrypi mopidy[901]: INFO     [MpdSession-23] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:34896
Dec 18 17:30:23 raspberrypi mopidy[901]: INFO     [MpdSession-24] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:34900
Dec 18 17:30:28 raspberrypi mopidy[901]: INFO     [MpdSession-25] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:34904
Dec 18 17:30:33 raspberrypi mopidy[901]: INFO     [MpdSession-26] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:34908
Dec 18 17:30:38 raspberrypi mopidy[901]: INFO     [MpdSession-27] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:34912

s-martin, 2020-12-18:
@tfischer90 looks like your login is not valid (anymore). Maybe you need to generate new client id/secret.

s-martin, 2020-12-18:
Please edit 
`/etc/mopidy/mopidy.conf`
and
`~/.config/mopidy/mopidy.conf`

you first have to get client-id and client-secret here: https://www.mopidy.com/authenticate/  This must be done manually.

 Put your username, password, client_id, client_secret into the spotify section.

closed 2022-12-20 by s-martin