MiczFlor/RPi-Jukebox-RFID #1210: Touch Sensor/Buttons

UndIch, 2020-12-16:
Hi everyone - hardly one year after finishing my first Phonie - a new one needs to be build and despite the old one was running ok‘ish - i planned some improvements- that send me back 2 days already. 
because the little one used to destroy most of the push buttons I ordered (Conrad, Reichelt, Berrybase) - I choose TTP223 Touch „buttons“

but I can’t get them to work. In my old Setup (old Version 1.9) I tested them and all was fine. Now in the new Setup (version 2) no matter what i do I can’t get them to work. The GND is connected to GND, VCC to the 3V Outlet and I/O to the PIN 26,20,19 and 16. touching them and the red led ad the touch chip flashes up - but even wiring pi states „gpio read“ with „0“ - and of course nothing happens 

Any ideas?

s-martin, 2020-12-16:
Did you configure the buttons in the new GPIO control?
see 
https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/develop/components/gpio_control
an here for touch buttons
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Capacitive-touch-sensors

UndIch, 2020-12-16:
yes.  Sure did. And at least gpio read should change if touched - but nothing.... puh kind of desperate again. It always sounds so easy to "reinstall" it and .... spending several days :D

UndIch, 2020-12-17:
So - Short Update: I can’t really say how - but at least the next/previous button work. But i can not get the volume buttons to work - if I switch the physical buttons - they all seem to work with next/previous and volume up (kind of strange thing happening if you push „previous“ at the beginning of a list - seems to knock out the gpio completely -
only reboot helps...) but i still can not figure out what’s wrong with the volume down thing. Using a voltmeter it shoes 2,7 V firing up as soon as i touch the button and this single wire is connect to e.g. PIN 16 - and it is set as Pin for volume down. Not working.

that-coding-human, 2020-12-27:
Did you install any other software? E.g. the oled Software, mentioned in the "splittscheid 3.0" Blog?

I had loads of trouble as these scripts are not compatible with the 2.x gpio handling of the phoniebox software. 
If you have any other software running, it might be some of these side effects.
Can you paste output of:

- sudo ps -ef | grep -i gpio

Just a shot in the dark, but we have to start somewhere.

that-coding-human, 2020-12-27:
Uh, and please some more information about the connected hardware (on off shim? Which sound device?) and your gpio configuration.

UndIch, 2020-12-27:
Sorry for the delay- Christmas stuff ... - with some night shifts I restarted from the beginning - and think I found the mistake: I used one 3V GPIO for all of the Touch Buttons - after reading the specs (2-5V) I thought about it and that one 3,3 V GPIO is to less for 4 times 2 V ... so I choose a different 3,3 V for V+/- and it starts to work ...

closed 2021-03-18 by UndIch