MiczFlor/RPi-Jukebox-RFID #1215: 🐛 | Registering Cards does not like Folders with single quotes

rimeraz, 2020-12-19:
## Bug

I tried to register a card to the folder `Kai Lüftner und Klabauterband - Rotz'n'Roll Radio` in the web interface.

### What happened

```
ERROR:
Seems you haven't selected anything! Add an URL and stream type, select a folder or a system command. Or 'Cancel' to go back to the home page.

(error 002)
```

### I expected this to happen

Properly register the card to the folder.

### Further information that might help

Nothing in the logs.

## Software

### Base image and version

`2020-12-02-raspios-buster-armhf-lite`

### Branch / Release

master

### Installscript

`scripts/installscripts/buster-install-default.sh`

## Hardware

### RaspberryPi version

Raspberry Pi 3 Model B Plus Rev 1.3

s-martin, 2022-12-20:
Can you check, if the issue still persists with current 2.4 version?

s-martin, 2023-12-05:
Closing as no interaction for a couple of months.
Feel free to reopen the issue, if the problem still occurs.

closed 2023-12-05 by s-martin