MiczFlor/RPi-Jukebox-RFID #1217: WM8960 Hi-Fi Sound Card HAT for Raspberry Pi - volume controls not working

mistersunshine2020, 2020-12-19:
Hi there, 

This is my second PhonieBox build - my 5 year old daughter loved the family player so much she wants one for her bedroom. I've managed to assemble everything together and it works just fine, but I'm using the WM8960 HAT and speakers so the whole unit is self contained. 

I can get sound output, and I've mapped GPIO buttons for forward/back/play, and was intending to add both a headphone socket and a rotary button so volume can be controlled. But I'm not able to change the output volume on the web browser interface. It's at 0% and any attempts to change the volume don't work - but if I log in to the Raspberry Pi GUI I can change the volume, and alsamixer also works. 

The instructions for this sound card feel a little out of date (I'm using the latest build of PhonieBox and the most recent version of Buster). Am I missing something? Various configurations below: 

**- GPIO buttons installed using /home/pi/RPi-Jukebox-RFID/settings/gpio_settings.ini (uses pins 7, 9, 14, 16, 32, 34)

- Soundcard HAT installed following manual installation sequence (replicated in the wiki here)

- 16x2 display installed using pins 2, 6, 3, 5

- I followed the steps [here](https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Sound_Card_WM8960_Hi-Fi_HAT#test-button) about GPIO 17 but not quite sure whether it's necessary in v 2?**

[note pins above are not GPIO numbers but pins]

**- aplay -l gives:** 

> **** List of PLAYBACK Hardware Devices ****
> card 0: b1 [bcm2835 HDMI 1], device 0: bcm2835 HDMI 1 [bcm2835 HDMI 1]
>   Subdevices: 4/4
>   Subdevice #0: subdevice #0
>   Subdevice #1: subdevice #1
>   Subdevice #2: subdevice #2
>   Subdevice #3: subdevice #3
> card 1: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
>   Subdevices: 4/4
>   Subdevice #0: subdevice #0
>   Subdevice #1: subdevice #1
>   Subdevice #2: subdevice #2
>   Subdevice #3: subdevice #3
> card 2: wm8960soundcard [wm8960-soundcard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 [bcm2835-i2s-wm8960-hifi wm8960-hifi-0]
>   Subdevices: 1/1
>   Subdevice #0: subdevice #0

**- cat /proc/asound/modules gives:** 

>  0 snd_bcm2835
>  1 snd_bcm2835
>  2 snd_soc_wm8960_soundcard

**- alsa.conf changed to:** 

> defaults.ctl.card 2
> defaults.pcm.card 2

**- mixer scontrols gives:**

> Simple mixer control 'Master',0
> Simple mixer control 'Capture',0

**- etc/mpd.conf changed to: **

> audio_output {
>   type "alsa"
>   name "Alsa Device"
>   mixer_type "Software"
>   mixer_device "bcm2835-i2s-wm8960-hifi"
>   mixer_control "playback"
> }
> 

**/home/pi/RPi-Jukebox-RFID/settings/Audio_iFace_Name changed to:

> playback

Under the web interface, volume control is set to mpd - if I change it to amixer and try to change the volume, I get: 

>An error occorured: Execution failed Command: /home/pi/RPi-Jukebox-RFID/scripts/playout_controls.sh -c=volumeup Output: RC: .1

Any help with this much appreciated! Love the work that's gone on here, and the player has made a huge difference to us, so really keen to try and get this fixed before Christmas for my daughter if I can...

Best,

Tom

s-martin, 2020-12-20:
Are you sure using pins 3, 5 both for the soundcard and the display work, I.e. have you tried disconnecting the display?

s-martin, 2020-12-20:
Please use `Master` instead of playback in mpd.conf and iface.

mistersunshine2020, 2020-12-20:
> Are you sure using pins 3, 5 both for the soundcard and the display work, I.e. have you tried disconnecting the display?

> Are you sure using pins 3, 5 both for the soundcard and the display work, I.e. have you tried disconnecting the display?

Hi, 

Thanks for the reply...

Yes, I'm sure they work together - the soundcard was configured first and this issue was there from the start - I added the display and it didn't seem to have any impact on the functionality, but I did wonder about two i2c devices on a single installation. I did a lot of searching which seemed to suggest that they could work in harmony, which they seem to be doing so far (and which is good news for future builds). 

I avoided using any of the other GPIO pins that were used by the soundcard.

mistersunshine2020, 2020-12-20:
> Please use `Master` instead of playback in mpd.conf and iface.

I will give this a go. Thanks so much for the help and advice. It's appreciated.

mistersunshine2020, 2020-12-22:
Hi, 

Reassembled everything (I was spray painting the enclosure gold and sticking unicorn stickers on - all very important stuff...) and made the changes. I amended mpd.conf, iface and global.conf to use 'Master'. I still had no volume control. I disconnected the display, and that made no difference too. 

I saw a response to another person suggesting amending asound.conf, which I attempted, adding: 

> pcm.!default {
>     type hw
>     card 2
> }
> ctl.!default {
>     type hw
>     card 2
> }

This disabled the sound entirely, until it was removed, and the sound worked again. The current contents of asound.conf are: 

>   GNU nano 3.2                              /etc/asound.conf                               Modified  
> 
> # The IPC key of dmix or dsnoop plugin must be unique
> # If 555555 or 666666 is used by other processes, use another one
> 
> 
> # use samplerate to resample as speexdsp resample is bad
> defaults.pcm.rate_converter "samplerate"
> 
> pcm.!default {
>     type asym
>     playback.pcm "playback"
>     capture.pcm "capture"
> }
> 
> pcm.playback {
>     type plug
>     slave.pcm "dmixed"
> }
> 
> pcm.capture {
>     type plug
>     slave.pcm "array"
> }
> 
> pcm.dmixed {
>     type dmix
>     slave.pcm "hw:wm8960soundcard"
>     ipc_key 555555
> }
> 
> pcm.array {
>     type dsnoop
>     slave {
>         pcm "hw:wm8960soundcard"
>         channels 2
>     }
>     ipc_key 666666
> }

Any ideas what might be the issue? I'm trying all these things with the 16x2 display disconnected from the GPIO pins. 

Massive thanks in advance for your help. 

Best, Tom

mistersunshine2020, 2020-12-22:
PS - I attempted editing asound.conf to:

defaults.pcm.rate_converter "samplerate"

> pcm.!default {
>     type asym
>     playback.pcm "Master"
>     capture.pcm "capture"
> }

But this disabled the sound entirely.

miohna, 2020-12-27:
Hi, Tom,

your `amixer scontrols` output does appear to show the scontrols of the wrong sound card. Can you run `amixer -c 2 scontrols` (the WM8960 Hi-Fi HAT is card number 2, in your case)? You should end up with a long list of 51 simple mixer controls, including "Playback". If you get this long list, your ALSA setting should be fine.

Your mpd config seems to have the same issue of using the default sound card. Give it a try to add  `device          "hw:CARD=wm8960soundcard,DEV=0"` to the audio output config. In total you may like to change to
```
audio_output {
        type            "alsa"
        name            "WM8960 HiFi HAT"
        device          "hw:CARD=wm8960soundcard,DEV=0"
        mixer_type      "hardware"
        mixer_device    "hw:CARD=wm8960soundcard"
        mixer_control   "Playback"
}
```

There are some more issues with the WM8960 Hi-Fi HAT, but let us start with the two above first :-)

michnixweiss, 2021-01-23:
Is this solved, because I have / had some problems with this HAT. Some solved, some worked around some not.
Also seen here [#549](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/549)
Sounds like @miohna figured out completely!?

jorgemia, 2021-02-26:
@mistersunshine2020 were you able to get the sound control working? I can adjust it via the web interface/rotary knob if I set the volume control to MPD instead of amixer. Amixer won't work. Is there any benefit of amixer over mpd?

ThomasP81, 2021-11-09:
Hello, 
i dont know, it it matters, but i found the slutuin here (mpd.conf)
https://forum-raspberrypi.de/forum/thread/47771-phoniebox-mit-wm8960-hi-fi-hat-onoff-shim-rc522-rotary-und-buttons/?postID=435020#post435020

`audio_output {
        type            "alsa"
        name            "My ALSA Device"
#       device          "hw:0,0"        # optional        
        mixer_type      "software"`

hdow1, 2023-04-30:
Just dumping this here incase anyone else has the same issue I had. 

This error: An error occorured: Execution failed Command: /home/pi/RPi-Jukebox-RFID/scripts/playout_controls.sh -c=volumeup Output: RC: .1

Was fixed by doing `sudo systemctl enable mpd.service`

Source: https://forum-raspberrypi.de/forum/thread/56032-phoniebox-kein-sound-nur-durch-mpd-restart/

mittler-works, 2023-07-09:
Maybe a little bit late, but I am also building a Phoniebox with this soundcard and had several issues...

You wrote:

> **/home/pi/RPi-Jukebox-RFID/settings/Audio_iFace_Name changed to:
> 
>    playback

The interface name on wm8960 is `Playback` (capital P). That should be it.