MiczFlor/RPi-Jukebox-RFID #1218: 🐛 | Status LED is not switched on

retrofreak83, 2020-12-20:
## Bug

### What I did

I installed the raspberry pi with above mentioned buster image and ran the installer script
Later-on, I re-ran the installation to have GPIO service enabled.

### What happened

There is no voltage between a GND pin and pin 32.

### I expected this to happen

Pin 32, according the https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Using-GPIO-hardware-status-led is said to be lit as soon as the Pi is booted an supported out of the box.

However, this is not the case: between pins 32 and 30, there is 0 V.

### Further information that might help

I am able to get 3,3V between pin 32 and GND when I activate the pin manually, like so:
```
import RPi.GPIO as GPIO
GPIO.setmode(GPIO.BOARD)
GPIO.setup(32, GPIO.OUT)
GPIO.output(32, 1)     
```

## Software

### Base image and version

2020-12-02-raspios-buster-armhf-lite

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

### Branch / Release

master

### Installscript

`scripts/installscripts/buster-install-default.sh`

## Hardware

### RaspberryPi version

`3 B+`

### RFID Reader

n.c. at the moment
### Soundcard

n.c. at the moment

### Other notable hardware

Note: the Github issue template says:
```
i.e. post your GPIO pin settings from `RPi-Jukebox-RFID/scripts/gpio-buttons.py`:
```
This file is missing on my Pi, as it is also in the current git master.

s-martin, 2020-12-20:
Since 2.1 we use [GPIO control](https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/develop/components/gpio_control), unfortunately the wiki page for status LED is not updated yet.

please check the description and see here https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/components/gpio_control/example_configs/gpio_settings_rotary_and_led.ini for examples.

retrofreak83, 2020-12-21:
Using GPIO control, it works as expected. Thanks!

closed 2020-12-21 by retrofreak83