MiczFlor/RPi-Jukebox-RFID #1220: No audio with phoniebox software v. 2.2 (sound not audible)

Andreas-Roth-27, 2020-12-20:
## Bug

I installed the raspberry pi 3 model B with above mentioned buster image and ran the current one line installer script (i.e. for phoniebox 2.2). My hardware has no external sound card just the audio jack with connected loud speakers.

I am able to run phoniebox, download mp3 and have access to Spotify (premium account). The sound files are visible and "played" (e.g. seconds are counted up).

I am also able to output sound like from a Youtube video or a test file *.WAV apart from Phoniebox.
 
I would have expected that I would hear my sound files through control of the phoniebox software as well.

cat /home/pi/RPi-Jukebox-RFID/settings/version shows this   "2.2 - 305325d - master"
amixer scontrols this "Simple mixer control 'Master',0" and "Simple mixer control 'Capture',0"
sudo cat /sys/firmware/devicetree/base/model shows this "Raspberry Pi 3 Model B Rev 1.2"

s-martin, 2020-12-20:
Did you check https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#audio-is-not-working

especially a)

Andreas-Roth-27, 2020-12-20:
Yes - I checked it. Portion a) is about „audio never worked“ which is true for my phoniebox but not for the raspberry. The raspberry is audible. And a) is about problems with an external sound card which I don’t have.

I haven’t found the file settings/Audio_iFace_Name which is because I am not very good at Linux. My programming skills stem from Fortran V and microVAX.  

Command cat /proc/asound/modules renders 0 snd_bcm2835 and 1 snd_bcm2835

Placing command sudo nano /usr/share/alsa/alsa.conf
opens a rather long file and changing parameters in there like described is like poking in the fog. I don’t want to change things that I haven’t truely understood.

Could you help here? I mean explain the parameters described and also where to find them in alsa.conf?

Best regards,
Andreas

s-martin, 2020-12-20:
> I haven’t found the file settings/Audio_iFace_Name which is because I am not very good at Linux.

Edit with `nano /home/pi/RPi-Jukebox-RFID/settings/Audio_iFace_Name`
Save with `CTRL+O`
Exit with `CTRL+X`

> Command cat /proc/asound/modules renders 0 snd_bcm2835 and 1 snd_bcm2835 

What’s the output of `aplay -l`?

> Placing command sudo nano /usr/share/alsa/alsa.conf opens a rather long file and changing parameters in there like described is like poking in the fog. 

Create the file with `sudo nano /etc/asound.conf`
Add the following content. Card should be the number from aplay -l above 

```
pcm.!default {
    type hw
    card 1
}
ctl.!default {
    type hw
    card 1
}
```

you probably need a reboot.

some more changes are probably needed (see https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#raspberry-pi-update-may-2020-broke-my-audio), but let’s get your answers first.

Andreas-Roth-27, 2020-12-20:
> What’s the output of aplay -l?

**** List of PLAYBACK Hardware Devices ****
card 0: b1 [bcm2835 HDMI 1], device 0: bcm2835 HDMI 1 [bcm2835 HDMI 1]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
card 1: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3

Andreas-Roth-27, 2020-12-20:
/home/pi/RPi-Jukebox-RFID/settings/Audio_iFace_Name
contains just 1 line with „Headphone“ in it (without „“ of course)

(I found the file while I got remembered to how to change directory levels within Unix (very similar to microVAX’s VMS))

Andreas

s-martin, 2020-12-20:
So please change the audio iface name to `Headphones` in   

` /etc/mpd.conf` (for **mixer_control**), `./settings/global.conf` (Audio iface) and `./settings/Audio_iFace_Name`

As stated in my comment above in  asound.conf card should both be 1.

Andreas-Roth-27, 2020-12-21:
Hi,

I changed all what you proposed, i.e.

1) set up file asound.conf with the card value of „1“
2) edited mpd.conf — „Headphones" instead of „Headphone"
3) edited global.conf and Audio-iFace_Name both — „Headphones" instead of „Headphone“

re-booted but still no sound.

I understand that this missing „s“ clearly is an error. 

And by the way, learned or remembered a bit about Linux….

So, what’s next to do? I understand that there is still something. So I highly appreciate your help.

Andreas

s-martin, 2020-12-21:
What’s the output of `tail-n 500 /var/log/syslog`?

Andreas-Roth-27, 2020-12-21:
many many lines 

and in between

Dec 21 11:16:35 raspberrypi lighttpd[894]: amixer: Unable to find simple control 'Headphones‘,0

and in log form

pi@raspberrypi:~ $ tail -n 500 /var/log/syslog

Dec 21 11:15:09 raspberrypi mopidy[518]: INFO     [MpdSession-12603] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42684

Dec 21 11:15:09 raspberrypi mopidy[518]: INFO     [MpdSession-12604] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42688

Dec 21 11:15:09 raspberrypi mopidy[518]: INFO     [MpdSession-12605] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42692

Dec 21 11:15:09 raspberrypi mopidy[518]: INFO     [MpdSession-12606] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42696

Dec 21 11:15:09 raspberrypi mopidy[518]: INFO     [MpdSession-12607] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42700

Dec 21 11:15:09 raspberrypi mopidy[518]: INFO     [MpdSession-12608] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42704

Dec 21 11:15:10 raspberrypi lighttpd[894]: amixer: Unable to find simple control 'Headphones',0

Dec 21 11:15:10 raspberrypi lighttpd[894]: amixer: Unable to find simple control 'Headphones',0

Dec 21 11:15:10 raspberrypi mopidy[518]: INFO     [MpdSession-12609] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42708

Dec 21 11:15:10 raspberrypi mopidy[518]: INFO     [MpdSession-12610] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42712

Dec 21 11:15:10 raspberrypi mopidy[518]: INFO     [MpdSession-12611] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42716

Dec 21 11:15:10 raspberrypi mopidy[518]: INFO     [MpdSession-12612] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42720

Dec 21 11:15:10 raspberrypi mopidy[518]: INFO     [MpdSession-12613] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42724

Dec 21 11:15:10 raspberrypi mopidy[518]: INFO     [MpdSession-12614] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42728

Dec 21 11:15:11 raspberrypi mopidy[518]: INFO     [MpdSession-12615] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42732

Dec 21 11:15:11 raspberrypi mopidy[518]: INFO     [MpdSession-12616] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42736

Dec 21 11:15:11 raspberrypi mopidy[518]: INFO     [MpdSession-12617] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42740

Dec 21 11:15:12 raspberrypi mopidy[518]: INFO     [MpdSession-12618] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42744

Dec 21 11:15:12 raspberrypi lighttpd[894]: amixer: Unable to find simple control 'Headphones',0

Dec 21 11:15:12 raspberrypi mopidy[518]: INFO     [MpdSession-12619] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42748

Dec 21 11:15:12 raspberrypi mopidy[518]: INFO     [MpdSession-12620] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42752

Dec 21 11:15:12 raspberrypi mopidy[518]: INFO     [MpdSession-12621] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42756

Dec 21 11:15:15 raspberrypi mopidy[518]: INFO     [MpdSession-12622] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42760

Dec 21 11:15:15 raspberrypi mopidy[518]: INFO     [MpdSession-12623] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42764

Dec 21 11:15:15 raspberrypi mopidy[518]: INFO     [MpdSession-12624] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42768

Dec 21 11:15:15 raspberrypi mopidy[518]: INFO     [MpdSession-12625] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42772

Dec 21 11:15:15 raspberrypi mopidy[518]: INFO     [MpdSession-12626] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42776

Dec 21 11:15:15 raspberrypi mopidy[518]: INFO     [MpdSession-12627] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42780

Dec 21 11:15:15 raspberrypi mopidy[518]: INFO     [MpdSession-12628] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42784

Dec 21 11:15:15 raspberrypi mopidy[518]: INFO     [MpdSession-12629] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:42788

[... 910 lines cut ...]
Dec 21 11:16:57 raspberrypi mopidy[518]: INFO     [MpdSession-13014] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:44370

Dec 21 11:16:57 raspberrypi mopidy[518]: INFO     [MpdSession-13015] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:44374

Dec 21 11:16:57 raspberrypi mopidy[518]: INFO     [MpdSession-13016] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:44378

Dec 21 11:16:58 raspberrypi mopidy[518]: INFO     [MpdSession-13017] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:44382

Dec 21 11:16:58 raspberrypi lighttpd[894]: amixer: Unable to find simple control 'Headphones',0

Dec 21 11:16:58 raspberrypi mopidy[518]: INFO     [MpdSession-13018] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:44386

Dec 21 11:16:58 raspberrypi mopidy[518]: INFO     [MpdSession-13019] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:44390

Dec 21 11:16:58 raspberrypi mopidy[518]: INFO     [MpdSession-13020] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:44394

Dec 21 11:17:01 raspberrypi CRON[21483]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)

Dec 21 11:17:02 raspberrypi mopidy[518]: INFO     [MpdSession-13021] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:44398

Dec 21 11:17:02 raspberrypi mopidy[518]: INFO     [MpdSession-13022] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:44402

Dec 21 11:17:02 raspberrypi mopidy[518]: INFO     [MpdSession-13023] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:44406

Dec 21 11:17:02 raspberrypi mopidy[518]: I

s-martin, 2020-12-21:
Please switch back to Headphone

Andreas-Roth-27, 2020-12-21:
I did it

Put it back to Headphone
here mpd.conf, here global.conf and here Audio-iFace_Name

> Am 21.12.2020 um 10:35 schrieb Andreas Roth <andreas.roth@hamburg.de>:
> 
> 2) edited mpd.conf — „Headphones" instead of „Headphone"
> 3) edited global.conf and Audio-iFace_Name both — „Headphones" instead of „Headphone“

Andreas-Roth-27, 2020-12-21:
Now after a re-boot and the tail command the output is

pi@raspberrypi:~ $ tail -n 500 /var/log/syslog

Dec 21 11:30:51 raspberrypi kernel: [    2.647389] mmcblk0: mmc0:e624 SN64G 59.5 GiB

Dec 21 11:30:51 raspberrypi kernel: [    2.650310]  mmcblk0: p1 p2

Dec 21 11:30:51 raspberrypi kernel: [    2.656805] Indeed it is in host mode hprt0 = 00021501

Dec 21 11:30:51 raspberrypi kernel: [    2.692767] EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Opts: (null)

Dec 21 11:30:51 raspberrypi kernel: [    2.692866] VFS: Mounted root (ext4 filesystem) readonly on device 179:2.

Dec 21 11:30:51 raspberrypi kernel: [    2.693688] devtmpfs: mounted

Dec 21 11:30:51 raspberrypi kernel: [    2.718682] Freeing unused kernel memory: 1024K

Dec 21 11:30:51 raspberrypi kernel: [    2.731079] mmc1: new high speed SDIO card at address 0001

Dec 21 11:30:51 raspberrypi kernel: [    2.757060] Run /sbin/init as init process

Dec 21 11:30:51 raspberrypi kernel: [    2.866671] usb 1-1: new high-speed USB device number 2 using dwc_otg

Dec 21 11:30:51 raspberrypi kernel: [    2.866848] Indeed it is in host mode hprt0 = 00001101

Dec 21 11:30:51 raspberrypi kernel: [    3.107060] usb 1-1: New USB device found, idVendor=0424, idProduct=9514, bcdDevice= 2.00

Dec 21 11:30:51 raspberrypi kernel: [    3.107079] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0

Dec 21 11:30:51 raspberrypi kernel: [    3.107874] hub 1-1:1.0: USB hub found

Dec 21 11:30:51 raspberrypi kernel: [    3.108003] hub 1-1:1.0: 5 ports detected

Dec 21 11:30:51 raspberrypi kernel: [    3.426669] usb 1-1.1: new high-speed USB device number 3 using dwc_otg

Dec 21 11:30:51 raspberrypi kernel: [    3.522429] NET: Registered protocol family 10

Dec 21 11:30:51 raspberrypi kernel: [    3.523858] Segment Routing with IPv6

Dec 21 11:30:51 raspberrypi kernel: [    3.557016] usb 1-1.1: New USB device found, idVendor=0424, idProduct=ec00, bcdDevice= 2.00

Dec 21 11:30:51 raspberrypi kernel: [    3.557034] usb 1-1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0

Dec 21 11:30:51 raspberrypi kernel: [    3.559805] smsc95xx v1.0.6

Dec 21 11:30:51 raspberrypi kernel: [    3.651853] smsc95xx 1-1.1:1.0 eth0: register 'smsc95xx' at usb-3f980000.usb-1.1, smsc95xx USB 2.0 Ethernet, b8:27:eb:0b:31:66

Dec 21 11:30:51 raspberrypi kernel: [    3.946727] usb 1-1.2: new low-speed USB device number 4 using dwc_otg

Dec 21 11:30:51 raspberrypi kernel: [    4.090086] usb 1-1.2: New USB device found, idVendor=16c0, idProduct=27db, bcdDevice= 0.01

Dec 21 11:30:51 raspberrypi kernel: [    4.090106] usb 1-1.2: New USB device strings: Mfr=1, Product=0, SerialNumber=0

Dec 21 11:30:51 raspberrypi kernel: [    4.090120] usb 1-1.2: Manufacturer: HXGCoLtd

Dec 21 11:30:51 raspberrypi kernel: [    4.102363] input: HXGCoLtd Keyboard as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1-1.2:1.0/0003:16C0:27DB.0001/input/input0

Dec 21 11:30:51 raspberrypi kernel: [    4.177348] hid-generic 0003:16C0:27DB.0001: input,hiddev96,hidraw0: USB HID v1.10 Keyboard [HXGCoLtd] on usb-3f980000.usb-1.2/input0

Dec 21 11:30:51 raspberrypi kernel: [    4.286683] usb 1-1.3: new low-speed USB device number 5 using dwc_otg

Dec 21 11:30:51 raspberrypi kernel: [    4.451254] usb 1-1.3: New USB device found, idVendor=046a, idProduct=0023, bcdDevice= 2.20

Dec 21 11:30:51 raspberrypi kernel: [    4.451274] usb 1-1.3: New USB device strings: Mfr=0, Product=0, SerialNumber=0

Dec 21 11:30:51 raspberrypi kernel: [    4.596674] usb 1-1.4: new low-speed USB device number 6 using dwc_otg

Dec 21 11:30:51 raspberrypi kernel: [    4.740943] random: systemd: uninitialized urandom read (16 bytes read)

[... 913 lines cut ...]

Dec 21 11:31:13 raspberrypi bash[400]: 0 is the mpg123 startup volume

Dec 21 11:31:13 raspberrypi bash[400]: High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3

Dec 21 11:31:13 raspberrypi bash[400]: #011version 1.25.10; written and copyright by Michael Hipp and others

Dec 21 11:31:13 raspberrypi bash[400]: #011free software (LGPL) without any warranty but with best wishes

Dec 21 11:31:13 raspberrypi bash[400]: Directory: /home/pi/RPi-Jukebox-RFID/shared/

Dec 21 11:31:13 raspberrypi bash[400]: Playing MPEG stream 1 of 1: startupsound.mp3 ...

Dec 21 11:31:13 raspberrypi bash[400]: MPEG 1.0 L III cbr64 44100 j-s

Dec 21 11:31:13 raspberrypi bash[400]: Artist:  Phoniebox

Dec 21 11:31:16 raspberrypi mopidy[553]: INFO     [MpdSession-19] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:47546

Dec 21 11:31:16 raspberrypi mopidy[553]: INFO     [MpdSession-20] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:47550

> Am 21.12.2020 um 10:35 schrieb Andreas Roth <andreas.roth@hamburg.de>:
> 
> 2) edited mpd.conf — „Headphones" instead of „Headphone"
> 3) edited global.conf and Audio-iFace_Name both — „Headphones" instead of „Headphone“

Andreas-Roth-27, 2020-12-21:
I did that. Checked the output of the tail -n command. And found a line with „undervoltage“ in it. I changed the power supply and now there is sound.
But…
…another problem showed up: when I swipe an RFID card (although there is a beep) the field in the web interface is not filled with the ID. Can you help here, as well, please?

Andreas

s-martin, 2020-12-21:
Please post
`journalctl -u mopidy` and 
`journalctl -u lighttpd`

Andreas-Roth-27, 2020-12-21:
looks like

Dec 21 12:39:33 raspberrypi mopidy[508]: INFO     [MpdSession-15650] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:50462

and

-- Logs begin at Mon 2020-12-21 12:39:21 CET, end at Mon 2020-12-21 14:28:15 CET. --
Dec 21 12:39:21 raspberrypi sudo[17858]: www-data : TTY=unknown ; PWD=/home/pi/RPi-Jukebox-RFID/htdocs/api ; USER=root ; COMMAND=/usr/bin/echo status
                                         currentsong
                                         close
Dec 21 12:39:21 raspberrypi sudo[17858]: pam_unix(sudo:session): session opened for user root by (uid=0)
Dec 21 12:39:21 raspberrypi sudo[17858]: pam_unix(sudo:session): session closed for user root
Dec 21 12:39:21 raspberrypi sudo[17880]: www-data : TTY=unknown ; PWD=/home/pi/RPi-Jukebox-RFID/htdocs/api ; USER=root ; COMMAND=/home/pi/RPi-Jukebox-RFID/scripts/playout_
Dec 21 12:39:21 raspberrypi sudo[17880]: pam_unix(sudo:session): session opened for user root by (uid=0)
Dec 21 12:39:21 raspberrypi sudo[17811]: pam_unix(sudo:session): session closed for user root
Dec 21 12:39:21 raspberrypi sudo[17905]: www-data : TTY=unknown ; PWD=/home/pi/RPi-Jukebox-RFID/htdocs/api ; USER=root ; COMMAND=/home/pi/RPi-Jukebox-RFID/scripts/playout_
Dec 21 12:39:21 raspberrypi sudo[17905]: pam_unix(sudo:session): session opened for user root by (uid=0)
Dec 21 12:39:21 raspberrypi sudo[17812]: pam_unix(sudo:session): session closed for user root
Dec 21 12:39:21 raspberrypi sudo[17923]: www-data : TTY=unknown ; PWD=/home/pi/RPi-Jukebox-RFID/htdocs/api ; USER=root ; COMMAND=/home/pi/RPi-Jukebox-RFID/scripts/playout_
Dec 21 12:39:21 raspberrypi sudo[17923]: pam_unix(sudo:session): session opened for user root by (uid=0)
Dec 21 12:39:22 raspberrypi sudo[17880]: pam_unix(sudo:session): session closed for user root
Dec 21 12:39:22 raspberrypi sudo[17990]: www-data : TTY=unknown ; PWD=/home/pi/RPi-Jukebox-RFID/htdocs/api ; USER=root ; COMMAND=/home/pi/RPi-Jukebox-RFID/scripts/playout_
Dec 21 12:39:22 raspberrypi sudo[17990]: pam_unix(sudo:session): session opened for user root by (uid=0)
Dec 21 12:39:22 raspberrypi sudo[17905]: pam_unix(sudo:session): session closed for user root
Dec 21 12:39:22 raspberrypi sudo[17923]: pam_unix(sudo:session): session closed for user root
Dec 21 12:39:22 raspberrypi sudo[17990]: pam_unix(sudo:session): session closed for user root
Dec 21 12:39:22 raspberrypi sudo[18035]: www-data : TTY=unknown ; PWD=/home/pi/RPi-Jukebox-RFID/htdocs/api ; USER=root ; COMMAND=/usr/bin/echo status
                                         currentsong
                                         close
Dec 21 12:39:22 raspberrypi sudo[18035]: pam_unix(sudo:session): session opened for user root by (uid=0)
Dec 21 12:39:22 raspberrypi sudo[18035]: pam_unix(sudo:session): session closed for user root
Dec 21 12:39:22 raspberrypi sudo[18040]: www-data : TTY=unknown ; PWD=/home/pi/RPi-Jukebox-RFID/htdocs/api ; USER=root ; COMMAND=/home/pi/RPi-Jukebox-RFID/scripts/playout_
Dec 21 12:39:22 raspberrypi sudo[18040]: pam_unix(sudo:session): session opened for user root by (uid=0)
Dec 21 12:39:22 raspberrypi sudo[18040]: pam_unix(sudo:session): session closed for user root
Dec 21 12:39:22 raspberrypi sudo[18078]: www-data : TTY=unknown ; PWD=/home/pi/RPi-Jukebox-RFID/htdocs/api ; USER=root ; COMMAND=/home/pi/RPi-Jukebox-RFID/scripts/playout_
Dec 21 12:39:22 raspberrypi sudo[18078]: pam_unix(sudo:session): session opened for user root by (uid=0)
Dec 21 12:39:23 raspberrypi sudo[18078]: pam_unix(sudo:session): session closed for user root
Dec 21 12:39:27 raspberrypi sudo[18113]: www-data : TTY=unknown ; PWD=/home/pi/RPi-Jukebox-RFID/htdocs/api ; USER=root ; COMMAND=/usr/bin/echo status
                                         currentsong
                                         close
Dec 21 12:39:27 raspberrypi sudo[18113]: pam_unix(sudo:session): session opened for user root by (uid=0)
Dec 21 12:39:27 raspberrypi sudo[18113]: pam_unix(sudo:session): session closed for user root
Dec 21 12:39:27 raspberrypi sudo[18116]: www-data : TTY=unknown ; PWD=/home/pi/RPi-Jukebox-RFID/htdocs/api ; USER=root ; COMMAND=/usr/bin/echo status
                                         currentsong
                                         close

Andreas-Roth-27, 2020-12-21:
How do I recognize if the RFID reader is installed correctly?

s-martin, 2020-12-21:
Which RFID reader do you use?

Andreas-Roth-27, 2020-12-21:
The one from the shopping list (Neuftech) 

https://www.amazon.de/Neuftech-Reader-Kartenlesegerät-Kartenleser-Kontaktlos/dp/B018OYOR3E/ref=sxts_sxwds-bia-wc-nc-drs1_0?__mk_de_DE=ÅMÅŽÕÑ&crid=203HBIG74EJ5R&cv_ct_cx=neuftech+usb+rfid+reader&dchild=1&keywords=neuftech+usb+rfid+reader&pd_rd_i=B018OYOR3E&pd_rd_r=15dad177-c9c2-4d6d-87c9-c905f8edd800&pd_rd_w=kuzp0&pd_rd_wg=lwkpA&pf_rd_p=58d6d532-cd4b-4346-81cc-26b9b56aa9ed&pf_rd_r=7PX6BTTX7ZBZRADG132T&psc=1&qid=1608586153&s=computers&sprefix=neuftech%2Ccomputers%2C228&sr=1-1-606822b7-04c2-4c74-a611-acbe80e94641

Andreas-Roth-27, 2020-12-21:
when I swipe the chip over the reader it beeps and the console displays the ID of the chip (0011959176) and on the next output line it says
bash: 0011959176: command not found

… I hope this is pretty obvious (to the expert) what is wrong

Andreas

s-martin, 2020-12-21:
What’s the output of `cat ~/RPi-Jukebox-RFID/scripts/deviceName.txt`?

s-martin, 2020-12-21:
If it’s empty or not existing please follow the steps in this sub chapter https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/CONFIGURE-stretch#register-your-usb-device-for-the-phoniebox

Andreas-Roth-27, 2020-12-21:
The output is
HID 046a:0023

> Am 21.12.2020 um 23:24 schrieb s-martin <notifications@github.com>:
> 
> cat ~/RPi-Jukebox-RFID/scripts/deviceName.txt

s-martin, 2020-12-21:
Ok, please swipe a card and afterwards the results of 
`tail -n 500 /var/log/syslog`
and
`journalctl -u lighttpd `

please post your answer here on the GitHub website and not via email. Replies via email can’t be formatted and can be hard to read (especially in long logs). Thanks 😊

Andreas-Roth-27, 2020-12-21:
The output you asked for is
	HID 046a:0023
so not empty or not existing

When I run Python or Python3 I get 2 options for my RFID reader

pi@raspberrypi:~/RPi-Jukebox-RFID/settings $ python3 /home/pi/RPi-Jukebox-RFID/scripts/RegisterDevice.py
Choose the reader from list
0 HID 046a:0023
1 HID 046a:0023
2 Logitech USB Mouse
3 HXGCoLtd Keyboard

0 and 1. But I tried both. For both options swiping the card over the reader gets acknowledged by a beep and the ID displayed in the console. 
But I don’t see the ID in the Web interface nor can I control phoniebox with it

Andreas-Roth-27, 2020-12-21:
pi@raspberrypi:~/RPi-Jukebox-RFID/settings $ tail -n 500 /var/log/syslog
Dec 21 23:31:26 raspberrypi mopidy[508]: INFO     [MpdSession-102190] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33320
Dec 21 23:31:26 raspberrypi mopidy[508]: INFO     [MpdSession-102191] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33324
Dec 21 23:31:26 raspberrypi mopidy[508]: INFO     [MpdSession-102192] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33328
Dec 21 23:31:30 raspberrypi mopidy[508]: INFO     [MpdSession-102193] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33332
Dec 21 23:31:31 raspberrypi mopidy[508]: INFO     [MpdSession-102194] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33336
Dec 21 23:31:31 raspberrypi mopidy[508]: INFO     [MpdSession-102195] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33340
Dec 21 23:31:31 raspberrypi mopidy[508]: INFO     [MpdSession-102196] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33344
Dec 21 23:31:31 raspberrypi mopidy[508]: INFO     [MpdSession-102197] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33348
Dec 21 23:31:31 raspberrypi mopidy[508]: INFO     [MpdSession-102198] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33352
Dec 21 23:31:31 raspberrypi mopidy[508]: INFO     [MpdSession-102199] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33356
Dec 21 23:31:31 raspberrypi mopidy[508]: INFO     [MpdSession-102200] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33360
Dec 21 23:31:31 raspberrypi mopidy[508]: INFO     [MpdSession-102201] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33364
Dec 21 23:31:31 raspberrypi mopidy[508]: INFO     [MpdSession-102202] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33368
Dec 21 23:31:31 raspberrypi mopidy[508]: INFO     [MpdSession-102203] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33372
Dec 21 23:31:31 raspberrypi mopidy[508]: INFO     [MpdSession-102204] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33376
Dec 21 23:31:31 raspberrypi mopidy[508]: INFO     [MpdSession-102205] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33380
Dec 21 23:31:31 raspberrypi mopidy[508]: INFO     [MpdSession-102206] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33384
Dec 21 23:31:36 raspberrypi mopidy[508]: INFO     [MpdSession-102207] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33388
Dec 21 23:31:36 raspberrypi mopidy[508]: INFO     [MpdSession-102208] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33392
Dec 21 23:31:36 raspberrypi mopidy[508]: INFO     [MpdSession-102209] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33396
Dec 21 23:31:36 raspberrypi mopidy[508]: INFO     [MpdSession-102210] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33400
Dec 21 23:31:36 raspberrypi mopidy[508]: INFO     [MpdSession-102211] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33404
Dec 21 23:31:36 raspberrypi mopidy[508]: INFO     [MpdSession-102212] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33408
Dec 21 23:31:37 raspberrypi mopidy[508]: INFO     [MpdSession-102213] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33412
Dec 21 23:31:37 raspberrypi mopidy[508]: INFO     [MpdSession-102214] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33416
Dec 21 23:31:37 raspberrypi mopidy[508]: INFO     [MpdSession-102215] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33420
Dec 21 23:31:37 raspberrypi mopidy[508]: INFO     [MpdSession-102216] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33424
Dec 21 23:31:37 raspberrypi mopidy[508]: INFO     [MpdSession-102217] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33428
Dec 21 23:31:37 raspberrypi mopidy[508]: INFO     [MpdSession-102218] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33432
Dec 21 23:31:37 raspberrypi mopidy[508]: INFO     [MpdSession-102219] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33436
Dec 21 23:31:37 raspberrypi mopidy[508]: INFO     [MpdSession-102220] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33440
Dec 21 23:31:42 raspberrypi mopidy[508]: INFO     [MpdSession-102221] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33444
Dec 21 23:31:42 raspberrypi mopidy[508]: INFO     [MpdSession-102222] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33448
Dec 21 23:31:42 raspberrypi mopidy[508]: INFO     [MpdSession-102223] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33452
Dec 21 23:31:42 raspberrypi mopidy[508]: INFO     [MpdSession-102224] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33456
Dec 21 23:31:42 raspberrypi mopidy[508]: INFO     [MpdSession-102225] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33460
Dec 21 23:31:43 raspberrypi mopidy[508]: INFO     [MpdSession-102226] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33464
Dec 21 23:31:43 raspberrypi mopidy[508]: INFO     [MpdSession-102227] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33468
Dec 21 23:31:44 raspberrypi mopidy[508]: INFO     [MpdSession-102228] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33472
[... cut ...]
Dec 21 23:35:19 raspberrypi mopidy[508]: INFO     [MpdSession-102686] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:35314
Dec 21 23:35:19 raspberrypi mopidy[508]: INFO     [MpdSession-102687] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:35318
Dec 21 23:35:20 raspberrypi mopidy[508]: INFO     [MpdSession-102688] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:35322
Dec 21 23:35:20 raspberrypi mopidy[508]: INFO     [MpdSession-102689] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:35326

Andreas-Roth-27, 2020-12-21:
This is the log from the tail command

Andreas-Roth-27, 2020-12-21:
pi@raspberrypi:~/RPi-Jukebox-RFID/settings $ tail -n 500 /var/log/syslog
Dec 21 23:31:26 raspberrypi mopidy[508]: INFO     [MpdSession-102190] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33320
Dec 21 23:31:26 raspberrypi mopidy[508]: INFO     [MpdSession-102191] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33324
Dec 21 23:31:26 raspberrypi mopidy[508]: INFO     [MpdSession-102192] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33328
Dec 21 23:31:30 raspberrypi mopidy[508]: INFO     [MpdSession-102193] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33332
Dec 21 23:31:31 raspberrypi mopidy[508]: INFO     [MpdSession-102194] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33336
Dec 21 23:31:31 raspberrypi mopidy[508]: INFO     [MpdSession-102195] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33340
Dec 21 23:31:31 raspberrypi mopidy[508]: INFO     [MpdSession-102196] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33344
Dec 21 23:31:31 raspberrypi mopidy[508]: INFO     [MpdSession-102197] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33348
Dec 21 23:31:31 raspberrypi mopidy[508]: INFO     [MpdSession-102198] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33352
Dec 21 23:31:31 raspberrypi mopidy[508]: INFO     [MpdSession-102199] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33356
Dec 21 23:31:31 raspberrypi mopidy[508]: INFO     [MpdSession-102200] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33360
Dec 21 23:31:31 raspberrypi mopidy[508]: INFO     [MpdSession-102201] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33364
Dec 21 23:31:31 raspberrypi mopidy[508]: INFO     [MpdSession-102202] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33368
Dec 21 23:31:31 raspberrypi mopidy[508]: INFO     [MpdSession-102203] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33372
Dec 21 23:31:31 raspberrypi mopidy[508]: INFO     [MpdSession-102204] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33376
Dec 21 23:31:31 raspberrypi mopidy[508]: INFO     [MpdSession-102205] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33380
Dec 21 23:31:31 raspberrypi mopidy[508]: INFO     [MpdSession-102206] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33384
Dec 21 23:31:36 raspberrypi mopidy[508]: INFO     [MpdSession-102207] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33388
Dec 21 23:31:36 raspberrypi mopidy[508]: INFO     [MpdSession-102208] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33392
Dec 21 23:31:36 raspberrypi mopidy[508]: INFO     [MpdSession-102209] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33396
Dec 21 23:31:36 raspberrypi mopidy[508]: INFO     [MpdSession-102210] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33400
Dec 21 23:31:36 raspberrypi mopidy[508]: INFO     [MpdSession-102211] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33404
Dec 21 23:31:36 raspberrypi mopidy[508]: INFO     [MpdSession-102212] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33408
Dec 21 23:31:37 raspberrypi mopidy[508]: INFO     [MpdSession-102213] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33412
Dec 21 23:31:37 raspberrypi mopidy[508]: INFO     [MpdSession-102214] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33416
Dec 21 23:31:37 raspberrypi mopidy[508]: INFO     [MpdSession-102215] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33420
Dec 21 23:31:37 raspberrypi mopidy[508]: INFO     [MpdSession-102216] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33424
Dec 21 23:31:37 raspberrypi mopidy[508]: INFO     [MpdSession-102217] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33428
Dec 21 23:31:37 raspberrypi mopidy[508]: INFO     [MpdSession-102218] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33432
Dec 21 23:31:37 raspberrypi mopidy[508]: INFO     [MpdSession-102219] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33436
Dec 21 23:31:37 raspberrypi mopidy[508]: INFO     [MpdSession-102220] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33440
Dec 21 23:31:42 raspberrypi mopidy[508]: INFO     [MpdSession-102221] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33444
Dec 21 23:31:42 raspberrypi mopidy[508]: INFO     [MpdSession-102222] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33448
Dec 21 23:31:42 raspberrypi mopidy[508]: INFO     [MpdSession-102223] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33452
Dec 21 23:31:42 raspberrypi mopidy[508]: INFO     [MpdSession-102224] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33456
Dec 21 23:31:42 raspberrypi mopidy[508]: INFO     [MpdSession-102225] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33460
Dec 21 23:31:43 raspberrypi mopidy[508]: INFO     [MpdSession-102226] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33464
Dec 21 23:31:43 raspberrypi mopidy[508]: INFO     [MpdSession-102227] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33468
Dec 21 23:31:44 raspberrypi mopidy[508]: INFO     [MpdSession-102228] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:33472
[... cut ...]
Dec 21 18:49:19 raspberrypi sudo[8660]: pam_unix(sudo:session): session opened for user root by (uid=0)
Dec 21 18:49:20 raspberrypi sudo[8604]: pam_unix(sudo:session): session closed for user root
Dec 21 18:49:20 raspberrypi sudo[8677]: www-data : TTY=unknown ; PWD=/home/pi/RPi-Jukebox-RFID/htdocs/api ; USER=root ; COMMAND=/home/pi/RPi-Jukebox-RFID/scripts
Dec 21 18:49:20 raspberrypi sudo[8677]: pam_unix(sudo:session): session opened for user root by (uid=0)
Dec 21 18:49:20 raspberrypi sudo[8660]: pam_unix(sudo:session): session closed for user root
lines 66-129

Andreas-Roth-27, 2020-12-21:
And these are the first lines of the output of the command journalctl -u lighttpd

Andreas-Roth-27, 2020-12-21:
Ok -sorry for accidentally closing the issue.

I put the lengthy outputs of both commands on the GitHub website.

The output of journalctl was very long and I actually did not know how to get full output. Instead I copied the first around 70 lines. Ok?

s-martin, 2020-12-21:
Is the rfid reader enabled in the web interface?
See screenshot here https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/MANUAL#external-devices--interfaces

Andreas-Roth-27, 2020-12-21:
Yes it is
[image]

Andreas-Roth-27, 2020-12-22:
Does the web interface with its indication that the RFID reader is enabled mean that just everything is ok until here? Where do I have to look now?

Andreas-Roth-27, 2020-12-22:
I mean, when I swipe the RFID chip over the reader its number is shown in the console (terminal) of the Raspbian right after. The green indication in the web interface is also present. What's missing?

Andreas-Roth-27, 2020-12-22:
And once again: how about this output line 
bash: "tag-ID": command not found      "tag-ID" equals the ID of the RFID tag
which means, that the system is able to read the RFID tag but ...

Andreas-Roth-27, 2021-01-03:
I should close this. With the help of the hint at the end of issue # 1202 there is sound. 
https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/1202#issuecomment-753669568

closed 2021-01-03 by Andreas-Roth-27