MiczFlor/RPi-Jukebox-RFID #1221: Wiki entry "Hack Playlist triggered by GPIO Button" not up to date with new GPIO-Server?

derdon222, 2020-12-20:
## Bug
Wiki entry "Hack Playlist triggered by GPIO Button" not up to date with new GPIO-Server?
### What I did
I tried to follow the Instructions in the wiki to assign playlists to hardware buttons. I did not find the mentioned file gpio-buttons.py but only the gpio-settings.ini. Doing the changes described in the wiki there did not have any effect. 
I could not find any hint on assigning playlists to GPIO-Buttons in the documentation for the new system either.
Volume rotary encoder, next prev play pause buttons all work fine!
Thank you for this great project!

### What happened

Buttons not working

### I expected this to happen

start the assigned playlist

### Further information that might help

I am on an very fresh installation (Dec 15. 2020) by provided script 

## Software
I am on an very fresh installation (Dec 15. 2020) by provided script on Raspberry Pi OS Lite 5.4
### Base image and version

### Branch / Release

### Installscript

## Hardware

### RaspberryPi version

### RFID Reader

### Soundcard

### Other notable hardware

s-martin, 2020-12-20:
You would need to add additional function calls to `components/gpio_control/function_calls.py` and set these function calls in gpio_settings.ini to a Button.

but be aware this hack assigns a playlist hardcoded to a button.

derdon222, 2020-12-27:
Thank you for the quick reply! I am afraid that I need a little more detail: I can open the files and as far as I understood, this for example is a function call: 

<CODE>
def functionHurra(*args):
    function_call("{command} -c=shutdown".format(command=playout_control), shell=True)
</CODE>

Now I have no idea how this must look like to trigger a (spotify) playlist
By the way: I would like to execute a shell script with one button - Is the way similar?

This is an example from gpio_settings.ini where I guess I have to do something with he function
<CODE>
[Hurra]
enabled: True
Type: Button
Pin: 27 #myPin here
pull_up: True
hold_time: 0.3
functionCall: functionCallPlayerPause
</CODE>

Best wishes!

closed 2024-03-01 by AlvinSchiller