MiczFlor/RPi-Jukebox-RFID #1223: Question: correct wires

UndIch, 2020-12-21:
So like always - having everything finishes I stumbled over some info which does concern me a little bit - because I don’t want to build a fire hazard.

I rebuilt my Phoniebox - with a new 3B+, Neuftech reader, USB Logitech W150 (incl. sound card, powered via Pi’s USB), touch buttons (TTP223), On-Off-shim.  
because of the small space in my box - I used a colored flat cable from reichel and plugged everything with crimped duponts in the flat cable - incl. the on-off-shim. 
now I read that for powering the Pi it is close to 2 A - and the flat cable is made out of AWG28 (~0,1mm²)- and google told me that it is max 0,3 Amps. Everything is finished - so I really need to start all over?
Why are all the breadboard sets equipped with those then? 
One additional question: when using the on-off-shim the upper power sources are blocked - so there is only one 3V left - but the touch buttons need 2-5 V - so two are. Working fine - with a third it’s tricky and 4 are not working. 

Thanks and have a merry Christmas and a better year  then this one

that-coding-human, 2020-12-26:
I am using the on off shim as well, but I plugged it directly on the gpio pin panel. The on off shim is a little sensible in regards of undervoltage if you connect it to the raspi using cables (e.g. jumper cables). It's kind of a good practice to avoid this.

It is correct that the on off shim blocks a lot more pins than it actually uses.
I handled that by using a 3x GPIO extension board from AZ-Delivery. Nothing special, but it replicates the gpio pin panel 3 times an can be mounted right on top of the pi. This also solved undervoltage issues.

This is very handy if you want to connect several boards to the gpio panel, that use more space than required. 
I connected buttons, on-off shim and a hifiberry miniamp using such a board.

It saves overall space, but increases the height required on top of the pi a bit.

Maybe this helps you with your design.

UndIch, 2020-12-26:
Thanks for the input. It is working properly and it is perfect for my design because due to the on-off-shim the plug is kind of at an awkward position - not suitable for most of the cases. My main concern is more that there is to much currency on those small wires - that this might catch fire or sth.

s-martin, 2020-12-26:
I currently also have some undervoltage issues with the onoffshim and jumper cables.

I also ordered the extension board to plug the shim directly on the GPIO headers.

that-coding-human, 2020-12-27:
Habe you tried using an angled plug on the one-off shim, if you have narrow space? Maybe i am playing "captain obvious", but it might be worth mentioning. Sometimes just using an angled plug solves it. 

Regarding the issue with your 0,3 amp wires, can you provide a link to some more documentation about it? 
Have you tried it for a while an observed a warming of the cable?