MiczFlor/RPi-Jukebox-RFID #1224: 🚀 | Make audio configuration more „bullet proof“

s-martin, 2020-12-21:
## Feature Description

### What functionality would you like to see in your phoniebox?

Configuration of audio during installation should be improved, e.g.

- automatically configure sound cards
- Consider RPi audio changes from [here](https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#raspberry-pi-update-may-2020-broke-my-audio) and [here](https://github.com/MiczFlor/RPi-Jukebox-RFID/discussions/1208)
- Improve wiki section https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#audio-is-not-working
- consider behavior, when hdmi cable is connected or not 

### How do you envision the feature to work from a users perspective?

Installation easier for users 

### Further information that might help

Users have to ask lots of questions(e.g. #1220), because it’s difficult and the current state is not very well documented.

Andreas-Roth-27, 2020-12-30:
Is there a short cut to circumference difficult installation for the time being?

s-martin, 2022-12-20:
New future3 implementation already addresses this.

closed 2022-12-20 by s-martin