MiczFlor/RPi-Jukebox-RFID #1225: GPIO Buttons do not work after update to new GPIO control

andisger, 2020-12-21:
Hi everybody,

I just updated my RPi Phoniebox to the new version 2.2. 
I had several issues during the update which I could, however, easily resolve, for example: taking the spotify account config didnt work.
So, after 30 mins: everything was up to date and running...everything but the GPIO controls which are now defined by the new settings file.
Regarding my setup :
- i use physical resistors very similar to the original setup describe in the wiki
- i use a a on-off shim
- RPi3B+

No button works actually and i am not able to identify the root cause. I just wanted to have the same functionality as before (see below). Since i use physical pull up resistors i deactivated them in the settings. I couldnt find any documentation to the new settings file, yet, so maybe you can help me with a few ideas regarding potential root causes.

I restarted the service, checked the pin numbering, updated the installation of the phoniebox 3 times, ...
Starting the install.sh script in the components/gpio-control folder does not work.
Is there anything obvious that i am missing? Is there anything else that i could provide to help debugging?

A minor finding : within the gpio settings example files there are a few controls where it is stated enable: True other enable**D**: True.

Does anybody else face similar issues with the new GPIO configuation?

Thanks a lot for your help.

here is the config of my old GPIO buttons.py file:
```
led = LED(22)
led.on()

shut = Button(12, hold_time=2)
vol0 = Button(13)
volU = Button(16,hold_time=0.3,hold_repeat=True)
volD = Button(19,hold_time=0.3,hold_repeat=True)
next = Button(26)
prev = Button(20)
halt = Button(21)
```
here is my new GPIO controls file
`[DEFAULT]
enabled: True

[PlayPause]
enabled: True
Type: Button
Pin: 21
#pull_up: True
hold_time: 0.3
functionCall: functionCallPlayerPause

[Shutdown]
enabled: True
Type:  Button
Pin: 12
#pull_up: True
hold_time: 2
functionCall: functionCallShutdown

[VolumeUp]
enabled: True
Type:  Button
Pin: 16
#pull_up: True
hold_time: 0.3
hold_repeat: True
functionCall: functionCallVolU

[VolumeDown]
enabled: True
Type:  Button
Pin: 19
#pull_up: True
hold_time: 0.3
hold_repeat: True
functionCall: functionCallVolD

[NextSong]
enabled: True
Type:  Button
Pin: 26
#pull_up: True
hold_time: 0.3
functionCall: functionCallPlayerNext

[PrevSong]
enabled: True
Type:  Button
Pin: 20
#pull_up: True
hold_time: 0.3
functionCall: functionCallPlayerPrev

[Halt]
enabled: True
Type:  Button
Pin: 21
#pull_up: True
hold_time: 0.3
functionCall: functionCallPlayerPauseForce

[StatusLED]
enabled: True
Type: MPDStatusLED
Pin: 22
`

s-martin, 2020-12-21:
I think you have set pull up explicitly to `False`, commenting defaults to True, if I remember correctly.

s-martin, 2020-12-21:
Please note also that Pin is actually the *GPIO number**

andisger, 2020-12-21:
thanks for the quick reply.
Regarding the GPIO number: the number is exactly as described here  https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Using-GPIO-hardware-buttons
--> GPIO26 - Pin37 --> i use 26 in the code, right? 

Regarding the pull up, i dont really understand what you mean: 
due to the usage of hardware resistors i thought that i do not need the pull up true statement. I didnt use the software pull ups in the old version and it worked. 
however, it doesnt work with Pull up : True either. 

What does the first line of the gpio-settings actually define?
```
[Default] 
Enabled: True

```  

thanks again!

s-martin, 2020-12-21:
Please set 
`pull_up: False`

andisger, 2020-12-21:
Doesnt work. 
when i manually start the python script gpio_control i get the following infos (not sure if this helps).
```
INFO:root:['PlayPause', 'Shutdown', 'VolumeUp', 'VolumeDown', 'NextSong', 'PrevSong', 'Halt', 'StatusLED']
INFO:root:adding GPIO-Device, PlayPause
PlayPause
INFO:root:adding GPIO-Device, Shutdown
Shutdown
INFO:root:adding GPIO-Device, VolumeUp
VolumeUp
INFO:root:adding GPIO-Device, VolumeDown
VolumeDown
INFO:root:adding GPIO-Device, NextSong
NextSong
INFO:root:adding GPIO-Device, PrevSong
PrevSong
INFO:root:adding GPIO-Device, Halt
Halt
Traceback (most recent call last):
  File "gpio_control.py", line 104, in <module>
    devices = get_all_devices(config)
  File "gpio_control.py", line 82, in get_all_devices
    device = generate_device(config[section], section)
  File "gpio_control.py", line 54, in generate_device
    pull_up_down=config.get('pull_up_down', fallback=GPIO.PUD_UP))
  File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/GPIODevices/simple_button.py", line 74, in __init__
    bouncetime=self.bouncetime)
RuntimeError: Conflicting edge detection already enabled for this GPIO channel
```

andisger, 2020-12-21:
here some addition log as a result of the following statement
$ journalctl -u phoniebox-gpio-control.service -f

`pi@raspberrypi:~/RPi-Jukebox-RFID/components/gpio_control $ journalctl -u phoniebox-gpio-control.service -f
-- Logs begin at Thu 2019-02-14 10:11:59 GMT. --
Dec 21 16:33:43 raspberrypi PhonieboxGPIOControl[21950]:     devices = get_all_devices(config)
Dec 21 16:33:43 raspberrypi PhonieboxGPIOControl[21950]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/gpio_control.py", line 82, in get_all_devices
Dec 21 16:33:43 raspberrypi PhonieboxGPIOControl[21950]:     device = generate_device(config[section], section)
Dec 21 16:33:43 raspberrypi PhonieboxGPIOControl[21950]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/gpio_control.py", line 54, in generate_device
Dec 21 16:33:43 raspberrypi PhonieboxGPIOControl[21950]:     pull_up_down=config.get('pull_up_down', fallback=GPIO.PUD_UP))
Dec 21 16:33:43 raspberrypi PhonieboxGPIOControl[21950]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/GPIODevices/simple_button.py", line 74, in __init__
Dec 21 16:33:43 raspberrypi PhonieboxGPIOControl[21950]:     bouncetime=self.bouncetime)
Dec 21 16:33:43 raspberrypi PhonieboxGPIOControl[21950]: RuntimeError: Conflicting edge detection already enabled for this GPIO channel
Dec 21 16:33:43 raspberrypi systemd[1]: phoniebox-gpio-control.service: Main process exited, code=exited, status=1/FAILURE
Dec 21 16:33:43 raspberrypi systemd[1]: phoniebox-gpio-control.service: Failed with result 'exit-code'.
Dec 21 16:33:46 raspberrypi systemd[1]: phoniebox-gpio-control.service: Service RestartSec=3s expired, scheduling restart.
Dec 21 16:33:46 raspberrypi systemd[1]: phoniebox-gpio-control.service: Scheduled restart job, restart counter is at 86.
Dec 21 16:33:46 raspberrypi systemd[1]: Stopped Phoniebox GPIO Control Service.
Dec 21 16:33:46 raspberrypi systemd[1]: Started Phoniebox GPIO Control Service.
Dec 21 16:33:47 raspberrypi PhonieboxGPIOControl[22055]: INFO:root:['PlayPause', 'Shutdown', 'VolumeUp', 'VolumeDown', 'NextSong', 'PrevSong', 'Halt', 'StatusLED']
Dec 21 16:33:47 raspberrypi PhonieboxGPIOControl[22055]: INFO:root:adding GPIO-Device, PlayPause
Dec 21 16:33:47 raspberrypi PhonieboxGPIOControl[22055]: INFO:root:adding GPIO-Device, Shutdown
Dec 21 16:33:47 raspberrypi PhonieboxGPIOControl[22055]: INFO:root:adding GPIO-Device, VolumeUp
Dec 21 16:33:47 raspberrypi PhonieboxGPIOControl[22055]: INFO:root:adding GPIO-Device, VolumeDown
Dec 21 16:33:47 raspberrypi PhonieboxGPIOControl[22055]: INFO:root:adding GPIO-Device, NextSong
Dec 21 16:33:47 raspberrypi PhonieboxGPIOControl[22055]: INFO:root:adding GPIO-Device, PrevSong
Dec 21 16:33:47 raspberrypi PhonieboxGPIOControl[22055]: INFO:root:adding GPIO-Device, Halt
Dec 21 16:33:47 raspberrypi PhonieboxGPIOControl[22055]: PlayPause
Dec 21 16:33:47 raspberrypi PhonieboxGPIOControl[22055]: Shutdown
Dec 21 16:33:47 raspberrypi PhonieboxGPIOControl[22055]: VolumeUp
Dec 21 16:33:47 raspberrypi PhonieboxGPIOControl[22055]: VolumeDown
Dec 21 16:33:47 raspberrypi PhonieboxGPIOControl[22055]: NextSong
Dec 21 16:33:47 raspberrypi PhonieboxGPIOControl[22055]: PrevSong
Dec 21 16:33:47 raspberrypi PhonieboxGPIOControl[22055]: Halt
Dec 21 16:33:47 raspberrypi PhonieboxGPIOControl[22055]: Traceback (most recent call last):
Dec 21 16:33:47 raspberrypi PhonieboxGPIOControl[22055]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/gpio_control.py", line 104, in <module>
Dec 21 16:33:47 raspberrypi PhonieboxGPIOControl[22055]:     devices = get_all_devices(config)
Dec 21 16:33:47 raspberrypi PhonieboxGPIOControl[22055]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/gpio_control.py", line 82, in get_all_devices
Dec 21 16:33:47 raspberrypi PhonieboxGPIOControl[22055]:     device = generate_device(config[section], section)
Dec 21 16:33:47 raspberrypi PhonieboxGPIOControl[22055]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/gpio_control.py", line 54, in generate_device
Dec 21 16:33:47 raspberrypi PhonieboxGPIOControl[22055]:     pull_up_down=config.get('pull_up_down', fallback=GPIO.PUD_UP))
Dec 21 16:33:47 raspberrypi PhonieboxGPIOControl[22055]:   File "/home/pi/RPi-Jukebox-RFID/components/gpio_control/GPIODevices/simple_button.py", line 74, in __init__
Dec 21 16:33:47 raspberrypi PhonieboxGPIOControl[22055]:     bouncetime=self.bouncetime)
Dec 21 16:33:47 raspberrypi PhonieboxGPIOControl[22055]: RuntimeError: Conflicting edge detection already enabled for this GPIO channel
Dec 21 16:33:47 raspberrypi systemd[1]: phoniebox-gpio-control.service: Main process exited, code=exited, status=1/FAILURE
Dec 21 16:33:47 raspberrypi systemd[1]: phoniebox-gpio-control.service: Failed with result 'exit-code'.
Dec 21 16:33:50 raspberrypi systemd[1]: phoniebox-gpio-control.service: Service RestartSec=3s expired, scheduling restart.
Dec 21 16:33:50 raspberrypi systemd[1]: phoniebox-gpio-control.service: Scheduled restart job, restart counter is at 87.
Dec 21 16:33:50 raspberrypi systemd[1]: Stopped Phoniebox GPIO Control Service.
Dec 21 16:33:50 raspberrypi systemd[1]: Started Phoniebox GPIO Control Service.
Dec 21 16:33:50 raspberrypi PhonieboxGPIOControl[22311]: INFO:root:['PlayPause', 'Shutdown', 'VolumeUp', 'VolumeDown', 'NextSong', 'PrevSong', 'Halt', 'StatusLED']
Dec 21 16:33:50 raspberrypi PhonieboxGPIOControl[22311]: INFO:root:adding GPIO-Device, PlayPause
[... cut ...]
Dec 21 16:33:50 raspberrypi PhonieboxGPIOControl[22311]: RuntimeError: Conflicting edge detection already enabled for this GPIO channel
Dec 21 16:33:50 raspberrypi systemd[1]: phoniebox-gpio-control.service: Main process exited, code=exited, status=1/FAILURE
Dec 21 16:33:50 raspberrypi systemd[1]: phoniebox-gpio-control.service: Failed with result 'exit-code'.
Dec 21 16:33:53 raspberrypi systemd[1]: phoniebox-gpio-control.service: Service RestartSec=3s expired, scheduling restart.
Dec 21 16:33:53 raspberrypi systemd[1]: phoniebox-gpio-control.service: Scheduled restart job, restart counter is at 88.
`

s-martin, 2020-12-21:
@veloxidSchweiz, can you help here?

oheuschele, 2020-12-23:
I think the problem is that you use a pin twice, "PlayPause" and "Halt" are using both pin 21.

andisger, 2020-12-24:
manchmal ist die Lösung so einfach...und man sucht nach den Fehlern im Code, dabei ist meine stumpfe Eingabe der Grund.
@oheuschele @s-martin Vielen Dank für Euren Support und eine schöne Weihnachtszeit

closed 2020-12-25 by s-martin