MiczFlor/RPi-Jukebox-RFID #1226: 🐛 | Audio and Spotify is not working

TheRealNic, 2020-12-21:
## Bug

### What I did

I installed a fresh raspberry pi 3b+ with the latest raspbian (2020-08-20-raspios-buster-armhf-lite). Then followed the installation procedure for the hifiberry mini amp (added asound.conf ) (which works via speaker-test), installed a wifi dongle and then installed phoniebox (Spotify edition, Master as IFace, mpd configured, did not start gpio control service, RC522 RFID reader). Then I added the section from https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/HiFiBerry-Soundcard-Details to the mpd config. 

### What happened

When I try to play an uploaded song I don't hear anything and the player doesn't seem to play the song (no time tracking is shown). An added spotify stream is not even starting. I rescanned the library multiple times.

### I expected this to happen

To hear the content, either spotify or local files.

### Further information that might help

[mopidy.log](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/5726946/mopidy.log)

## Software

### Base image and version

2020-08-20-raspios-buster-armhf-lite

### Branch / Release

master up-to-date

### Installscript

One line installer

## Hardware

### RaspberryPi version

3 B+

### RFID Reader

RC522

### Soundcard

HiFiBerry Miniamp

TheRealNic, 2020-12-22:
Additional information: Mopidy and Iris is working, I can play music via that interface and get an output via the miniamp... I reinstalled the phoniebox software a couple of times in the last 2 hours, but the phoniebox interface is still not working...

0x0lli, 2020-12-22:
did you run the "search files and folders"-command in phoniebox web-frontend?

TheRealNic, 2020-12-22:
Yes, a couple of times

TheRealNic, 2020-12-22:
Another question, just to shure up all bases: If I want to add a card to the following spotify URL: https://open.spotify.com/album/3esEgXmUs1zjBVqijbZ6RQ
What exactly do I need to enter into the Stream URL filed on the card id tab? Maybe I am doing that wrong?

s-martin, 2020-12-22:
Please check here how to use Spotify URI

https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/1153#issuecomment-736366718

And maybe (but that’s pretty dated info):
https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/docs/SPOTIFY-INTEGRATION.md#how-to-use-spotify

TheRealNic, 2020-12-22:
Thanks so much, that actually helped, so now I'm able to play spotify links in the phonie web interface... Guess I tried every other way of adding a spotify link except this one ☹️  You can close this, now I only need to get the buttons working...

closed 2020-12-23 by s-martin