MiczFlor/RPi-Jukebox-RFID #1227: 🐛 | BUG SUMMARY

Andreas-Roth-27, 2020-12-22:
SETUP
======
I installed the raspberry pi 3 model B with the buster image and ran the current one line installer script (i.e. for phoniebox 2.2). My hardware has no external sound card just the audio jack with connected loud speakers.

WHAT I HAVE
===========
I am able to run phoniebox, download mp3 and have access to Spotify (premium account). The sound is ok. I am able to listen to music both locally stored mp3's and Spotify.
When I swipe the RFID chip over the reader its number or id is shown in the console (terminal) of the Raspbian right after. The green indication in the web interface for an activated RFID reader is also present. 

[image]

BUG
====
When I swipe the RFID chip over the reader its number or id is shown in the console (terminal) of the Raspbian right after and there is also an output line with 
                bash: "tag-ID": command not found  (correct ID but...)
And I cannot use the RFID tag-ID in the web interface of the phoniebox. What is wrong?

cat /home/pi/RPi-Jukebox-RFID/settings/version shows this "2.2 - 305325d - master"
amixer scontrols this "Simple mixer control 'Master',0" and "Simple mixer control 'Capture',0"
sudo cat /sys/firmware/devicetree/base/model shows this "Raspberry Pi 3 Model B Rev 1.2"

s-martin, 2020-12-22:
Can you activate debug logging in the web interface 
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/MANUAL#debug-log-settings

and swipe again?

you should see the log entries also in the web interface

Andreas-Roth-27, 2020-12-22:
Yes, I am using Spotify (premium).
Here is the screen shot after having set all the debug parameters to "TRUE"
[image]

Andreas-Roth-27, 2020-12-22:
Array
(
)
Array
(
    [base_url] => 
    [base_path] => /home/pi/RPi-Jukebox-RFID
    [local_url] => 192.168.178.177
    [scripts_abs] => /home/pi/RPi-Jukebox-RFID/scripts
    [shared_abs] => /home/pi/RPi-Jukebox-RFID/shared
    [settings_abs] => /home/pi/RPi-Jukebox-RFID/settings
    [settings_lang] => en-UK
    [url_abs] => http://192.168.178.177/cardRegisterNew.php
)

Andreas-Roth-27, 2020-12-22:
[image]

Andreas-Roth-27, 2020-12-22:
[image]

Andreas-Roth-27, 2020-12-22:
VAR COMMAND: getchapters
VAR VALUE: 
current file: /home/pi/RPi-Jukebox-RFID/shared/audiofolders/
chapters file: /home/pi/RPi-Jukebox-RFID/shared/audiofolders.chapters.json
chapters for extension enabled: 0
chapters for duration enabled: 0

  # function handleGet() 
$responseList: {"volume":"","repeat":"0","random":"0","single":"0","consume":"0","playlist":"4","playlistlength":"26","xfade":"0","state":"stop","nextsong":"0","nextsongid":"27","chapters":[]}GET
# WebApp API # /home/pi/RPi-Jukebox-RFID/htdocs/api/player.php
  # $_SERVER['REQUEST_METHOD']: GET
  # function execSuccessfully: echo 'status
currentsong
close' | nc -w 1 localhost 6600
  # function execSuccessfully: /home/pi/RPi-Jukebox-RFID/scripts/playout_controls.sh -c=getvolume########### SCRIPT playout_controls.sh (2020-12-22.22:02:05) ##
  ######### SCRIPT inc.readArgsFromCommandLine.sh (2020-12-22.22:02:05) ##
########### SCRIPT inc.writeGlobalConf.sh (2020-12-22.22:02:05) ##
  # VAR CARDID: 
  # VAR COMMAND: getvolume
  # VAR FOLDER: 
  # VAR VALUE: 
VAR COMMAND: getvolume
VAR VALUE: 
current file: /home/pi/RPi-Jukebox-RFID/shared/audiofolders/
chapters file: /home/pi/RPi-Jukebox-RFID/shared/audiofolders.chapters.json
chapters for extension enabled: 0
chapters for duration enabled: 0
#  getvolume
   VOLPERCENT: | VOLUMEMANAGER:amixer

  # function execSuccessfully: /home/pi/RPi-Jukebox-RFID/scripts/playout_controls.sh -c=getchapters########### SCRIPT playout_controls.sh (2020-12-22.22:02:05) ##
  ######### SCRIPT inc.readArgsFromCommandLine.sh (2020-12-22.22:02:05) ##
  # VAR CARDID: 
  # VAR COMMAND: getchapters
  # VAR FOLDER: 
  # VAR VALUE: 
VAR COMMAND: getchapters
VAR VALUE: 
current file: /home/pi/RPi-Jukebox-RFID/shared/audiofolders/
chapters file: /home/pi/RPi-Jukebox-RFID/shared/audiofolders.chapters.json

sebrep, 2021-07-16:
I struggled with this soo much. Let me guess you are using RFC522 as a card reader right? And you have no cable from the IRQ pin to the raspberry right? If so, just add a cable from the RFC522 IRQ port to the pin 18 or GPIO 24

s-martin, 2021-07-17:
@Andreas-Roth-27 if you can confirm this solves your issue, I would add that to the wiki.

Andreas-Roth-27, 2021-07-18:
Sorry for answering so late. I was busy and could not get to my main computer.
In fact, I did not put a cable of any kind to solve the problem.
I manually registered the RFID reader like that:

"...You can manually register a USB device manually like this:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/CONFIGURE-stretch#register-your-usb-device-for-the-phoniebox

Register your USB device for the Phoniebox
Now we know a lot about the RFID reader attached to our RPi. In order to use it as a controller for our Phoniebox, the software needs to know which device to listen to.
To register your device, go to the directory containing all the scripts and run the file RegisterDevice.py.
cd /home/pi/RPi-Jukebox-RFID/scripts/
python3 RegisterDevice.py
This will bring up a list of one or more devices. Spot the device you are using as a RFID reader, type the number (in this case 2) and hit Enter.
Choose the reader from list
   0 Chicony USB Keyboard
   1 Chicony USB Keyboard
   2 Sycreader USB Reader
   3 Logitech USB-PS/2 Optical Mouse 
You can check if your device was registered properly by taking a look inside the file that was just generated, called deviceName.txt.
cat deviceName.txt
Now your Phoniebox knows which device to listen to when you are swiping your cards or keyrings."

In my case there was only one device called (0) called  HXGCoLtd.
This fixed it.

Since I am still a newly to this platform I highly appreciate a feedback. if my comment on this subject was right or misleading or so.

Thank you

s-martin, 2021-08-10:
Was already in the wiki, see https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Wiring_for_RC522_card_reader

closed 2021-08-10 by s-martin