MiczFlor/RPi-Jukebox-RFID #1237: Spotify: Albums and single tracks work, playlists do not

derdon222, 2020-12-27:
## Bug

### What I did

I assigned a spotify playlist(5 songs) via web interface. Links had the format <code>spotify:playlist:0g7Wc... </code>

### What happened
The player in the web interface shows the right picture for the first song, but playback does not start. Not even if I click the play button in the web interface

### I expected this to happen
Playback of the playlist starts

### Further information that might help
any spotify links with "album" instead of "playlist" work fine, like this one:
`spotify:album:32UfyQkxTGaQ1Vw1LJ6u9F`

## Software
I am on an very fresh installation (Dec 15. 2020) by provided script on Raspberry Pi OS Lite 5.4

### RaspberryPi version

Pi 3

s-martin, 2020-12-27:
What's the output of `journalctl -u mopidy`?

derdon222, 2020-12-27:
it's this:
`ournalctl -u mopidy
-- Logs begin at Sun 2020-12-27 13:02:46 CET, end at Sun 2020-12-27 13:11:41 CET. --
Dec 27 13:02:57 JaroniBox systemd[1]: Starting Mopidy music server...
Dec 27 13:02:58 JaroniBox systemd[1]: Started Mopidy music server.
Dec 27 13:03:03 JaroniBox mopidy[3099]: INFO     [MainThread] mopidy.__main__ Starting Mopidy 3.0.2
Dec 27 13:03:03 JaroniBox mopidy[3099]: INFO     [MainThread] mopidy.config Loading config from builtin defaults
Dec 27 13:03:03 JaroniBox mopidy[3099]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d
Dec 27 13:03:03 JaroniBox mopidy[3099]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d
Dec 27 13:03:03 JaroniBox mopidy[3099]: INFO     [MainThread] mopidy.config Loading config from file:///etc/mopidy/mopidy.conf
Dec 27 13:03:03 JaroniBox mopidy[3099]: INFO     [MainThread] mopidy.config Loading config from command line options
Dec 27 13:03:03 JaroniBox mopidy[3099]: ERROR    [MainThread] mopidy.internal.log Loading logging config '/etc/mopidy/logging.c
Dec 27 13:03:04 JaroniBox mopidy[3099]: INFO     [MainThread] mopidy.__main__ Enabled extensions: file, spotify, softwaremixer,
Dec 27 13:03:04 JaroniBox mopidy[3099]: INFO     [MainThread] mopidy.__main__ Disabled extensions: none
Dec 27 13:03:05 JaroniBox mopidy[3099]: INFO     [MainThread] mopidy.commands Starting Mopidy mixer: SoftwareMixer
Dec 27 13:03:05 JaroniBox mopidy[3099]: INFO     [MainThread] mopidy.commands Mixer volume set to 30
Dec 27 13:03:05 JaroniBox mopidy[3099]: INFO     [MainThread] mopidy.commands Starting Mopidy audio
Dec 27 13:03:05 JaroniBox mopidy[3099]: INFO     [MainThread] mopidy.commands Starting Mopidy backends: FileBackend, M3UBackend
lines 1-16...skipping...
-- Logs begin at Sun 2020-12-27 13:02:46 CET, end at Sun 2020-12-27 13:11:41 CET. --
Dec 27 13:02:57 JaroniBox systemd[1]: Starting Mopidy music server...
Dec 27 13:02:58 JaroniBox systemd[1]: Started Mopidy music server.
Dec 27 13:03:03 JaroniBox mopidy[3099]: INFO     [MainThread] mopidy.__main__ Starting Mopidy 3.0.2
Dec 27 13:03:03 JaroniBox mopidy[3099]: INFO     [MainThread] mopidy.config Loading config from builtin defaults
Dec 27 13:03:03 JaroniBox mopidy[3099]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy-local.conf
Dec 27 13:03:03 JaroniBox mopidy[3099]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy.conf
Dec 27 13:03:03 JaroniBox mopidy[3099]: INFO     [MainThread] mopidy.config Loading config from file:///etc/mopidy/mopidy.conf
Dec 27 13:03:03 JaroniBox mopidy[3099]: INFO     [MainThread] mopidy.config Loading config from command line options
Dec 27 13:03:03 JaroniBox mopidy[3099]: ERROR    [MainThread] mopidy.internal.log Loading logging config '/etc/mopidy/logging.conf' failed. 'formatters'
Dec 27 13:03:04 JaroniBox mopidy[3099]: INFO     [MainThread] mopidy.__main__ Enabled extensions: file, spotify, softwaremixer, local, iris, mpd, stream, http, m3u
Dec 27 13:03:04 JaroniBox mopidy[3099]: INFO     [MainThread] mopidy.__main__ Disabled extensions: none
Dec 27 13:03:05 JaroniBox mopidy[3099]: INFO     [MainThread] mopidy.commands Starting Mopidy mixer: SoftwareMixer
Dec 27 13:03:05 JaroniBox mopidy[3099]: INFO     [MainThread] mopidy.commands Mixer volume set to 30
Dec 27 13:03:05 JaroniBox mopidy[3099]: INFO     [MainThread] mopidy.commands Starting Mopidy audio
Dec 27 13:03:05 JaroniBox mopidy[3099]: INFO     [MainThread] mopidy.commands Starting Mopidy backends: FileBackend, M3UBackend, StreamBackend, SpotifyBackend, LocalBackend
Dec 27 13:03:06 JaroniBox mopidy[3099]: INFO     [Audio-2] mopidy.audio.actor Audio output set to "alsasink"
Dec 27 13:03:07 JaroniBox mopidy[3099]: INFO     [SpotifyEventLoop] mopidy_spotify.backend Logged in to Spotify in online mode
Dec 27 13:03:07 JaroniBox mopidy[3099]: ERROR    [SpotifyBackend-6] mopidy_spotify.web Authorization failed, not attempting Spotify API request. Please get new credentials from https://www.mopidy.com/authenti
Dec 27 13:03:07 JaroniBox mopidy[3099]: ERROR    [SpotifyBackend-6] mopidy_spotify.web OAuth token refresh failed: invalid_client Client not known.
Dec 27 13:03:07 JaroniBox mopidy[3099]: ERROR    [SpotifyBackend-6] mopidy_spotify.web Failed to load Spotify user profile
Dec 27 13:03:07 JaroniBox mopidy[3099]: INFO     [MainThread] mopidy.commands Starting Mopidy core
Dec 27 13:03:07 JaroniBox mopidy[3099]: INFO     [MainThread] mopidy.commands Starting Mopidy frontends: HttpFrontend, IrisFrontend, MpdFrontend
Dec 27 13:03:07 JaroniBox mopidy[3099]: INFO     [HttpFrontend-12] mopidy.http.actor HTTP server running at [::ffff:0.0.0.0]:6680
Dec 27 13:03:07 JaroniBox mopidy[3099]: INFO     [IrisFrontend-13] mopidy_iris.core Starting Iris 3.45.0
Dec 27 13:03:07 JaroniBox mopidy[3099]: INFO     [MainThread] mopidy_mpd.actor MPD server running at [::ffff:0.0.0.0]:6600
Dec 27 13:03:07 JaroniBox mopidy[3099]: INFO     [MainThread] mopidy.commands Starting GLib mainloop
Dec 27 13:03:07 JaroniBox mopidy[3099]: INFO     [MpdSession-15] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36128
Dec 27 13:03:07 JaroniBox mopidy[3099]: INFO     [MpdSession-17] mopidy_mpd.session New MPD connection from [::ffff:127.0.0.1]:36134
`

s-martin, 2020-12-27:
You are not correctly logged in to spotify

derdon222, 2020-12-27:
that is what I thought first too, bur Iris Settings look like this:
[image]

s-martin, 2020-12-27:
Please check again. A user opened an issue a couple of days ago with exactly the same scenario. Iris said in the web Interface authorized, but there was an error in the log. After redoing it everything worked.

@kingosticks, did you experience such behavior, too?

derdon222, 2020-12-27:
which steps should I redo?

kingosticks, 2020-12-27:
What behaviour? Iris Spotify credentials are separate to Mopidy's Spotify credentials.

s-martin, 2020-12-27:
> What behaviour? Iris Spotify credentials are separate to Mopidy's Spotify credentials.

Ok, that explains a lot ;) thanks 
I don’t use our Spotify edition myself, but I try to help our users.

derdon222, 2020-12-27:
I logged out from Spotify in the Iris interface an logged back in, forced a token refresh as well. With journalctl -u mopidy
 I still get the 
` [SpotifyEventLoop] mopidy_spotify.backend Logged in to Spotify in online mode
Dec 27 13:03:07 JaroniBox mopidy[3099]: ERROR    [SpotifyBackend-6] mopidy_spotify.web Authorization failed, not attempting Spotify API request. Please get new credentials from https://www.mopidy.com/authenti
Dec 27 13:03:07 JaroniBox mopidy[3099]: ERROR    [SpotifyBackend-6] mopidy_spotify.web OAuth token refresh failed: invalid_client Client not known.
Dec 27 13:03:07 JaroniBox mopidy[3099]: ERROR    [SpotifyBackend-6] mopidy_spotify.web Failed to load Spotify user profile
`part

derdon222, 2020-12-27:
So I have to fix the Mopidy Spotify credentials and that is not possible with iris, right?

kingosticks, 2020-12-27:
How did you force a token refresh?

derdon222, 2020-12-27:
in iris again

kingosticks, 2020-12-27:
You need to fix this for Mopidy. These are Mopidy-Spotify errors.

kingosticks, 2020-12-27:
I suggest obtaining new credentials from https://mopidy.com/authenticate/ and entering those credentials in the Mopidy config file.

derdon222, 2020-12-27:
thanks for this great support! Where do I find the mopidy config file? It is not under settings...

s-martin, 2020-12-27:
In 

` /etc/mopidy/mopidy.conf `
and
`~/.config/mopidy/mopidy.conf`

you need  to update the following parameters 
```
[spotify]
username = 
password = 
client_id = 
client_secret = 
```

s-martin, 2020-12-27:
We should add that to https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ or https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Spotify-FAQ

derdon222, 2020-12-27:
now it looks like this 
`journalctl -u mopidy
-- Logs begin at Sun 2020-12-27 14:08:03 CET, end at Sun 2020-12-27 14:08:27 CET. --
Dec 27 14:08:16 JaroniBox systemd[1]: Starting Mopidy music server...
Dec 27 14:08:16 JaroniBox systemd[1]: Started Mopidy music server.
Dec 27 14:08:22 JaroniBox mopidy[3823]: INFO     [MainThread] mopidy.__main__ Starting Mopidy 3.0.2
Dec 27 14:08:22 JaroniBox mopidy[3823]: INFO     [MainThread] mopidy.config Loading config from builtin defaults
Dec 27 14:08:22 JaroniBox mopidy[3823]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy-local.conf
Dec 27 14:08:22 JaroniBox mopidy[3823]: INFO     [MainThread] mopidy.config Loading config from file:///usr/share/mopidy/conf.d/mopidy.conf
Dec 27 14:08:22 JaroniBox mopidy[3823]: INFO     [MainThread] mopidy.config Loading config from file:///etc/mopidy/mopidy.conf
Dec 27 14:08:22 JaroniBox mopidy[3823]: INFO     [MainThread] mopidy.config Loading config from command line options
Dec 27 14:08:22 JaroniBox mopidy[3823]: ERROR    [MainThread] mopidy.internal.log Loading logging config '/etc/mopidy/logging.conf' failed. 'formatters'
Dec 27 14:08:23 JaroniBox mopidy[3823]: INFO     [MainThread] mopidy.__main__ Enabled extensions: iris, stream, file, local, m3u, http, softwaremixer, mpd, spotify
Dec 27 14:08:23 JaroniBox mopidy[3823]: INFO     [MainThread] mopidy.__main__ Disabled extensions: none
Dec 27 14:08:25 JaroniBox mopidy[3823]: INFO     [MainThread] mopidy.commands Starting Mopidy mixer: SoftwareMixer
Dec 27 14:08:25 JaroniBox mopidy[3823]: INFO     [MainThread] mopidy.commands Mixer volume set to 30
Dec 27 14:08:25 JaroniBox mopidy[3823]: INFO     [MainThread] mopidy.commands Starting Mopidy audio
Dec 27 14:08:25 JaroniBox mopidy[3823]: INFO     [MainThread] mopidy.commands Starting Mopidy backends: FileBackend, M3UBackend, StreamBackend, SpotifyBackend, LocalBackend
Dec 27 14:08:25 JaroniBox mopidy[3823]: INFO     [Audio-2] mopidy.audio.actor Audio output set to "alsasink"
Dec 27 14:08:26 JaroniBox mopidy[3823]: INFO     [SpotifyEventLoop] mopidy_spotify.backend Logged in to Spotify in online mode
Dec 27 14:08:27 JaroniBox mopidy[3823]: INFO     [SpotifyBackend-6] mopidy_spotify.web Logged into Spotify Web API as derdon222
`
and everything works fine! Thank you!!

closed 2020-12-27 by derdon222

kingosticks, 2020-12-27:
That client ID and secret can be reused for other Mopidy-Spotify installations for that Spotify account. Glad it's working.

baschior, 2021-02-15:
You made my day... Couldn't figure out iris and mdp have different credentials...
And also couldn't believe that wrong credentials would let me play albums nonetheless. Thanks for the help

baschior, 2021-02-15:
Actually the problem was using the spotify developer id ans secret instead of the client ones...
A link inbthe description where to get the credentials would be helpful...
Thank for your work, I love this project